Koenkk/zigbee2mqtt #3245: "Failed to call 'DeviceReport' 'onZigbeeEvent'" error on latest dev version of zigbee2mqtt

vke, 2020-03-30:
# Bug Report
## What happened

Click on any of two buttons of IKEA E1743 remote:

```6|z2m      | zigbee2mqtt:debug 2020-03-30 13:29:45: Received Zigbee message from 'onoff_switch01', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 901
6|z2m      | zigbee2mqtt:info  2020-03-30 13:29:45: MQTT publish: topic '/zigbee2mqtt/onoff_switch01', payload '{"linkquality":111,"last_seen":1585567785726,"update_available":true,"battery":34,"click":"on"}'
6|z2m      | zigbee2mqtt:error 2020-03-30 13:29:45: Failed to call 'DeviceReport' 'onZigbeeEvent' (TypeError: Cannot read property 'constructor' of null
6|z2m      |     at /opt/zigbee2mqtt/lib/extension/deviceReport.js:186:52
6|z2m      |     at Array.filter (<anonymous>)
6|z2m      |     at DeviceReport.poll (/opt/zigbee2mqtt/lib/extension/deviceReport.js:186:29)
6|z2m      |     at DeviceReport.onZigbeeEvent (/opt/zigbee2mqtt/lib/extension/deviceReport.js:163:18)
6|z2m      |     at Controller.callExtensionMethod (/opt/zigbee2mqtt/lib/controller.js:346:44))
```

The same behavior with IKEA E1524 five key remote:

```6|z2m      | zigbee2mqtt:debug 2020-03-30 13:59:59: Received Zigbee message from 'fivekey_remote02', type 'commandStepWithOnOff', cluster 'genLevelCtrl', data '{"stepmode":0,"stepsize":43,"transtime":5}' from endpoint 1 with groupID 12335
6|z2m      | zigbee2mqtt:info  2020-03-30 13:59:59: MQTT publish: topic '/zigbee2mqtt/fivekey_remote02', payload '{"linkquality":42,"last_seen":1585569599316,"update_available":true,"battery":34,"action":"brightness_up_click"}'
6|z2m      | zigbee2mqtt:error 2020-03-30 13:59:59: Failed to call 'DeviceReport' 'onZigbeeEvent' (TypeError: Cannot read property 'constructor' of null
6|z2m      |     at /opt/zigbee2mqtt/lib/extension/deviceReport.js:186:52
6|z2m      |     at Array.filter (<anonymous>)
6|z2m      |     at DeviceReport.poll (/opt/zigbee2mqtt/lib/extension/deviceReport.js:186:29)
6|z2m      |     at DeviceReport.onZigbeeEvent (/opt/zigbee2mqtt/lib/extension/deviceReport.js:163:18)
6|z2m      |     at Controller.callExtensionMethod (/opt/zigbee2mqtt/lib/controller.js:346:44))
```

## What did you expect to happen

No error log messages

## How to reproduce it (minimal and precise)

Just click on button of IKEA E1743 or E1524 remote on latest dev version of zigbee2mqtt (commit  #545eca6)

## Debug Info
zigbee2mqtt version: 1.12.0-dev (commit #545eca6)
CC253X firmware version: I'm using cc1352p, {"type":"zStack3x0","meta":{"transportrev":2,"product":1,"majorrel":2,"minorrel":7,"maintrel":1,"revision":20200312}}

Koenkk, 2020-03-30:
Can you check if this has been fixed in the latest dev branch?

vke, 2020-03-30:
> Can you check if this has been fixed in the latest dev branch?

Thank you! Fixed!

closed 2020-03-30 by Koenkk