Koenkk/zigbee2mqtt #3250: Error since adding Kwikset SMARTCODE_DEADBOLT_10_L

Botched1, 2020-03-31:
I know this has something to do with the homeassistant integration, but I am unsure how to fix it.

Since adding the SMARTCODE_DEADBOLT_10_L support I've been getting this error when the device sends updates:
```
zigbee2mqtt:info  2020-03-30 19:00:43: MQTT publish: topic 'zigbee2mqtt/Front Door Lock', payload '{"state":"UNLOCK","linkquality":117,"user":65535,"source":2,"battery":100,"last_seen":"2020-03-30T19:00:43-05:00"}'
(node:17) UnhandledPromiseRejectionWarning: TypeError: Cannot read property 'includes' of undefined
    at HomeAssistant.onPublishEntityState (/app/lib/extension/homeassistant.js:1529:55)
    at EventBus.<anonymous> (/app/lib/extension/homeassistant.js:1436:63)
    at EventBus.emit (events.js:311:20)
    at EventBus.emit (/app/lib/eventBus.js:14:15)
    at Controller.publishEntityState (/app/lib/controller.js:302:23)
    at publish (/app/lib/extension/deviceReceive.js:146:22)
    at DeviceReceive.onZigbeeEvent (/app/lib/extension/deviceReceive.js:176:13)
    at Controller.callExtensionMethod (/app/lib/controller.js:346:44)
    at Controller.onZigbeeEvent (/app/lib/controller.js:231:14)
    at Zigbee.emit (events.js:311:20)
(node:17) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 7)

```

Can someone give me a  pointer on what I need to update to make that go away? I'm sure it is simple, but it just isn't something I've had to do before.

Koenkk, 2020-03-31:
Can you update to the latest-dev first? I currently cannot match the line numbers with the current code.

Botched1, 2020-03-31:
> Can you update to the latest-dev first? I currently cannot match the line numbers with the current code.

I don't see the error in the current latest-dev docker. Sorry for the trouble.

closed 2020-03-31 by Botched1