Koenkk/zigbee2mqtt #3251: Coordinator is near but sensors connect to the router which is far away and the link very weak.

spider7611, 2020-03-31:
Have problem with the link quality. Coordinator the strong CC2530 + CC2591 and have one same type router. My problem is some aquara door sensor which is near to the coordinator not connect to him. Connect to the router with very weak link which more far away than the coordinator.

So 1 sensor is near to the coordinator (about 2 meter without wall) but this sensor connect to the router which is far away about 12 meter and 3 wall with very weak link...

How can I solve this problem?

zigbee2mqtt version: 1.12.0-dev
CC2530 + CC2591 coordinator firmware version: 2019.06.19
CC2530 + CC2591 router

Koenkk, 2020-03-31:
Xiaomi devices are known that stick to their parent where they were originally paired with. It's recommended to repair the sensor close to the desired parent.

Also as you are running the 20190619 firmware, not that this firmware can only have 5 direct children. (so if it has 5 children now it's expected that no more sensors connect to it).

spider7611, 2020-04-01:
Ok, paired it again and now it is working good.
Ok will see another firmware, hope will find this.

closed 2020-04-01 by spider7611