Koenkk/zigbee2mqtt #3253: Philips Hue Link Quality Not Reported

danieledwardgeorgehitchcock, 2020-03-31:
# Bug Report
## What happened
Philips Hue devices only report link quality and availability when they are triggered
## What did you expect to happen
Looking at a WireShark capture of when the devices were part of the Hue Bridge network, it would appear that devices periodically report link quality amongst current status. I would assume the same behaviou should occur on a Zigbee2MQTT network
## How to reproduce it (minimal and precise)
Add Hue devices (bulbs) to Zigbee2MQTT network and watch MQTT traffic and also device statuses in Zigbee2MQTT Assistant
## Debug Info
zigbee2mqtt version: 1.12.0
CC253X firmware version: 20190608

coolguymatt, 2020-03-31:
I also have this issue.

Koenkk, 2020-03-31:
It's expected that bulbs don't publish linkquality out-of-the-box as they don't send messages to the coordinator. In case you want to now the linkquality use the network map for bulbs (or setup reporting but it's not recommended if this is your only use case)

closed 2020-03-31 by Koenkk