Koenkk/zigbee2mqtt #3254: How to backup and restore ？

ShallowGreen, 2020-03-31:
Is it possible to do gateway backup and restore？
I backup the data folder，And replace the data folder on another computer
I shutdown the before computer to avoid panid duplication

But the panId still increased by one, because of the router or endpoint Occupied before panId

And How to tell if it was restarted or replaced？

Koenkk, 2020-03-31:
A copy of the data folder should do it, in case you are using a CC26X2R1 or CC1352P2 make sure to also copy `coordinator_backup.json`.

Botched1, 2020-03-31:
> A copy of the data folder should do it, in case you are using a CC26X2R1 or CC1352P2 make sure to also copy `coordinator_backup.json`.

Is it really that simple to changer coordinator hardware? So if my CC26x2R1 dies, I connect up a new one and then start Docker up. Magically all works with no re-pairing?

ShallowGreen, 2020-04-01:
> > A copy of the data folder should do it, in case you are using a CC26X2R1 or CC1352P2 make sure to also copy `coordinator_backup.json`.
> 
> Is it really that simple to change coordinator hardware? So if my CC26x2R1 dies, I connect up a new one and then start Docker up. Magically all works with no re-pairing?

This is what I want to do, with no re-pairing to restore. But there still some problems, for example,
the new one start, the panId will auto increased(debug), routers and endpoints will lost.

ShallowGreen, 2020-04-01:
> A copy of the data folder should do it, in case you are using a CC26X2R1 or CC1352P2 make sure to also copy `coordinator_backup.json`.

How to solve the problem of self-incrementing panid when the new one start. Causing the device to lose connectivity

ShallowGreen, 2020-04-01:
I have solved the problem, The condition for restore is the new one should **not startup** or **startup with the old one PanID**, I made UI for users to restore, so I must startup the new one. when I startup, the panID should the same with the old one, then I restore the data folder and restart, routers and endpoints all works with no re-pairing. anyway, thanks!

closed 2020-04-01 by ShallowGreen

nonyhaha, 2021-02-02:
Hello. Sorry to bring this back up. do you know what folder should be backed on a home assistant supervised instance setup as a vm from the official image? 

i could find a zigbee2mqtt folder in /share/ but there is no data folder there.

many thanks.

oberdannunes, 2025-01-24:
Hi, I'm seeing that this post is a little old, but I couldn't find so much information about it. I had a problem with my stack folder and loose all my installation. Now I have the backup of configuration.yaml, coordinator_backup.json and state.json, but I don't have the database.db. I've recreated the container, using those files and I'm able to open zigbee2mqtt but I can't see any devices. 

Is there a way to restore them? I'm using ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus.

Tks!