Koenkk/zigbee2mqtt #3257: Redo device interview by mqtt request

vke, 2020-03-31:
Hello.

Can zigbee2mqtt do yet another device interview on mqtt reqest? I have to do this again due to a change in device firmware during development. Now I have to remove and pair the device again. I read the documentation, but I do not see such an opportunity there.

## Debug Info
zigbee2mqtt version: latest dev-version of zigbee2mqtt
CC253X firmware version: cc1352p with Zigbee 3 stack

Koenkk, 2020-03-31:
You can re-trigger an interview by
- Stop z2m
- Change `"interviewCompleted":false` in the `database.db` for this device
- Start z2m
- Let the device send a message

vke, 2020-03-31:
Thank you, this is enough for me, closing issue.

closed 2020-03-31 by vke

frankis78, 2022-11-09:
> You can re-trigger an interview by
> 
> * Stop z2m
> * Change `"interviewCompleted":false` in the `database.db` for this device
> * Start z2m
> * Let the device send a message

Hi, i have the same question after upgrading a loot of Aqara T1 switches, and they all have very different binding, reporting, cluster settings. So i want to wipe it and reinterview the devices so that they get new settings.

Regards to this i have 3 simple questions that might help others aswell.
1. After it has been changed to "interviewCompleted":false in database.db file, 
    is it enough to click any button on the device, or does the device needs to be in pairing/including mode ?
2. Is it needed to set "Permit Join":True ?
3. Is it needed to click on the orange "Reconfigure" button on the device in the UI ?
Thank you.