Koenkk/zigbee2mqtt #3258: Follow homie convetion for experimental attribute for setters

krzyk, 2020-03-31:
Reading attributes follows [homie convention](https://homieiot.github.io/) (when I use `experimental: output: attribute`): `homie/zigbee/notification/state`.

But when I try to set values I have to use: `homie/zigbee/notification/set/state` instead of the same attribute as for reading.

I think homie enforces usage of the same tree path for that (and it doesn't allow such deep nesting).

sjorge, 2020-03-31:
I'm kind of surprised reading works, as z2m doesn't follow the homie convention at all (well if it does, it is by pure luck).

I think to get everything to work in homie something like this:
https://github.com/Koenkk/zigbee2mqtt/blob/master/lib/extension/homeassistant.js

Will have to be written, but for homie obviously.

krzyk, 2020-04-01:
@sjorge well, it does (if you change the main topic name to `homie` and provide the boiler plate and leave just the attributes for zigbee2mqtt) - I can succesfully use it in openhab with auto mqtt discovery.

The only issue is the `/set` part for settable attributes. If `/set` would be optional then it would magically work.

Koenkk, 2020-04-01:
Zigbee2mqtt indeed doesn't support the Homie convention (it was worked on sometime ago but never finished: https://github.com/Koenkk/zigbee2mqtt/pull/855), so everything that currently does currently work isn't intended to. 

As adding Homie supports seems to be a lot of work and there is not a lost of interest from the community ATM, an easy workaround would be to write an intermediate python script which does the homie -> z2m translation.

sjorge, 2020-04-01:
That old PR might still be a good starting point if someone wants to pick it up. It doesn’t look bad code wise at a quick glance.

closed 2020-06-07 by stale[bot]

flatsiedatsie, 2021-01-29:
Just wanted to say that following the Homie convention would allow Zigbee2MQTT to instantly work with a lot of smart home controllers nowadays. 

Although it could be most of those are already supported.