Koenkk/zigbee2mqtt #3261: Cannot pair with SureFlap Microchip PetFlap

koenbulcke, 2020-04-01:
Hi all,

After having followed the tutorial (All latest available software: firmware flashing, host software setup, etc. all without problem, thank you very much for the clear instructions!!), I tried to pair with a Zigbee supporting "door pet flap" branded "Sureflap"   (product details at: https://s3-eu-west-1.amazonaws.com/sureflap/website/support/manuals/97/multi-lang/manual.pdf  and https://fccid.io/XO9-IMPD00003/Test-Report/Test-Report-3666566 ).

However, I cannot see that the USB stick "discovers" the door flap (put in pairing mode), just nothing happens and I have no idea how to continue (like to force a search or discovery for pariring):

My command log (Ubuntu 19.10 based):

> zigbee2mqtt@1.12.0 start /opt/zigbee2mqtt
> node index.js

zigbee2mqtt:info  2020-03-29 13:19:35: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2020-03-29.13-19-35' filename: log.txt
zigbee2mqtt:debug 2020-03-29 13:19:35: Removing old log directory '/opt/zigbee2mqtt/data/log/2020-03-29.12-17-55'
zigbee2mqtt:debug 2020-03-29 13:19:35: Loaded state from file /opt/zigbee2mqtt/data/state.json
zigbee2mqtt:info  2020-03-29 13:19:35: Starting zigbee2mqtt version 1.12.0 (commit #840b9d9)
zigbee2mqtt:info  2020-03-29 13:19:35: Starting zigbee-herdsman...
zigbee2mqtt:debug 2020-03-29 13:19:35: Using zigbee-herdsman with settings: '{"network":{"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[15],"networkKey":"HIDDEN"},"databasePath":"/opt/zigbee2mqtt/data/database.db","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/ttyACM0"}}'
zigbee2mqtt:info  2020-03-29 13:19:37: zigbee-herdsman started
zigbee2mqtt:info  2020-03-29 13:19:37: Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}}'
zigbee2mqtt:debug 2020-03-29 13:19:37: Zigbee network parameters: {"panID":6754,"extendedPanID":"0xdddddddddddddddd","channel":15}
zigbee2mqtt:info  2020-03-29 13:19:37: Currently 0 devices are joined:
zigbee2mqtt:warn  2020-03-29 13:19:37: `permit_join` set to  `true` in configuration.yaml.
zigbee2mqtt:warn  2020-03-29 13:19:37: Allowing new devices to join.
zigbee2mqtt:warn  2020-03-29 13:19:37: Set `permit_join` to `false` once you joined all devices.
zigbee2mqtt:info  2020-03-29 13:19:37: Zigbee: allowing new devices to join.
zigbee2mqtt:info  2020-03-29 13:19:37: Connecting to MQTT server at mqtt://localhost
zigbee2mqtt:info  2020-03-29 13:19:37: Connected to MQTT server
zigbee2mqtt:info  2020-03-29 13:19:37: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
zigbee2mqtt:info  2020-03-29 13:19:37: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.12.0","commit":"840b9d9","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}},"log_level":"debug","permit_join":true}'

-------------
Afterwards, I consequently flashed the sniffer firmware and could discover the door flap on channel 15 (broadcast packets seen in Wireshark) so that seems all to be fine (see attached screenshot)

Hope that somebody can help me with some instructions what I could do? I feel I'm close to pairing the Sureflap petflap but just missing a few critical steps :)

If I'm missing some info, please let me know, this is my first week into the Zigbee and MTQQ world.
[image]

Thanks in advance,
Koen.

Koenkk, 2020-04-01:
Normally after a beacon request you see a beacon response, example:

[image]

In this case it doesn't seem to happen, I wonder if that has to do with malformed message Wireshark is talking about ("Inavlid addressing for beacon request").

With what bridge is this device normally used with?

koenbulcke, 2020-04-01:
Hi,

The device is a dog flap that allows a dog to come in or come out of the house (basically unlocks the dog flap), it has two radio's:
 - for near field RFID tag that the dog carries and basically authenticates the animal.
 - for an optional Zigbee network that connects to a hub (that then basically further connects over Wifi and probably a cloud infrastructure to a mobile app where the owner can see the pet's activity). This is ilustrate on page 14 of the following PDF: https://s3-eu-west-1.amazonaws.com/sureflap/website/support/manuals/97/multi-lang/manual.pdf and see attached screenshot from the FCC certification of the device.

Basically, I would like to all bypass that (expensive) 3rd party hub and infrastructure and have a (later) RBI Zero doing the bridge and further signalling.

Could I do/try something more? I suppose Zigbee channel 15 is the only channel there would be traffic of is this like a multichannel protocol? (I have technical background but Zigbee is really new to me).

Thanks!
br, Koen.

[image]

Koenkk, 2020-04-01:
When I search for "Zigbee" in the manual it doesn't yield any results. Are you sure it supports Zigbee? (it could also be any other 802.15.4 based protocol)

koenbulcke, 2020-04-01:
Ha good point, is there anyway to figure it out?

I'm googling a bit, and some make suggest of Bluetooth but that cannot work over 802.15.4 as physical layer?

Here:  https://github.com/alextoft/sureflap/blob/master/README.md I found the chipset but no more "practical" details :
_It’s all Microchip stuff, based around a 32bit/80MHz PIC32MX695F MIPS SoC with 512KB flash, 128KB RAM, 10/100 Ethernet and USB2 OTG. Also on the board are a WRF24J48MA 2.4GHz wireless module (to communicate with the flap) and obligatory ENC424J600 Ethernet controller to manage it._ 

I'll try to look further to find something more. Seems that most people buy the hub and then use home-automation on top of it (at least try). I just wanted a more "hacky" solution and connect something directly to the dog flap.

koenbulcke, 2020-05-04:
Hi,

@Koenkk  ; you were right, the "upper layer" protocol on the 2.4 GHz 802.15.4 seems to be Microchip's proprietary "MiWi P2P" that is being used. So zigbee2mqtt won't be possible to use, thans neverhteless

br, Koen.

closed 2020-05-04 by koenbulcke