Koenkk/zigbee2mqtt #3264: Interview fails OSRAM Smart+ gardenpole 8.7W RGBW

julian2701, 2020-04-01:
# Bug Report
## What happened
Interviews fails every time when trying to pair "Smart+ gardenpole 8.7W RGBW"

## What did you expect to happen
Pair device

## How to reproduce it (minimal and precise)
Interviewed and resetted device many times, restarted zigbee2mqtt many times

## Debug Info
zigbee2mqtt version: 1.11.0
CC253X firmware version: 20190608
Log: https://pastebin.com/auAP4pWn

Tom85649, 2020-04-05:
Hi Julian,
I've got 4 garden pole units, and all of them I can only pair directly with the coordinator. For me it never worked with routers in between.  Maybe you got the same prob ? So what I do is bring them close to the coordinator, pair them, and then bring them to their final destination.
KR Tom

Koenkk, 2020-04-05:
Can you try again with the latest dev branch? (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

julian2701, 2020-04-06:
Thank you for your replies.

@Koenkk Installing dev-branch lead to a more detailed error, but interview failed although. Log: https://pastebin.com/YVH9se34

@Tom85649 Thank you, I will try to do that in the afternoon, need to get my gardenpoles out of the garden, because I managed to successfully join the LEDs 3 weeks ago until they were not reachable any more the last days ("no ack")

Koenkk, 2020-04-06:
I've pushed another possible fix to the dev branch, can you try pairing it at the wanted location?

julian2701, 2020-04-08:
> I've pushed another possible fix to the dev branch, can you try pairing it at the wanted location?

Thank you, Koenkk! I moved my raspberry just an hour before your answer right near to the gardenpoles, and from then on it was working fine. I think it was the distance and now also all other devices are working. Does the new dev branch and a log might also be interesting for you?

Koenkk, 2020-04-09:
If its working OK now no further action is required, could you close this issue?

closed 2020-06-17 by stale[bot]

odelma, 2021-09-09:
Hello, I am running 1.21.0-dev commit: fc2e8304. I tried to pair my Ledvance (I assume it is quite the same model as sold under OSRAM brand) Gardenpole Mini, but the interview process keeps failing with similar log info as presented here. Is the solution to try to pair it closer to controller (zzh)?

In database I only see:
`{"id":53,"type":"Unknown","ieeeAddr":"0x7cb03eaa0a078e9a","nwkAddr":23902,"epList":[],"endpoints":{},"interviewCompleted":false,"meta":{},"lastSeen":16xxxxxxxxxxx}` 

I assume this model should be supported?
[image]