Koenkk/zigbee2mqtt #3265: Zigbee2MQTT doesn't find new devices

coolcoons, 2020-04-01:
## What happened
Zigbee2MQTT didn't find new devices.
## What did you expect to happen
I thought that resetting my lights would connect them to Zigbee2MQTT.
## How to reproduce it (minimal and precise)
I started eclipse-mosquitto:latest container in docker. I started MQTT-Explorer. I started zigbee2mqtt in PowerShell. I turned on my lights and made sure they were reset. Then I started svrooij/diyhue:latest in docker.  All three servers connected to the mosquitto server. Then I went into the Philips Hue app and connected to the diyHue server. I told it to look for lights and it sent this in the log. https://pastebin.com/kGD5YpBx Here is the Zigbee2MQTT log. https://pastebin.com/p2pwv1u1 Here is MQTT-Explorer 
[image] Here is Mosquitto. https://pastebin.com/YyU0kGgU

Svrooji made it so diyHue can integrate with the MQTT messaging system and connect to Zigbee2MQTT. Then we can use zigbee devices with the server instead of just ESP8266. I don't know much about Zigbee2MQTT, but isn't it supposed to search for devices on its own without a command from homeassistant or in my case diyHue?
## Debug Info
zigbee2mqtt and CC2531 versions : {"version":"1.12.0","commit":"unknown","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}},"log_level":"debug","permit_join":true}

coolcoons, 2020-04-02:
I switched the ZigBee channel to 25 instead of 11 and it found the lights. It isn't a supported device, and I read this https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html but I don't have enough info from the first time I tried to connect to add it to supported devices. Here is the log file when I tried to connect to the lightbulb.
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/4423427/log.txt)

coolcoons, 2020-04-02:
It says that the devices zigbeeModel is 'undefined'. The vendor model number is also already supported so I don't know why I am having problems at all.  https://www.zigbee2mqtt.io/devices/73739.html My device is a Sylvania Smart+ BR30 RGBW LED Bulb. (Here is a link to it. https://assets2.sylvania.com/media/bin/asset-3466185/SYLVANIA%20SMART%2B%20ZigBee%20BR30%20Full%20Color.pdf) The device that is already supported is an Osram Lightify BR30 RGBW LED Bulb.

Koenkk, 2020-04-03:
Make sure to check https://www.zigbee2mqtt.io/information/FAQ.html#why-does-my-device-not-or-fail-to-pair

coolcoons, 2020-04-04:
Here is the Log after trying the fixes you sent me. https://pastebin.com/ikmpHbY7 It tries to interview multiple times and fails but the device name is '0xf0d1b800000645ff'. Would that be the last issue?
If device joins with 0x000000000000000 as ieeeAddress (you will see: Starting interview of '0x0000000000000000' in the Zigbee2mqtt log) your CC253X might be broken. See issue #2761

Koenkk, 2020-04-04:
No that's not the issue, can you provide the herdsman debug logging when pairing the device?

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

coolcoons, 2020-04-04:
https://pastebin.com/6NEV4v5f Here it is.

Koenkk, 2020-04-05:
Can you try again with the latest dev branch? (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

coolcoons, 2020-04-06:
The dev-branch didn't change anything. Here's the new log file with zigbee-herdsman-debug. https://pastebin.com/PH0UV17B

Koenkk, 2020-04-06:
Sorry I made a mistake with the fix, please try again with the lates dev branch.

coolcoons, 2020-04-07:
My first light bulb '0xf0d1b8000006671c' connected on the first try and works fine but the second one, '0xf0d1b800000645ff', has this error message:   zigbee-herdsman:controller:error Interview failed for '0xf0d1b800000645ff with error 'Error: Interview failed because can not get node descriptor ('0xf0d1b800000645ff')' +0ms https://pastebin.com/CuHbiWRw

coolcoons, 2020-04-07:
I think this is the problem: zigbee-herdsman:controller:device:log Interview - node descriptor request failed for '0xf0d1b800000645ff'

Koenkk, 2020-04-08:
Can you try with zigbee2mqtt 1.12.1 and pair the device close to the coordinator?

coolcoons, 2020-04-09:
'0xf0d1b800000645ff' still won't connect. https://pastebin.com/6GYzs7sS. I have the coordinator within 3 inches of the base of the lightbulbs. When I wasn't doing that it wouldn't find the lights at all.

coolcoons, 2020-04-09:
I tried reinstalling Zigbee2MQTT and the first light, '0xf0d1b8000006671c', connects fine but does have an error at the end: zigbee-herdsman:controller:error Interview failed for '0xf0d1b8000006671c with error 'Error: Interview failed because can not get active endpoints ('0xf0d1b8000006671c')' +0ms. https://pastebin.com/7zyj8TdP The second one, '0xf0d1b800000645ff', still has the same error though and won't connect: zigbee-herdsman:controller:error Interview failed for '0xf0d1b800000645ff with error 'Error: Interview failed because can not get node descriptor ('0xf0d1b800000645ff')' +0ms https://pastebin.com/iwhQcpXA

Koenkk, 2020-04-11:
@coolcoons this looks like some intereference problems. is the CC2531 connected through an USB extension cable?

coolcoons, 2020-04-11:
I have it connected through a USB extension cable. We also have a strong wifi connection in the house which might be interfering with it. I think the ZigBee and wifi channels were overlapping but when I switched the ZigBee channel to something that wouldn't conflict the lightbulbs couldn't be found.

coolcoons, 2020-04-15:
Should I turn off the wifi while I am pairing the bulbs?

Koenkk, 2020-04-16:
you can try

coolcoons, 2020-04-17:
I reflashed the CC2531, reinstalled Zigbee2MQTT and Node.js and reset the lightbulbs. I then turned off the Wifi and even Bluetooth to be sure nothing was interfering and it is still having issues. I tried connecting '0xf0d1b8000006671c' again. Do you want me to try some other things or do you need some more info? https://pastebin.com/UcU5gCJc

coolcoons, 2020-04-17:
These issues all look very similar or the same just not with my light. Do any of them contain fixes I could use or does it only work on the Xiamoi devices that these people are using? 
https://github.com/Koenkk/zigbee2mqtt/issues/837
https://github.com/Koenkk/zigbee2mqtt/issues/1504
https://github.com/Koenkk/zigbee2mqtt/issues/1779

Koenkk, 2020-04-17:
@coolcoons for Xiaomi devices this is related to something different which has already been solved in later versions.

Based on the logging I don't know what we can do, maybe sniffing will reveal more info: https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

coolcoons, 2020-04-19:
I tried every zigbee channel in zboss 11-26 and it found the lightbulbs on every channel. I guess I will run Zigbee2MQTT on its default channel 11 and see if it can still find the lights.

Koenkk, 2020-04-21:
You need to sniff at the channel that Zigbee2mqtt is running on (which currently doesn't seem to be the case).

coolcoons, 2020-04-22:
So I need to run Zigbee2MQTT while sniffing with Zboss+Wireshark?

Koenkk, 2020-04-22:
yes

coolcoons, 2020-04-22:
I need two CC2531's right? Since I can't flash two firmwares onto one device?

coolcoons, 2020-04-22:
I don't have two CC2531's at the moment.

Koenkk, 2020-04-22:
Yes you need two

coolcoons, 2020-04-22:
I'm making an order on aliexpress. It should be here in the next month. Do you want me to close this issue and then open it again in a month?

Koenkk, 2020-04-23:
Feel free to keep it open.

coolcoons, 2020-06-22:
I thought the Zigbee dongle would be here months ago. It says it may take longer than normal to get here due to Covid19. I hope it will be here soon and I will let you know when it arrives.

closed 2020-08-01 by stale[bot]

intositeme, 2020-10-28:
Can't seem to connect to E1743 Tradfri On/OFF switch. Previously had it working.
```
> node index.js
Zigbee2MQTT:info  2020-10-28 15:59:24: Logging to console and directory: '/share/zigbee2mqtt/log/2020-10-28.15-59-24' filename: log.txt
Zigbee2MQTT:info  2020-10-28 15:59:25: Starting Zigbee2MQTT version 1.15.0 (commit #1.15.0)
Zigbee2MQTT:info  2020-10-28 15:59:25: Starting zigbee-herdsman...
2020-10-28T07:59:26.059Z zigbee-herdsman:adapter Failed to validate path: 'Error: spawn udevadm ENOENT'
CREATED DECONZ ADAPTER
2020-10-28T07:59:26.066Z zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13],"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"path":"/dev/ttyACM0","adapter":"deconz"},"databasePath":"/share/zigbee2mqtt/database.db","databaseBackupPath":"/share/zigbee2mqtt/database.db.backup","backupPath":"/share/zigbee2mqtt/coordinator_backup.json","adapter":{"concurrent":null}}'
2020-10-28T07:59:26.069Z zigbee-herdsman:deconz:driver Opening with /dev/ttyACM0
2020-10-28T07:59:26.085Z zigbee-herdsman:deconz:driver Serialport opened
2020-10-28T07:59:26.088Z zigbee-herdsman:controller:log Started with result 'resumed'
2020-10-28T07:59:26.088Z zigbee-herdsman:controller:log Injected database: true, adapter: true
2020-10-28T07:59:26.114Z zigbee-herdsman:deconz:driver send read parameter request from queue. seqNr: 1 paramId: 1
2020-10-28T07:59:26.128Z zigbee-herdsman:deconz:frameParser MAC: 0x00212effff05ce97
2020-10-28T07:59:26.163Z zigbee-herdsman:deconz:driver send read parameter request from queue. seqNr: 2 paramId: 7
2020-10-28T07:59:26.172Z zigbee-herdsman:deconz:frameParser NWKADDR: 0
Zigbee2MQTT:info  2020-10-28 15:59:26: zigbee-herdsman started
2020-10-28T07:59:26.214Z zigbee-herdsman:deconz:driver send read firmware version request from queue. seqNr: 3
2020-10-28T07:59:26.220Z zigbee-herdsman:deconz:frameParser read firmware version response - version: 0,7,102,38
Zigbee2MQTT:info  2020-10-28 15:59:26: Coordinator firmware version: '{"meta":{"maintrel":0,"majorrel":38,"minorrel":102,"product":0,"revision":"0x26660700","transportrev":0},"type":"ConBee2/RaspBee2"}'
2020-10-28T07:59:26.264Z zigbee-herdsman:deconz:driver send read parameter request from queue. seqNr: 4 paramId: 5
2020-10-28T07:59:26.268Z zigbee-herdsman:deconz:frameParser PANID: 1a62
2020-10-28T07:59:26.314Z zigbee-herdsman:deconz:driver send read parameter request from queue. seqNr: 5 paramId: 11
2020-10-28T07:59:26.318Z zigbee-herdsman:deconz:frameParser APS_EXT_PANID: 0xdddddddddddddddd
2020-10-28T07:59:26.364Z zigbee-herdsman:deconz:driver send read parameter request from queue. seqNr: 6 paramId: 28
2020-10-28T07:59:26.368Z zigbee-herdsman:deconz:frameParser CHANNEL: 11
2020-10-28T07:59:26.414Z zigbee-herdsman:deconz:driver send read parameter request from queue. seqNr: 7 paramId: 24
2020-10-28T07:59:26.419Z zigbee-herdsman:deconz:frameParser NETWORK_KEY: hidden
Zigbee2MQTT:info  2020-10-28 15:59:26: Currently 0 devices are joined:
Zigbee2MQTT:warn  2020-10-28 15:59:26: `permit_join` set to  `true` in configuration.yaml.
Zigbee2MQTT:warn  2020-10-28 15:59:26: Allowing new devices to join.
Zigbee2MQTT:warn  2020-10-28 15:59:26: Set `permit_join` to `false` once you joined all devices.
Zigbee2MQTT:info  2020-10-28 15:59:26: Zigbee: allowing new devices to join.
2020-10-28T07:59:26.438Z zigbee-herdsman:controller:log Permit joining
Zigbee2MQTT:info  2020-10-28 15:59:26: Started frontend on port 8099
2020-10-28T07:59:26.454Z zigbee-herdsman:deconz:driver DATA_REQUEST - destAddr: 0xfffc EP:0 SeqNr. 8 request id: 1
2020-10-28T07:59:26.457Z zigbee-herdsman:deconz:frameParser DATA_REQUEST RESPONSE - request id: 1 status: 0
2020-10-28T07:59:26.458Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2020-10-28T07:59:26.590Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10100110
2020-10-28T07:59:26.590Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 1 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2020-10-28T07:59:26.604Z zigbee-herdsman:deconz:driver DATA_CONFIRM - sending data state request - SeqNr. 9
2020-10-28T07:59:26.610Z zigbee-herdsman:deconz:frameParser DATA_CONFIRM RESPONSE - destAddr: 0xfffc request id: 1 confirm status: 0
2020-10-28T07:59:26.611Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2020-10-28T07:59:26.612Z zigbee-herdsman:deconz:adapter PERMIT_JOIN - 254 seconds
2020-10-28T07:59:26.622Z zigbee-herdsman:deconz:adapter zclFrame to all - zclFrame.payload:
2020-10-28T07:59:26.630Z zigbee-herdsman:deconz:adapter { options: 11, commisioningWindow: 254 }
2020-10-28T07:59:26.630Z zigbee-herdsman:deconz:adapter sendZclFrameToAll - message send
2020-10-28T07:59:26.632Z zigbee-herdsman:deconz:driver send write parameter request from queue. seqNr: 10 paramId: 33 param: 254
2020-10-28T07:59:26.635Z zigbee-herdsman:deconz:frameParser write parameter response - parameter id: 33 - status: 0
2020-10-28T07:59:26.654Z zigbee-herdsman:deconz:driver DATA_REQUEST - destAddr: 0xfffd EP:242 SeqNr. 11 request id: 2
2020-10-28T07:59:26.658Z zigbee-herdsman:deconz:frameParser DATA_REQUEST RESPONSE - request id: 2 status: 0
[... cut ...]
2020-10-28T08:00:26.075Z zigbee-herdsman:deconz:frameParser device state: 10100010
2020-10-28T08:00:26.076Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2020-10-28T08:00:36.078Z zigbee-herdsman:deconz:driver send read device state from queue. seqNr: 27
2020-10-28T08:00:36.081Z zigbee-herdsman:deconz:frameParser device state: 10100010
2020-10-28T08:00:36.082Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
```