Koenkk/zigbee2mqtt #3268: Docker-compose does not recognize device

MartijndeHeus, 2020-04-01:
# Bug Report
## What happened
When I run the service as part of a docker-compose it does not recognize my device. I can run Zigbee2mqtt when I clone the repo using npm start. Following error: 
```
zigbee2mqtt:error 2020-04-01 18:27:16: Error: Error while opening serialport 'Error: Error: No such file or directory, cannot open /dev/tty.usbmodem14501
```

## What did you expect to happen
Zigbee2mqtt start normally, similar to when I run it using the original project.

## How to reproduce it (minimal and precise)
Flash the CC2531, plug it in a USB-C adaptor, run docker-compose:

```
zigbee2mqtt:
        container_name: zigbee2mqtt
        image: koenkk/zigbee2mqtt:1.10.0
        volumes: 
            - ./zigbee2mqtt-data/:/app/data/
        devices:
            - /dev/tty.usbmodem14501
        privileged: true
```

I do also have ```/dev/tty.usbmodem14501``` in my ```configuration.yaml```.

## Debug Info
zigbee2mqtt version: 1.12.0
CC253X firmware version: CC2531_DEFAULT_20190608

MartijndeHeus, 2020-04-01:
I found that this is actually a problem with Docker and Mac. I am now trying to get it to work with docker-machine.

Koenkk, 2020-04-01:
Docker on mac doesnt allow device passthrough AFAIK

MartijndeHeus, 2020-04-01:
Yes, this is true. I am now passing it to a VirtualBox where I run a docker-machine with boot2docker. I can find it in the VirtualBox under the following path: ```/dev/bus/usb/001/003```

But I get this error: ```'Error: Error: Not a tty setting custom baud rate of 115200'```

Does this sound familiar to you? I am sorry I am really new to using hardware.

Koenkk, 2020-04-01:
Do you see it under: ls /dev ?

MartijndeHeus, 2020-04-01:
No. I can only seem to find it in this path ```/dev/bus/usb/001/003```. Here is also disappears and comes back when I plug in and out.

It is possible to show up in 2 places? I will look again in other places in ```/dev```.

MartijndeHeus, 2020-04-01:
I think I found the issue. I was passing it to VirtualBox as a USB, while I think I should pass it as a 'Serial Port'. VirtualBox now seems to crash when I start it.

I am not sure why it is crashing now. Trying to figure it out now.

Maybe it has to do with how to configure the IRQ and I/O port on Mac OS for this to work.

MartijndeHeus, 2020-04-01:
I can not seem to find a solution for now. I think I read somewhere that you develop this on a Mac, do you solely do this using `npm` or do you have any suggestions?

Koenkk, 2020-04-02:
I only start it with `npm start`

MartijndeHeus, 2020-04-02:
Ah okay. I think I will run my docker-compose and then run zigbee2mqtt seperately while working on my Mac to workaround this. 

I will close this issue now as it is not a problem with zigbee2mqtt. Thanks a lot for the help and creating/maintaining zigbee2mqtt!

closed 2020-04-02 by MartijndeHeus

SHxKM, 2021-06-19:
@martijndeheus - I know this is pretty old, but any chance you found a workaround for running Z2M inside a docker container on a Mac?

martijndeheus, 2021-06-21:
> @martijndeheus - I know this is pretty old, but any chance you found a workaround for running Z2M inside a docker container on a Mac?

No. I think it might be doable, but for me it ended up being easier to run other services using docker compose and Z2M standalone.

zella, 2021-06-26:
I run on intel-nuc with ubuntu:
```
docker run \
-it \
-v $(pwd)/data:/app/data \
--device=/dev/bus/usb/001/006:/dev/ttyACM0 \
-e TZ=Europe/Amsterdam \
-v /run/udev:/run/udev:ro \
--privileged=true \
koenkk/zigbee2mqtt
```
And gets:
` Error: Error while opening serialport 'Error: Error: Not a tty setting custom baud rate of 115200'`

robwolff3, 2023-02-21:
@zella I got mine to work by referencing it as /dev/ttyACM0 on both the host machine and the container device. Referring to it by /dev/bus/usb/ would not work.