Koenkk/zigbee2mqtt #3269: unstable / slow performance

rmalchow, 2020-04-01:
# Bug Report
## What happened
i have switched all devices from my hue bridge to z2m. when i switch single devices, all works fine. as soon as i switch multiple devices, all goes wonky. some of the commands are delivered eventually, but may devices never do the switch, and what gets deliver gets delivered with up to 20s delay. and subsequent commands also get caught up in this. it takes about a minute to stabilize again. then, single switching is ok again. 

at the moment, i am not using any hardware accessories for automation - all is done through hass UI. those device are connected, though.

i am running a CC2531, with z2m dockerized and /dev/ttyACM0 mounted to /dev/ttyAM0

zigbee2mqtt:info  2020-04-01 18:55:31: Currently 46 devices are joined:`

the devices are:
- about half lamps (mains powered), mostly philips with a few tradfri
- other half accessories (mixed bag of philips, tradfri and xiaomi)

## What did you expect to happen

devices switch (eventually). i was hoping to find out how i can tweak the queue settings so that i can maybe play with concurrency and delay between commands a bit?

## How to reproduce it (minimal and precise)

switch more than 2 or 3 devices simultanously

## Debug Info

this is my network map:

[image]

notice the 4 balls in the top right corner: those are 1 hue dimmer, 1 aqara door sensor, 1 tradfri motion sensor and one hue motion sensor. they are not connected to the graph, but i DO receive messages from them (unreliably)

zigbee2mqtt version:
```Starting zigbee2mqtt version 1.12.0 (commit #840b9d9)```

CC253X firmware version:
```Coordinator firmware version: '{"type":"zStack12","meta": "transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190619}}' ```

rmalchow, 2020-04-01:
i have 4 old tradfri gu10s ... very early models. i am currently trying a OTA update of these bulbs. i had 3 devices fail so far, one with "device did not respond", 2 with "cancelled by device" at <10% ... 4th attempt is at > 50% after ~10m

Koenkk, 2020-04-01:
46 devices is a lot for he CC2531, when having such a large network a more powerful coordinator is recommended: https://www.zigbee2mqtt.io/information/supported_adapters.html#texas-instruments-cc26x2r1

rmalchow, 2020-04-02:
hi. ok. so let's see how digikey delivers these days. i have quite a few non-essentials "toys" on the network. for now, i'll remove some and see when it becomes usable. thank you.

rmalchow, 2020-04-02:
so, i'm down to 34 now. and yes, it is working smoothly again. thank you! i think i read some FAQ entry about the number of devices and which firmware to use, and misunderstood it to mean way more than is actually possible.

closed 2020-04-02 by rmalchow

rmalchow, 2020-04-08:
i got a CC26X2R1 now, revision is E (2.1) ... i am having a lot of issues flashing it. it constantly reports errors, mostly one "error -261" .. the only reference to this that i can find is about a hardware issue on a different TI board. i did manage to flash it twice without any error, but it still doesnt work, on both ports, i get this:
```
zigbee2mqtt:debug 2020-04-08 16:34:46: Using zigbee-herdsman with settings: '{"network":{"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11],"networkKey":"HIDDEN"},"databasePath":"/app/data/database.db","databaseBackupPath":"/app/data/database.db.backup","backupPath":"/app/data/coordinator_backup.json","serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/ttyACM1"}}'
zigbee2mqtt:error 2020-04-08 16:34:53: Error while starting zigbee-herdsman
zigbee2mqtt:error 2020-04-08 16:34:53: Failed to start zigbee
zigbee2mqtt:error 2020-04-08 16:34:53: Exiting...
zigbee2mqtt:error 2020-04-08 16:34:53: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:72:23)
    at Generator.throw (<anonymous>)
    at rejected (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:6:65)
```

any ideas?

Koenkk, 2020-04-09:
did you execute step 5 of https://www.zigbee2mqtt.io/information/flashing_via_uniflash.html ?

rmalchow, 2020-04-09:
yes, i did - although, i had to remove the plug and plug it back in after step 4. but other than that, yes.

rmalchow, 2020-04-09:
using this here:

https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/Z-Stack_3.x.0/bin/CC26X2R1_20191106.zip 

unzipping, and flashing with "binary" unchecked.

rmalchow, 2020-04-09:
`[4/9/2020, 7:52:10 PM] [INFO] Cortex_M3_0: GEL Output: Memory Map Initialization Complete.
[4/9/2020, 7:52:12 PM] [INFO] Cortex_M4_0: GEL Output: Memory Map Initialization Complete.
[4/9/2020, 7:52:14 PM] [INFO] Cortex_M4_0: MassErase(): Initializing.
[4/9/2020, 7:52:14 PM] [INFO] Cortex_M4_0: MassErase(): Issuing Board Reset.
[4/9/2020, 7:52:16 PM] [INFO] Cortex_M4_0: MassErase(): Mass erase complete.
[4/9/2020, 7:52:23 PM] [INFO] Cortex_M3_0: GEL Output: Memory Map Initialization Complete.
[4/9/2020, 7:52:25 PM] [INFO] Cortex_M4_0: GEL Output: Memory Map Initialization Complete.
[4/9/2020, 7:52:35 PM] [INFO] Cortex_M4_0: Blank Check: Device flash is blank
[4/9/2020, 7:52:48 PM] [INFO] Cortex_M3_0: GEL Output: Memory Map Initialization Complete.
[4/9/2020, 7:52:50 PM] [INFO] Cortex_M4_0: GEL Output: Memory Map Initialization Complete.
[4/9/2020, 7:52:56 PM] [ERROR] IcePick_C: Error: (Error -261 @ 0xFFFFFEFB) Invalid response was received from the XDS110. (Emulation package 8.4.0.00006)
[4/9/2020, 7:53:12 PM] [ERROR] IcePick_C: Error: (Error -261 @ 0xFFFFFEFB) Invalid response was received from the XDS110. (Emulation package 8.4.0.00006)
[4/9/2020, 7:53:14 PM] [INFO] Cortex_M4_0: GEL Output: Board Reset Complete.
[4/9/2020, 7:53:24 PM] [ERROR] IcePick_C: Error: (Error -261 @ 0xFFFFFEFB) Invalid response was received from the XDS110. (Emulation package 8.4.0.00006)
[4/9/2020, 7:53:30 PM] [ERROR] Cortex_M4_0: Error: (Error -261 @ 0x0) Invalid response was received from the XDS110. (Emulation package 8.4.0.00006)
[4/9/2020, 7:53:32 PM] [INFO] Cortex_M4_0: Target timed out! (Block 1)
[4/9/2020, 7:53:32 PM] [ERROR] Cortex_M4_0: Status 0xFFFFFFFF: Target flashloader has not returned any status. Unknown error!
[4/9/2020, 7:53:32 PM] [INFO] Cortex_M4_0: Command=18 -- addr=0x00024800 -- length=0x00001800
[4/9/2020, 7:53:32 PM] [ERROR] Cortex_M4_0: File Loader: Memory write failed: Timed out waiting for target flashloader to execute command`

Koenkk, 2020-04-10:
It looks that something is seriously wrong with your CC2652R, I would recommend asking TI for a replacement.

rmalchow, 2020-04-10:
yep. that was my impression too. thank you!