Koenkk/zigbee2mqtt #3272: breaking changes regarding MQTT discovery in HA 0.108

netweaver1970, 2020-04-02:
Hi,
as per the HA beta release notes, [here](https://rc.home-assistant.io/blog/2020/03/27/release-108/)
there are some breaking changes in the MQTT Discovery component. Before I make the jump to HA 0.108B0, can someone (@Koenkk) confirm if this is no issue for us?

Thanks,
Geert

MQTT - Deprecated features from MQTT platforms are now removed. - (@emontnemery - #32909) (mqtt docs)

- For MQTT lights setup through MQTT discovery, it’s no longer possible to specify schema with platform configuration variable. Instead, schema must be specified by the schema configuration variable.
- For MQTT alarm_control_panel, MQTT binary_sensor and MQTT sensor set up through MQTT discovery: The undocumented implicit setting of state_topic is no longer supported. Instead, state_topic must be explicitly set.
- Support for json_attributes is removed from MQTT sensor, json_attributes_topic should be used instead.

Koenkk, 2020-04-02:
I've checked all 3 points, but none of these are used by zigbee2mqtt (1.12.0) so should be fine.

netweaver1970, 2020-04-02:
thanks for confirming. And thanks for the Siterwell TRV valve support assistance... If I get back to Belgium I'll have 5 working TRV's hopefully ! :)

closed 2020-04-02 by netweaver1970