Koenkk/zigbee2mqtt #3276: Pairing Osram Smart+ lamp: Interview fails

tinuthir, 2020-04-03:
# Bug Report
## What happened
I am trying to pair an Osram Smart+ lamp, which is Zigbee enabled, with Homeassistant via zigbee2mqtt. I put both the lamp and Homeassistant into pairing mode, and the two devices do communicate briefly, but then the interview fails.
## What did you expect to happen
The lamp is paired to Homeassistant.
## How to reproduce it (minimal and precise)
1. Put Homeassistant/zigbee2mqtt into pairing mode
2. Put the lamp into pairing mode
## Debug Info
zigbee2mqtt version: 1.12.0
CC253X firmware version: 20190608

Herdsman logs can be found here: https://pastebin.com/N5zeBnt9

Koenkk, 2020-04-03:
Did you check: https://www.zigbee2mqtt.io/information/FAQ.html#interview-fails ?

tinuthir, 2020-04-03:
Sure did. Point 1 didn't help (tried a total of 6 times before activating herdsman logging) and point 3 doesn't apply. That left point 2, which brought me here ☺

Koenkk, 2020-04-03:
It's not listed there, but did you also try to replug the stick and pair again?

tinuthir, 2020-04-04:
All right. That didn't work either. See the new log here: https://pastebin.com/6qt6yjrt

To me, the "node descriptor request failed" is the most likely culprit. But this is my first foray into all of this, so I don't know what it means.

Koenkk, 2020-04-04:
Can you try powering off all routers and pairing it close to the coordinator?

tinuthir, 2020-04-04:
Tried it, got a new log: https://pastebin.com/n1Jqmu9s - didn't work, unfortunately.

What I find interesting here is that the herdsman logger is two hours off from the Raspi - it changes after switching mqtt into pairing mode.

Koenkk, 2020-04-04:
Can you try replacing the contents of `/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adpater/zStackAdapter.js` with https://gist.github.com/Koenkk/f74e42ca8f4caf698abd5924ceb3316d and provide the log again?

tinuthir, 2020-04-05:
Turns out, the answer might be no (unless you can give me some pointers). I can't read or write to that file because Home Assistant doesn't have access through the File Editor addon. I also can't see those files if I pop the microSD card into a card reader.

I didn't point this out initially because I wasn't aware it might be important, but I use zigbee2mqtt "regular" rather than the dev channel. Any chance you could replace the content in the dev channel and I then try to use that? Might be the easiest way forward, but of course I don't want to screw over everyone on the dev channel if this goes wrong.

tinuthir, 2020-04-05:
All right, with some help from my more experienced friend (@lukastillmann created a fork and replaced the file), it worked immediately.

Logged the experience here: https://pastebin.com/0hxeGMpX

Thanks for being on top of things!

Koenkk, 2020-04-05:
Thanks, fixed is implemented in the latest dev branch!

closed 2020-04-05 by Koenkk