Koenkk/zigbee2mqtt #3279: RTCGQ11LM issues a ssIasZone that does not have a converter

sjorge, 2020-04-03:
# Bug Report
## What happened
My RTCGQ11LM dropped 
```
zigbee2mqtt:debug 2020-04-03 17:28:48: Received Zigbee message from 'bedroom/motion', type 'readResponse', cluster 'ssIasZone', data '{}' from endpoint 1 wit
h groupID 0
zigbee2mqtt:debug 2020-04-03 17:28:48: No converter available for 'RTCGQ11LM' with cluster 'ssIasZone' and type 'readResponse' and data '{}'
zigbee2mqtt:info  2020-04-03 17:28:48: Successfully interviewed 'bedroom/motion', device has successfully been paired
```

## What did you expect to happen
I've never seen these before, I wonder if this is something we haven't implemented in the past?
As there were a lot of changes to IasZones lately.

Probably Ok to close it as normal behavior but mayb we can add the ignore convert like we have for some others?

## How to reproduce it (minimal and precise)
Just pair a RTCGQ11LM with current dev.

## Debug Info
zigbee2mqtt version:
CC253X firmware version:

Koenkk, 2020-04-03:
If I understand correctly, your RTCGQ11LM drops from the network after this has been send?

sjorge, 2020-04-03:
Not immediately... it used to be very stable until it would randomly drop form the network, (not unheard of for xiami devices though( I started seeing those ssIasZone messages on repair... I never noticed them in the past when the device was not dropping from the network.

But now I get them once after a repair and I think the max the device has stayed in the network has been ~7 days. I need to repair it weekly (also why I noticed the repair stuff brakes after a lot of uptime... anyway it was working fine today though)

Koenkk, 2020-04-03:
Strange, what is the LQI of the sensor and what is the parent?

sjorge, 2020-04-03:
Probably the ikea repeater, let me pull a network map...

closed 2020-06-09 by stale[bot]