Koenkk/zigbee2mqtt #3284: Stelpro stm402AD

TePazz, 2020-04-03:
Hello,
I can't get a Stelpro STM402 thermosstat to work.
My hub detects it my tells me that it is not recognized.
In my opinion the problem is due to the fact that it is a new model the smt402AD. On the other hand I believe that it is exactly the same. I think I should only add it.

I attach my log here and more information.
I'm on hassio raspberryPi: HassOS 3.12
Home Assistant 0.107.7
Mosquitto broker 5.1
zigbee2mqtt 1.12.0

thank you.

log zigbee2mqtt
info  2020-04-03 15:34:33: Logging to console and directory: '/share/zigbee2mqtt/log/2020-04-03.15-34-33' filename: log.txt
debug 2020-04-03 15:34:33: Removing old log directory '/share/zigbee2mqtt/log/2020-03-30.01-44-14'
debug 2020-04-03 15:34:34: Loaded state from file /share/zigbee2mqtt/state.json
info  2020-04-03 15:34:34: Starting zigbee2mqtt version 1.12.0 (commit #unknown)
info  2020-04-03 15:34:34: Starting zigbee-herdsman...
debug 2020-04-03 15:34:34: Using zigbee-herdsman with settings: '{"network":{"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11],"networkKey":"HIDDEN"},"databasePath":"/share/zigbee2mqtt/database.db","databaseBackupPath":"/share/zigbee2mqtt/database.db.backup","backupPath":"/share/zigbee2mqtt/coordinator_backup.json","serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/ttyACM0"}}'
info  2020-04-03 15:34:36: zigbee-herdsman started
info  2020-04-03 15:34:36: Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}}'
debug 2020-04-03 15:34:36: Zigbee network parameters: {"panID":6754,"extendedPanID":"0xdddddddddddddddd","channel":11}
info  2020-04-03 15:34:36: Currently 4 devices are joined:
info  2020-04-03 15:34:36: 0x00158d00042d8b8c (0x00158d00042d8b8c): SJCGQ11LM - Xiaomi Aqara water leak sensor (EndDevice)
info  2020-04-03 15:34:36: 0xf8f005ffffd0d701 (0xf8f005ffffd0d701): ST218 - Stelpro Built-in electronic thermostat (Router)
info  2020-04-03 15:34:36: 0x00158d0003fa7166 (0x00158d0003fa7166): WSDCGQ01LM - Xiaomi MiJia temperature & humidity sensor (EndDevice)
info  2020-04-03 15:34:36: 0x00158d0003f1cb12 (0x00158d0003f1cb12): RTCGQ11LM - Xiaomi Aqara human body movement and illuminance sensor (EndDevice)
warn  2020-04-03 15:34:36: `permit_join` set to  `true` in configuration.yaml.
warn  2020-04-03 15:34:36: Allowing new devices to join.
warn  2020-04-03 15:34:36: Set `permit_join` to `false` once you joined all devices.
info  2020-04-03 15:34:36: Zigbee: allowing new devices to join.
info  2020-04-03 15:34:36: Connecting to MQTT server at http://192.168.31.10:1883
info  2020-04-03 15:34:36: Connected to MQTT server
info  2020-04-03 15:34:36: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
info  2020-04-03 15:34:36: MQTT publish: topic 'zigbee2mqtt/0x00158d00042d8b8c', payload '{"battery":100,"voltage":3045,"linkquality":107,"water_leak":false}'
info  2020-04-03 15:34:36: MQTT publish: topic 'zigbee2mqtt/0xf8f005ffffd0d701', payload '{"local_temperature":20.5,"linkquality":13}'
info  2020-04-03 15:34:36: MQTT publish: topic 'zigbee2mqtt/0x00158d0003fa7166', payload '{"battery":86,"voltage":2975,"temperature":20.63,"humidity":48.64,"linkquality":136}'
info  2020-04-03 15:34:36: MQTT publish: topic 'zigbee2mqtt/0x00158d0003f1cb12', payload '{"battery":100,"voltage":3025,"illuminance":0,"illuminance_lux":0,"linkquality":44,"occupancy":false}'
info  2020-04-03 15:34:36: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.12.0","commit":"unknown","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}},"log_level":"debug","permit_join":true}'
info  2020-04-03 15:34:36: MQTT publish: topic 'homeassistant/binary_sensor/0x00158d00042d8b8c/water_leak/config', payload '{"payload_on":true,"payload_off":false,"value_template":"{{ value_json.water_leak }}","device_class":"moisture","state_topic":"zigbee2mqtt/0x00158d00042d8b8c","json_attributes_topic":"zigbee2mqtt/0x00158d00042d8b8c","name":"0x00158d00042d8b8c_water_leak","unique_id":"0x00158d00042d8b8c_water_leak_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x00158d00042d8b8c"],"name":"0x00158d00042d8b8c","sw_version":"Zigbee2mqtt 1.12.0","model":"Aqara water leak sensor (SJCGQ11LM)","manufacturer":"Xiaomi"},"availability_topic":"zigbee2mqtt/bridge/state"}'
info  2020-04-03 15:34:36: MQTT publish: topic 'homeassistant/sensor/0x00158d00042d8b8c/battery/config', payload '{"unit_of_measurement":"%","device_class":"battery","value_template":"{{ value_json.battery }}","state_topic":"zigbee2mqtt/0x00158d00042d8b8c","json_attributes_topic":"zigbee2mqtt/0x00158d00042d8b8c","name":"0x00158d00042d8b8c_battery","unique_id":"0x00158d00042d8b8c_battery_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x00158d00042d8b8c"],"name":"0x00158d00042d8b8c","sw_version":"Zigbee2mqtt 1.12.0","model":"Aqara water leak sensor (SJCGQ11LM)","manufacturer":"Xiaomi"},"availability_topic":"zigbee2mqtt/bridge/state"}'
info  2020-04-03 15:34:36: MQTT publish: topic 'homeassistant/sensor/0x00158d00042d8b8c/linkquality/config', payload '{"icon":"mdi:signal","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}","state_topic":"zigbee2mqtt/0x00158d00042d8b8c","json_attributes_topic":"zigbee2mqtt/0x00158d00042d8b8c","name":"0x00158d00042d8b8c_linkquality","unique_id":"0x00158d00042d8b8c_linkquality_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x00158d00042d8b8c"],"name":"0x00158d00042d8b8c","sw_version":"Zigbee2mqtt 1.12.0","model":"Aqara water leak sensor (SJCGQ11LM)","manufacturer":"Xiaomi"},"availability_topic":"zigbee2mqtt/bridge/state"}'
info  2020-04-03 15:34:36: MQTT publish: topic 'homeassistant/sensor/0xf8f005ffffd0d701/linkquality/config', payload '{"icon":"mdi:signal","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}","state_topic":"zigbee2mqtt/0xf8f005ffffd0d701","json_attributes_topic":"zigbee2mqtt/0xf8f005ffffd0d701","name":"0xf8f005ffffd0d701_linkquality","unique_id":"0xf8f005ffffd0d701_linkquality_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0xf8f005ffffd0d701"],"name":"0xf8f005ffffd0d701","sw_version":"Zigbee2mqtt 1.12.0","model":"Built-in electronic thermostat (ST218)","manufacturer":"Stelpro"},"availability_topic":"zigbee2mqtt/bridge/state"}'
info  2020-04-03 15:34:36: MQTT publish: topic 'homeassistant/sensor/0x00158d0003fa7166/temperature/config', payload '{"unit_of_measurement":"°C","device_class":"temperature","value_template":"{{ value_json.temperature }}","state_topic":"zigbee2mqtt/0x00158d0003fa7166","json_attributes_topic":"zigbee2mqtt/0x00158d0003fa7166","name":"0x00158d0003fa7166_temperature","unique_id":"0x00158d0003fa7166_temperature_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x00158d0003fa7166"],"name":"0x00158d0003fa7166","sw_version":"Zigbee2mqtt 1.12.0","model":"MiJia temperature & humidity sensor (WSDCGQ01LM)","manufacturer":"Xiaomi"},"availability_topic":"zigbee2mqtt/bridge/state"}'
info  2020-04-03 15:34:36: MQTT publish: topic 'homeassistant/sensor/0x00158d0003fa7166/humidity/config', payload '{"unit_of_measurement":"%","device_class":"humidity","value_template":"{{ value_json.humidity }}","state_topic":"zigbee2mqtt/0x00158d0003fa7166","json_attributes_topic":"zigbee2mqtt/0x00158d0003fa7166","name":"0x00158d0003fa7166_humidity","unique_id":"0x00158d0003fa7166_humidity_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x00158d0003fa7166"],"name":"0x00158d0003fa7166","sw_version":"Zigbee2mqtt 1.12.0","model":"MiJia temperature & humidity sensor (WSDCGQ01LM)","manufacturer":"Xiaomi"},"availability_topic":"zigbee2mqtt/bridge/state"}'
info  2020-04-03 15:34:36: MQTT publish: topic 'homeassistant/sensor/0x00158d0003fa7166/battery/config', payload '{"unit_of_measurement":"%","device_class":"battery","value_template":"{{ value_json.battery }}","state_topic":"zigbee2mqtt/0x00158d0003fa7166","json_attributes_topic":"zigbee2mqtt/0x00158d0003fa7166","name":"0x00158d0003fa7166_battery","unique_id":"0x00158d0003fa7166_battery_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x00158d0003fa7166"],"name":"0x00158d0003fa7166","sw_version":"Zigbee2mqtt 1.12.0","model":"MiJia temperature & humidity sensor (WSDCGQ01LM)","manufacturer":"Xiaomi"},"availability_topic":"zigbee2mqtt/bridge/state"}'
[... cut ...] [0,3,513,516,4,1029],"outClusterList":[3,10,1026],"clusters":{"genBasic":{"attributes":{"modelId":"SMT402AD\u0000\u0000\u0000\u0000\u0000�\u0012\u0000 A*\u0000\b\u0003\u0000\u0000\u0000AT\u0000\b��","powerSource":1,"zclVersion":2,"appVersion":100,"stackVersion":50,"hwVersion":1,"dateCode":"20000001 45575\u0000"}}},"binds":[]}},"appVersion":100,"stackVersion":50,"hwVersion":1,"dateCode":"20000001 45575\u0000","zclVersion":2,"interviewCompleted":true,"meta":{},"lastSeen":1585942529644}

thank you.

Koenkk, 2020-04-04:
This device is currently unsupported, please follow: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

TePazz, 2020-04-12:
hi, i made a pull requests for devices.js and homeassistant.js to add the 2 thermostat. I tested them and it works very well with this code. A big thank you to @carldebilly for the precious help on the files to be improved and much more.

Koenkk, 2020-04-12:
merged the prs, thanks!

closed 2020-04-12 by Koenkk