Koenkk/zigbee2mqtt #3285: Ikea Symfonisk does not work anymore since newest zigbee2mqtt version

Baggerfahrer, 2020-04-04:
Hey Guys,

since my last update to the actual zigbee2mqtt version, the Ikea Symfonisk Sound Button [https://www.zigbee2mqtt.io/devices/E1744.html](url) does not work anymore.
The only reason i can image that my coordinator is on an old version, because i do not own the usb flashing device.
can anyone confirm this issue?

# Bug Report
## What happened
Since update of the zigbee2mqtt version to the actual version 1.12.0 the Ikea Symfonsik Button do not work anymore. Log see below.
Before that, the Button worked like a charm.
I tried to pair the button several times, as well very close and as well 2 meters away from the coordinator.

## What did you expect to happen
Should work as expected and as it did before the update.

## How to reproduce it (minimal and precise)
It occurs every time.

## Debug Info
zigbee2mqtt version: zigbee2mqtt version 1.12.0 
CC253X firmware version: Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190315}}'

See the following log of debugging of zigbee2mqtt:

`Apr 04 10:36:32 openhab npm[13247]: zigbee2mqtt:info  2020-04-04 10:36:32: Starting interview of '0x680ae2fffe372522'
Apr 04 10:36:32 openhab npm[13247]: zigbee2mqtt:info  2020-04-04 10:36:32: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x680ae2fffe372522"}}'
Apr 04 10:37:16 openhab npm[13247]: zigbee2mqtt:debug 2020-04-04 10:37:16: Device '0x680ae2fffe372522' announced itself
Apr 04 10:37:16 openhab npm[13247]: zigbee2mqtt:info  2020-04-04 10:37:16: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"0x680ae2fffe372522"}}'
Apr 04 10:37:32 openhab npm[13247]: zigbee2mqtt:error 2020-04-04 10:37:32: Failed to interview '0x680ae2fffe372522', device has not successfully been paired
Apr 04 10:37:32 openhab npm[13247]: zigbee2mqtt:info  2020-04-04 10:37:32: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_failed","meta":{"friendly_name":"0x680ae2fffe372522"}}'`

johras, 2020-04-04:
Doesn't directly answer your problem but CC-debugger is not needed to flash the stick.
There are many alternative methods: https://www.zigbee2mqtt.io/information/alternative_flashing_methods.html

I did it using only Raspberry Pi's GPIO.

Baggerfahrer, 2020-04-05:
honestly, it was the best tip you could give me! :)
I flashed it accordingly without the debugger et voila, the symfonisk button works again.

great thanks!

closed 2020-04-05 by Baggerfahrer