Koenkk/zigbee2mqtt #3287: ZigBee sniffing with ZBOSS

alflinusjonsson, 2020-04-04:
Hi,
Successfully installed zigbee2mqtt (cc2531) and running a mosquitto server on a rpi. Now i want to sniff the traffic, followed the guide provided here in the repository but cant download ZBOSS. Created an account but haven't received any confirmation mail from them allowing me to log in. Anyone else with similar problem? Or is there any other way to sniff the traffic?

Thanks in advance!

Kryzek, 2020-04-04:
Check junkmail also of using Gmail.

alflinusjonsson, 2020-04-04:
Checked junkmail, nothing there :(

pixeldoc2000, 2020-04-04:
There is also https://www.ti.com/tool/PACKET-SNIFFER from TI. I've use it with good success.

https://sunmaysky.blogspot.com/2018/10/how-to-use-cc2531emk-and-wireshark-as.html

Use something like http://bugmenot.com/view/zboss.dsr-wireless.com to download ;-)

alflinusjonsson, 2020-04-04:
The login details from your link worked, thanks a lot! :)

alflinusjonsson, 2020-04-05:
Successfully installed ZBOSS and made all settings showed here:  [https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html](url)

But getting connection errors while running my zigbee2mqtt at the same time. Is it not possible to run zigbee2mqtt **and** sniff traffic with ZBOSS at the same time using cc2531?

If I close my zigbee2mqtt server, I´m allowed to sniff traffic. But it´s empty since my server is not running.

[image]
[image]

dinhchinh82, 2020-04-06:
@alflinusjonsson Have you set the zigbee channel & USB port correctly yet?

alflinusjonsson, 2020-04-06:
Yes, ZigBee channel is set to (0x0B (11)) and USB port is COM3 (CC2531)

alflinusjonsson, 2020-04-06:
But as I understand now, its different CC2531 firmwares for zigbee2mqtt and sniffing and therefore its not possible to sniff and run zigbee2mqtt on the **same** dongle at the **same** time.

brendanmullan, 2020-04-18:
> 
> 
> But as I understand now, its different CC2531 firmwares for zigbee2mqtt and sniffing and therefore its not possible to sniff and run zigbee2mqtt on the **same** dongle at the **same** time.

Correct, you need the ZBOSS firmware installed to sniff traffic. Would suggest closing this issue to help keep the list clear :-)

closed 2020-04-18 by alflinusjonsson