Koenkk/zigbee2mqtt #3288: Publish 'set' 'state' to device failed after coordinator is resetted

zrainx, 2020-04-04:
# Bug Report

## What happened
Publish 'set' 'state' to device failed after coordinator is resetted

## What did you expect to happen
success to publish 'set' 'state' message

## How to reproduce it (minimal and precise)
1. reset the coordinator(CC2530)  with MQTT message on 'zigbee2mqtt/bridge/config/reset' with data 'Coordinator'

2. set device(Xiaomi Aqara smart LED bulb) state with  MQTT message on 'zigbee2mqtt/0x00158d000398e6b7/set' with data '{"state":"OFF"}'

## Debug Info
zigbee2mqtt version: 1.12.0 (commit #840b9d9)

CC253X firmware version: 20190608

The hardware is CC2530 + CC2592 and CH340 (usb to serial)

https://pastebin.com/sArJ3NA2

Koenkk, 2020-04-04:
Does replugging the stick fix it?

zrainx, 2020-04-05:
replugging  the stick can‘t fix it.
killing zigbee2mqtt  and running it again can fix it.

Koenkk, 2020-04-05:
But when replugging the stick, you also restart z2m right?

zrainx, 2020-04-06:
Hi Koenkk,
I try it again. You are right. replugging stick can fix it. 
The z2m exits when I remove the stick. I have to restart z2m after I replug the stick.
It seems that restarting z2m can fix the problem. 
But I don't know why, and how to fix it without restarting z2m.

bochelork, 2020-04-06:
Here I see the same. 
`zigbee2mqtt:info  2020-04-06 23:12:47: Soft resetted ZNP
zigbee2mqtt:debug 2020-04-06 23:12:59: Saving state to file /volumeUSB1/usbshare/zigbee2mqtt3/data/state.json
zigbee2mqtt:debug 2020-04-06 23:13:00: Received MQTT message on 'zigbee2mqtt/E14/set' with data '{"state": "ON"}'
zigbee2mqtt:debug 2020-04-06 23:13:00: Publishing 'set' 'state' to 'E14'
zigbee2mqtt:error 2020-04-06 23:13:00: Publish 'set' 'state' to 'E14' failed: 'Error: Command 0x00124b001ed00787/11 genLevelCtrl.moveToLevelWithOnOff({"level":250,"transtime":0}, {"timeout":6000,"manufacturerCode":null,"disableDefaultResponse":false}) failed (Error: SREQ '--> AF - dataRequest - {"dstaddr":24051,"destendpoint":11,"srcendpoint":1,"clusterid":8,"transid":11,"options":0,"radius":30,"len":6,"data":{"type":"Buffer","data":[1,10,4,250,0,0]}}' failed with status '2' (expected '0'))'
zigbee2mqtt:debug 2020-04-06 23:13:00: Error: Command 0x00124b001ed00787/11 genLevelCtrl.moveToLevelWithOnOff({"level":250,"transtime":0}, {"timeout":6000,"manufacturerCode":null,"disableDefaultResponse":false}) failed (Error: SREQ '--> AF - dataRequest - {"dstaddr":24051,"destendpoint":11,"srcendpoint":1,"clusterid":8,"transid":11,"options":0,"radius":30,"len":6,"data":{"type":"Buffer","data":[1,10,4,250,0,0]}}' failed with status '2' (expected '0'))
    at Endpoint.<anonymous> (/volumeUSB1/usbshare/zigbee2mqtt3/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:349:23)
    at Generator.throw (<anonymous>)
    at rejected (/volumeUSB1/usbshare/zigbee2mqtt3/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65)
    at <anonymous>
zigbee2mqtt:info  2020-04-06 23:13:00: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to 'E14' failed: 'Error: Command 0x00124b001ed00787/11 genLevelCtrl.moveToLevelWithOnOff({\"level\":250,\"transtime\":0}, {\"timeout\":6000,\"manufacturerCode\":null,\"disableDefaultResponse\":false}) failed (Error: SREQ '--> AF - dataRequest - {\"dstaddr\":24051,\"destendpoint\":11,\"srcendpoint\":1,\"clusterid\":8,\"transid\":11,\"options\":0,\"radius\":30,\"len\":6,\"data\":{\"type\":\"Buffer\",\"data\":[1,10,4,250,0,0]}}' failed with status '2' (expected '0'))'","meta":{"friendly_name":"E14"}}'`

Just restarting zigbee2mqtt fixes the problem.
Can't find details about the reset in documentation. When should we use it?

Koenkk, 2020-04-08:
It's an old command, not sure if its really useful anymore. @zrainx why did you use it?

zrainx, 2020-04-09:
Nothing special, I just tested MQTT topics one by one  according to https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html.

zigbee2mqtt/bridge/config/reset  is in this web page.

closed 2020-06-17 by stale[bot]