Koenkk/zigbee2mqtt #3291: Missing package while following howto for adding unsupported device

Boldie, 2020-04-04:
I just followed the how to to add a new device for using inside a docker containner. I also checked out the zigbee-herdsman-converters stuff in version 12.0.51 like it is in package.json but no luck, a package named axios is missing.

```
Using '/app/data' as data directory

> zigbee2mqtt@1.12.0 start /app
> node index.js

internal/modules/cjs/loader.js:985
  throw err;
  ^

Error: Cannot find module 'axios'
Require stack:
- /app/node_modules/zigbee-herdsman-converters/ota/ledvance.js
- /app/node_modules/zigbee-herdsman-converters/ota/index.js
- /app/node_modules/zigbee-herdsman-converters/devices.js
- /app/node_modules/zigbee-herdsman-converters/index.js
- /app/lib/util/utils.js
- /app/lib/util/settings.js
- /app/lib/util/logger.js
- /app/lib/mqtt.js
- /app/lib/controller.js
- /app/index.js
    at Function.Module._resolveFilename (internal/modules/cjs/loader.js:982:15)
    at Function.Module._load (internal/modules/cjs/loader.js:864:27)
    at Module.require (internal/modules/cjs/loader.js:1044:19)
    at require (internal/modules/cjs/helpers.js:77:18)
    at Object.<anonymous> (/app/node_modules/zigbee-herdsman-converters/ota/ledvance.js:1:15)
    at Module._compile (internal/modules/cjs/loader.js:1158:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1178:10)
    at Module.load (internal/modules/cjs/loader.js:1002:32)
    at Function.Module._load (internal/modules/cjs/loader.js:901:14)
    at Module.require (internal/modules/cjs/loader.js:1044:19) {
  code: 'MODULE_NOT_FOUND',
  requireStack: [
    '/app/node_modules/zigbee-herdsman-converters/ota/ledvance.js',
    '/app/node_modules/zigbee-herdsman-converters/ota/index.js',
    '/app/node_modules/zigbee-herdsman-converters/devices.js',
    '/app/node_modules/zigbee-herdsman-converters/index.js',
    '/app/lib/util/utils.js',
    '/app/lib/util/settings.js',
    '/app/lib/util/logger.js',
    '/app/lib/mqtt.js',
    '/app/lib/controller.js',
    '/app/index.js'
  ]
}
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.12.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.12.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2020-04-04T22_59_28_510Z-debug.log
```

Boldie, 2020-04-04:
The problem is easy, the node_modules path is also been overwritten with the HowTo stuff. Now I just added the devices.js as single file to the container and the stuff works. -> So the HowTo shall be updated accordingly.

Koenkk, 2020-04-05:
@Boldie what did override it?

Boldie, 2020-04-05:
I was using this hoto: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
And there doing this for docker is not a great idea: -v /opt/zigbee-herdsman-converters:/app/node_modules/zigbee-herdsman-converters -v /opt/homeassistant.js:/app/lib/extension/homeassistant.js
because it will hide the whole /app/node_modules/zigbee-herdsman-converters/node_modules path which leads to the error shown above. Just adding a single file is more promising and my current solution.

Koenkk, 2020-04-06:
After the clone a `npm ci` has to be done to install the dependencies. I've updated the instructions accordingly. thanks!

closed 2020-04-06 by Koenkk