Koenkk/zigbee2mqtt #3293: No Support for essentials ZigBee Thermostat by tuya

sander363, 2020-04-05:
need help with essential zigbee Thermostat.
I don't know how I add new unsupported devices. 

Hersteller: _TYST11_jeaxp72v
Modell: eaxp72v

Westcott1, 2020-04-06:
Do you have a URL for where you bought this device?

sander363, 2020-04-08:
https://www.wesmartify.de/heizung/heizkoerperradiator/281/essentials-smart-home-heizkoerperthermostat-premium/

maddinthe, 2020-04-18:
It seems to be a whitelabel for the Siterwell GS361A-H04

sander363, 2020-04-19:
okay...but the log says that is an unsupported device...how i can get an support for this thermostat?

maddinthe, 2020-04-19:
I have entered the information into the devices.js myself.
Looks like this at the siterwell valve entry now:
```
// Siterwell
    {
        zigbeeModel: ['ivfvd7h', 'eaxp72v\u0000'],
        model: 'GS361A-H04',
        vendor: 'Siterwell',
        description: 'Radiator valve with thermostat',
        supports: 'thermostat, temperature',
        fromZigbee: [
            fz.tuya_thermostat,
            fz.tuya_thermostat_on_set_data,
            fz.ignore_basic_report,
        ],
        toZigbee: [
            tz.tuya_thermostat_child_lock,
            tz.tuya_thermostat_window_detection,
            tz.tuya_thermostat_valve_detection,
            tz.tuya_thermostat_current_heating_setpoint,
            tz.tuya_thermostat_system_mode,
            tz.tuya_thermostat_auto_lock,
            tz.tuya_thermostat_calibration,
            tz.tuya_thermostat_min_temp,
            tz.tuya_thermostat_max_temp,
            tz.tuya_thermostat_boost_time,
            tz.tuya_thermostat_comfort_temp,
            tz.tuya_thermostat_eco_temp,
            tz.tuya_thermostat_force,
        ],
        whiteLabel:[
            {vendor: '_TYST11_jeaxp72v', description: 'Essentials Smart Home Heizkoerperthermostat Premium', model: 'eaxp72v\u0000'},
        ],
    },
```

closed 2020-06-20 by sander363

ajenjo85, 2020-08-04:
I bought this device to add it to OpenHAB environment using zigbee2Mqtt. 
I can make the first part of the process, see the devices in my MQTT server (using MQTT explorer). I can see all the information that the device provides.
But in the second part, I don't know how to proceed. I don't know the command we have to use to change the configuration of the device 

Could you help me? I need the command I have to use.

Gringohekus, 2020-10-25:
Hi I am also think to purchase this TR for my openHAB. Can you help me to come to decision? 
Is this capable to report the opening % of the valve? 
What is the valve detection functionality?

BinGeradeOnline, 2021-12-08:
I have the same issue and it says that it is unsupported.

All though i have the same device i think it has not the same modell number could this be the problem and how can i fix that?
I'm pretty new to all of this.
thanks for any help