Koenkk/zigbee2mqtt #3294: How to support new device tozigbee.js ?

hyzgit, 2020-04-05:
I am adding a new device now, the status back reading is OK, but send command do not know where to start, because in the log, there is no any information related to send command.

 Here is a three-position wall switch, the equipment manufacturer is Xiaomi, the model is: QBKG25LM , zigbeeModel：lumi.switch.l3acn3.
 fromzigbee used fz.aqara_opple_multistate. 

 Switch btn 1 (on) :
zigbee2mqtt:debug 2020-04-05 20:50:26: Received Zigbee message from '0x04cf8cdf3c7a4456', type 'attributeReport', cluster 'genOnOff', data '{"245":56798720,"onOff":1}' from endpoint 1 with groupID 0
,zigbee2mqtt:info  2020-04-05 20:50:26: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c7a4456', payload '{"state":"ON","linkquality":89}'
,zigbee2mqtt:debug 2020-04-05 20:50:27: Received Zigbee message from '0x04cf8cdf3c7a4456', type 'attributeReport', cluster 'genMultistateInput', data '{"presentValue":1}' from endpoint 41 with groupID 0
,zigbee2mqtt:info  2020-04-05 20:50:27: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c7a4456', payload '{"state":"ON","linkquality":92,"action":"button_41_single"}'

 Switch btn 1 (off):
zigbee2mqtt:debug 2020-04-05 20:50:27: Received Zigbee message from '0x04cf8cdf3c7a4456', type 'attributeReport', cluster 'genOnOff', data '{"245":56798720,"onOff":0}' from endpoint 1 with groupID 0
,zigbee2mqtt:info  2020-04-05 20:50:27: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c7a4456', payload '{"state":"OFF","linkquality":92}'
,zigbee2mqtt:debug 2020-04-05 20:50:28: Received Zigbee message from '0x04cf8cdf3c7a4456', type 'attributeReport', cluster 'genMultistateInput', data '{"presentValue":1}' from endpoint 41 with groupID 0
,zigbee2mqtt:info  2020-04-05 20:50:28: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c7a4456', payload '{"state":"OFF","linkquality":97,"action":"button_41_single"}'

btn 1 endpoint  is 1 and 41 , btn 2 endpoint  is 2 and 42 , btn 3 endpoint  is 3 and 43.

How do you know tozigbee endpoint and data format for each key now?

thanks!

Koenkk, 2020-04-05:
Can you provide your `database.db`?

hyzgit, 2020-04-06:
Thank you, I've debugged it, and at the moment this can be used, but still want to know how to debug for tozigbee.js, thank you
 Here is the download link for the file ( datase.db , devices.js , fromzigbee.js )
https://pan.baidu.com/s/1Ejp-QK7bt5EM9LFymAb5qA 
Extract code:9fyv

Koenkk, 2020-04-06:
I'm not sure how to download from that site, it shows me that I need an app or something and my Chinese is not really good. Could you upload it to pastebin.com and link it here?

hyzgit, 2020-04-09:
> I'm not sure how to download from that site, it shows me that I need an app or something and my Chinese is not really good. Could you upload it to pastebin.com and link it here?

https://pastebin.com/J0GdRbAB

hyzgit, 2020-04-09:
Now there's another problem, the light attached to the center button, and when it's on, sometimes turns itself on, off, on, off, off

hyzgit, 2020-04-09:
this is home-assistant log:

PM 2:03:28
ke_ting_deng turned on
PM 2:03:28
ke_ting_deng turned off
PM 2:03:27
ke_ting_deng turned on
PM 2:03:26
ke_ting_deng turned off
PM 2:03:26
ke_ting_deng turned on
PM 2:03:25
ke_ting_deng turned off
PM 2:03:24
ke_ting_deng turned on
PM 2:03:24
ke_ting_deng turned off
PM 2:03:23
ke_ting_deng turned on
PM 2:03:23
ke_ting_deng turned off
PM 2:03:22
ke_ting_deng turned on
PM 2:03:21
ke_ting_deng turned off
PM 2:03:21
ke_ting_deng turned on
PM 2:03:20
ke_ting_deng turned off
PM 2:03:20
ke_ting_deng turned on
PM 2:03:19
ke_ting_deng turned off
PM 2:03:19
ke_ting_deng turned on
PM 2:03:18
ke_ting_deng turned off
PM 2:03:17
ke_ting_deng turned on

hyzgit, 2020-04-09:
this is zigbee2mqtt log :

zigbee2mqtt:info  2020-04-09 14:03:20: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c7a4456', payload '{"state":"ON","linkquality":105,"state_btn3":"ON","battery":100,"voltage":25856,"state_btn1":"OFF","state_btn2":"ON","state_right":"OFF","state_left":"OFF","state_center":"OFF"}'
,zigbee2mqtt:debug 2020-04-09 14:03:20: Received Zigbee message from '0x04cf8cdf3c7a4456', type 'attributeReport', cluster 'genOnOff', data '{"245":117440588,"onOff":0}' from endpoint 2 with groupID 0
,zigbee2mqtt:info  2020-04-09 14:03:20: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c7a4456', payload '{"state":"ON","linkquality":102,"state_btn3":"ON","battery":100,"voltage":25856,"state_btn1":"OFF","state_btn2":"ON","state_right":"OFF","state_left":"OFF","state_center":"OFF"}'
,zigbee2mqtt:info  2020-04-09 14:03:21: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c7a4456', payload '{"state":"ON","linkquality":102,"state_btn3":"ON","battery":100,"voltage":25856,"state_btn1":"OFF","state_btn2":"ON","state_right":"OFF","state_left":"OFF","state_center":"ON"}'
,zigbee2mqtt:debug 2020-04-09 14:03:21: Received Zigbee message from '0x04cf8cdf3c7a4456', type 'attributeReport', cluster 'genOnOff', data '{"245":117440589,"onOff":1}' from endpoint 2 with groupID 0
,zigbee2mqtt:info  2020-04-09 14:03:21: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c7a4456', payload '{"state":"ON","linkquality":105,"state_btn3":"ON","battery":100,"voltage":25856,"state_btn1":"OFF","state_btn2":"ON","state_right":"OFF","state_left":"OFF","state_center":"ON"}'
,zigbee2mqtt:info  2020-04-09 14:03:21: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c7a4456', payload '{"state":"ON","linkquality":105,"state_btn3":"ON","battery":100,"voltage":25856,"state_btn1":"OFF","state_btn2":"ON","state_right":"OFF","state_left":"OFF","state_center":"OFF"}'
,zigbee2mqtt:info  2020-04-09 14:03:21: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c7a4456', payload '{"state":"ON","linkquality":102,"state_btn3":"ON","battery":100,"voltage":25856,"state_btn1":"OFF","state_btn2":"ON","state_right":"OFF","state_left":"OFF","state_center":"OFF"}'
,zigbee2mqtt:debug 2020-04-09 14:03:21: Received Zigbee message from '0x04cf8cdf3c7a4456', type 'attributeReport', cluster 'genOnOff', data '{"245":117440590,"onOff":0}' from endpoint 2 with groupID 0
,zigbee2mqtt:info  2020-04-09 14:03:22: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c7a4456', payload '{"state":"ON","linkquality":102,"state_btn3":"ON","battery":100,"voltage":25856,"state_btn1":"OFF","state_btn2":"ON","state_right":"OFF","state_left":"OFF","state_center":"ON"}'
,zigbee2mqtt:debug 2020-04-09 14:03:22: Received Zigbee message from '0x04cf8cdf3c7a4456', type 'attributeReport', cluster 'genOnOff', data '{"245":117440591,"onOff":1}' from endpoint 2 with groupID 0
,zigbee2mqtt:info  2020-04-09 14:03:22: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c7a4456', payload '{"state":"ON","linkquality":105,"state_btn3":"ON","battery":100,"voltage":25856,"state_btn1":"OFF","state_btn2":"ON","state_right":"OFF","state_left":"OFF","state_center":"ON"}'
,zigbee2mqtt:debug 2020-04-09 14:03:23: Received Zigbee message from '0x04cf8cdf3c7a4456', type 'attributeReport', cluster 'genOnOff', data '{"245":117440592,"onOff":0}' from endpoint 2 with groupID 0
,zigbee2mqtt:info  2020-04-09 14:03:23: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c7a4456', payload '{"state":"ON","linkquality":105,"state_btn3":"ON","battery":100,"voltage":25856,"state_btn1":"OFF","state_btn2":"ON","state_right":"OFF","state_left":"OFF","state_center":"OFF"}'
,zigbee2mqtt:info  2020-04-09 14:03:23: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c7a4456', payload '{"state":"ON","linkquality":102,"state_btn3":"ON","battery":100,"voltage":25856,"state_btn1":"OFF","state_btn2":"ON","state_right":"OFF","state_left":"OFF","state_center":"OFF"}'
,zigbee2mqtt:info  2020-04-09 14:03:23: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c7a4456', payload '{"state":"ON","linkquality":102,"state_btn3":"ON","battery":100,"voltage":25856,"state_btn1":"OFF","state_btn2":"ON","state_right":"OFF","state_left":"OFF","state_center":"ON"}'
,zigbee2mqtt:debug 2020-04-09 14:03:23: Received Zigbee message from '0x04cf8cdf3c7a4456', type 'attributeReport', cluster 'genOnOff', data '{"245":117440593,"onOff":1}' from endpoint 2 with groupID 0
,zigbee2mqtt:info  2020-04-09 14:03:23: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c7a4456', payload '{"state":"ON","linkquality":105,"state_btn3":"ON","battery":100,"voltage":25856,"state_btn1":"OFF","state_btn2":"ON","state_right":"OFF","state_left":"OFF","state_center":"ON"}'
,zigbee2mqtt:info  2020-04-09 14:03:24: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c7a4456', payload '{"state":"ON","linkquality":105,"state_btn3":"ON","battery":100,"voltage":25856,"state_btn1":"OFF","state_btn2":"ON","state_right":"OFF","state_left":"OFF","state_center":"OFF"}'
,zigbee2mqtt:debug 2020-04-09 14:03:24: Received Zigbee message from '0x04cf8cdf3c7a4456', type 'attributeReport', cluster 'genOnOff', data '{"245":117440594,"onOff":0}' from endpoint 2 with groupID 0
,zigbee2mqtt:debug 2020-04-09 14:03:24: Received Zigbee message from '0x04cf8cdf3c7a4456', type 'attributeReport', cluster 'genOnOff', data '{"245":117440595,"onOff":1}' from endpoint 2 with groupID 0
,zigbee2mqtt:info  2020-04-09 14:03:24: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c7a4456', payload '{"state":"ON","linkquality":105,"state_btn3":"ON","battery":100,"voltage":25856,"state_btn1":"OFF","sta [... cut ...]
,zigbee2mqtt:debug 2020-04-09 14:03:28: Received Zigbee message from '0x04cf8cdf3c7a4456', type 'attributeReport', cluster 'genOnOff', data '{"245":117440602,"onOff":1}' from endpoint 2 with groupID 0
,zigbee2mqtt:info  2020-04-09 14:03:29: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c7a4456', payload '{"state":"ON","linkquality":105,"state_btn3":"ON","battery":100,"voltage":25856,"state_btn1":"OFF","state_btn2":"ON","state_right":"OFF","state_left":"OFF","state_center":"OFF"}'
,zigbee2mqtt:debug 2020-04-09 14:03:29: Received Zigbee message from '0x04cf8cdf3c7a4456', type 'attributeReport', cluster 'genOnOff', data '{"245":117440603,"onOff":0}' from endpoint 2 with groupID 0

hyzgit, 2020-04-09:
The "state_btn1" , "state_btn2" , "state_btn3" in the log was written when I started debugging, the back has been removed, replaced by left ,center ,right, but why have these contents been saved in the mqtt payload?

Koenkk, 2020-04-10:
Please try with the following definition:

```js
{
    zigbeeModel: ['lumi.switch.l3acn3'],
    model: 'QBKG25LM',
    vendor: 'Xiaomi',
    description: 'Aqara D1 3 gang smart wall switch',
    supports: 'on/off',
    fromZigbee: [fz.on_off],
    toZigbee: [tz.on_off],
    meta: {multiEndpoint: true},
    endpoint: (device) => {
        return {'left': 1, 'center': 2, 'right': 3};
    },
},
```

You should be able to control it with `zigbee2mqtt/0x04cf8cdf3c7a4456/left/set` payload `{"state": "ON"}`. Instead of `left` you can also use `center` and `right`.

hyzgit, 2020-04-10:
Okay, I'll try, thank you

hyzgit, 2020-04-15:
It's best to add this so you can recognize the manual switch . 
fromZigbee:[fz.on_off , fz.QBKG25LM_click],

QBKG25LM_click: {
        cluster: 'genMultistateInput',
        type: ['attributeReport', 'readResponse'],
        convert: (model, msg, publish, options, meta) => {
            if ([1, 2].includes(msg.data.presentValue)) {             
                const mapping = {41: 'left', 42: 'center', 43: 'right'};
                const times = {1: 'single', 2: 'double', 3: 'triple', 0: 'hold', 255: 'release'};
                const button = mapping[msg.endpoint.ID];
                return {click: `${button}_${times[msg.data.presentValue]}`};
            }
        },            
    },

woshitom, 2020-05-25:
> Please try with the following definition:
>
> ```js
> {
>     zigbeeModel: ['lumi.switch.l3acn3'],
>     model: 'QBKG25LM',
>     vendor: 'Xiaomi',
>     description: 'Aqara D1 3 gang smart wall switch', [...]

Hi, I'm also tryin gto add 'lumi.switch.l3acn3' so I copy/paste your code, but when I'm trying to turn it on I'm getting this message on the log file:

> Failed to call 'EntityPublish' 'onMQTTMessage' (AssertionError [ERR_ASSERTION]: Endpoint name 'left' is given but device is unsupported

Koenkk, 2020-05-25:
As this device is supported in the latest dev branch now, I assume this can be closed.

closed 2020-05-25 by Koenkk