Koenkk/zigbee2mqtt #3296: Feature request. Make `retain: true` the default for some kind of devices

bessarabov, 2020-04-05:
# Bug Report

Well, this is not actually a bug, but the current behaviour, that I think should be changed.

## What happened

I'm using z2m with Home Assistant. I have several Aqara temp/humidity sensors and Aqara door sensor.
The problem is that when I restart Home Assistant I need to wait some time for the temp/humidity to appear in the interface. (And in the case of door sensor it is always "closed" after the restart)

The solution is simple. Just edit `/usr/share/hassio/share/zigbee2mqtt/devices.yaml` and add `retain: true` to the device description:

````
'0x00158d000422fde9':
  friendly_name: '0x00158d000422fde9'
  retain: true
```

## What did you expect to happen

I'm not an expert in zigbee2mqtt nor in Home Assistant. My expectation was that zigbee2mqtt default setting is to send messages with retain flag, to prevent confusing behaviour.

I can't say for any type of devices, but for temp/humidity sensors and for door sensors I can't imagine situation when somebody needs to disable sending retain flag.

So, I would like to propose a feature — make the default `retain: true` for some kind of devices to simplify initial configuration for the users.

## How to reproduce it (minimal and precise)

## Debug Info
zigbee2mqtt version: 1.9.0
CC253X firmware version: not sure

Botched1, 2020-04-06:
I actually think there should be a universal "retain" flag that applies to all devices. 100% of my devices have retain set, and I would guess that is true of the vast majority of z2m users.

presslab-us, 2020-04-06:
I think that's the point of the birth/will message as shown here, to fix this problem:
https://www.zigbee2mqtt.io/integration/home_assistant.html

Do you have Home Assistant configured to use those?

Koenkk, 2020-04-06:
Setting retain true for all devices can easily be done with (https://www.zigbee2mqtt.io/information/configuration.html#changing-device-type-specific-defaults)

```yaml
device_options:
  retain: true
```

Botched1, 2020-04-06:
> Setting retain true for all devices can easily be done with (https://www.zigbee2mqtt.io/information/configuration.html#changing-device-type-specific-defaults)
> 
> ```yaml
> device_options:
>   retain: true
> ```

My apologies for missing that. I read that in the documentation, but still did not understand exactly how to use device_options: in the configuration.yaml. 

Are there other examples of configuration.yaml files this use device_options? Maybe another real-world example would make it more clear to me.

Koenkk, 2020-04-06:
Yes it's as simple as that.

Botched1, 2020-04-06:
> Yes it's as simple as that.

Thanks. Not sure why I was making it so hard. Sorry for the dumb/newbie questions.

closed 2020-06-12 by stale[bot]

Patrick1610, 2021-05-06:
Hi @Koenkk, I came across this post and this raises a question for me. Is there a reason that it is turned off by default? What is the downside of enabling retain for all devices? Or is there no downside and is it best to enable this for all devices?

Koenkk, 2021-05-06:
@Patrick1610 you don't want to enable this for e.g. devices that send an `action`. If the last published message is `"action": "single"` (because of a button click) and you enable retain, everytime your client connect it thinks that the button has clicked.

Patrick1610, 2021-05-08:
Alright, that's good to know! I have purged it for all devices to prevent events like that. 
Does it really have an advantage for other devices? For example motion sensors and lights? 
Because, I run on Home Assistant, the latest state is always stored in the entities of a device, even after a brief interruption or restart.

Koenkk, 2021-05-08:
@Patrick1610 for HA it this doesn't matter indeed (since we already publish all states when HA comes online)

th3w1zard1, 2023-03-08:
> Hi @Koenkk, I came across this post and this raises a question for me. Is there a reason that it is turned off by default? What is the downside of enabling retain for all devices? Or is there no downside and is it best to enable this for all devices?

Necroing 2 years later to say thank you for asking this. I've been pulling my hair out trying to figure out why my mqtt broker was constantly pushing action calls over and over again. I went as far as uninstalling zigbee2mqtt, uninstalling hassio broker, and the messages were STILL being published. I was incredibly confused.

Hope this helps someone else out that runs into this problem: be careful using retention settings, you can easily overwhelm your devices if they don't update quick enough.