Koenkk/zigbee2mqtt #3297: [feature] expand extension framework to make it easier to use

sjorge, 2020-04-05:
Some discussion in #3229 led to the need for an easier way to add extension that would not require them to live in z2m repo.

Some things from that:

Excellent, I'll clean this up some more then and it can be a example module.

Perhaps we can move some of the on/off toggles to load extensions to a list where we list what we want to load. (Keeping the old behavior for ha and availability) That would remove the need to add a new field for each extension to the internal configuration file definition.

```yaml
extensions:
  - name: initialize
    configuration:
      config: option
      another: option here too
  - availability
```

Not sure how we can easily allow and extension to add per device config though 🤔 

I'll create a new issue for this, as this one is closed.

sjorge, 2020-04-05:
Possible extension ideas:
- DeviceInitialization (send zigbee messages on device announcement)
- DevicePoll (poll set of attributes at an interval, aka how Hue Bridge does it)

sjorge, 2020-04-07:
Some more food for thought,

- It would be nice to have extensions being able to call other extensions (and have them as a dependency some how.
  e.g. have `Homekit` one required DevicePoll and be able to trigger a poll of a device (say when HomeKit requests a refresh of a service)
- Extend zigbee-herdsman-convertor with generic `device type` tag that has `features`
   - bulb [brightness, colorxy, colorhs, colortemp]
   - switch
   - thermostat
   - sensor [temperature, humidity, ...]
  This can then be consumed by extensions, e.g. home assistant and homekit, they can then still have a 'custom' time for device that don't fit any of the predefined molds... that should hopefully also take away the long list of device mappings we have in the ha extension and only add a few exceptions. Or better yet allow users to add them via the config file. But looking at the list that is currently there... the above type suggestions should cover most and it will also b easy to adapt to things like homekit or and others.

sjorge, 2020-04-11:
That can probably also lead to a refactor of `entityPublish.js` where there are some higher level methods expose via the api to do stuff like `get` and `set` that take a json payload of key-value pairs. 

The existing `entityPublish.js` can just pass through most stuff from mqtt, but then things like DevicePull can simple call `get` with the list of keys configured to poll the device. That would eliminate a lot of the code duplication that the current ugly hacked one has.

Koenkk, 2020-04-19:
I've implemented the extension loading. Extensions that are put in `data/extension` will automatically be loaded.

E.g. by putting https://gist.github.com/Koenkk/16f00544fb7e46f6681495363dfad078 in `data/extension/example.js`.

sjorge, 2020-04-21:
Is there an easy way to get to the logger?
I tried with
```
const logger = require('../util/logger');
```

and 

```
const logger = require('../../util/logger');
```

But that just makes z2m fail to start, perphaps also make errors in extensions non fatal

sjorge, 2020-04-21:
So I was adapting the silly deviceInitialize extension to the new framework and ran into some issues.
- unable to load the logger (see above)
- a broken extension prevents zigbee2mqtt from starting
- error on using this.publishEntityState()

For the `this.publishEntityState()` call I am getting this: (now I might be using the call wrong)

```
(node:39577) UnhandledPromiseRejectionWarning: TypeError: Cannot read property 'constructor' of undefined
    at Zigbee.resolveEntity (/opt/zigbee2mqtt/lib/zigbee.js:148:17)
    at Controller.publishEntityState (/opt/zigbee2mqtt/lib/controller.js:246:44)
    at Initializer.onDeviceAnnounce (/opt/zigbee2mqtt/data/extension/initializer.js:40:14)
    at Initializer.onZigbeeEvent (/opt/zigbee2mqtt/data/extension/initializer.js:45:18)
    at Controller.callExtensionMethod (/opt/zigbee2mqtt/lib/controller.js:348:44)
(node:39577) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 1)
(node:39577) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
```

Here is the entire code
```javascript
/*
 * This extension will apply a state to a device on deviceAnnouncement
 *
 * In `configuration.yaml` onder the device add a `initial_state` object.
 * devices:
 *   '0x000d6ffffe8e8d4f':
 *     friendly_name: masterbedroom/repeater
 *     initial_state:
 *       state: OFF
 *       color: #FF0000
 *
 */
class Initializer {
    constructor(zigbee, mqtt, state, publishEntityState, eventBus) {
        this.zigbee = zigbee;
        this.mqtt = mqtt;
        this.state = state;
        this.publishEntityState = publishEntityState;
        this.eventBus = eventBus;
    }

    onZigbeeStarted() {
        // XXX: logger.info('Loaded Device Initializer');
    }

    onDeviceAnnounce(resolvedEntity) {
        // abort early if we are interviewing
        if (!resolvedEntity.device.interviewCompleted) return;

        // retreive state to apply on initialization
        // XXX: device_options?
        const settings = resolvedEntity.settings;
        const state = settings.hasOwnProperty("initial_state") ? settings.initial_state : null;

        // apply state (if we have any)
        if (!state) return;
        // XXX: logger.debug(`Applying initial state '${JSON.stringify(state}' to '${resolvedEntity.name}'`);
        await this.publishEntityState(resolvedEntity.ieeeAddr, state);
    }

    onZigbeeEvent(type, data, resolvedEntity) {
        if ((type === 'deviceAnnounce') && resolvedEntity)
            this.onDeviceAnnounce(resolvedEntity);
    }
}

module.exports = Initializer;
```

Koenkk, 2020-04-21:
- Logger is now passed down as 6th argument in the constructor
- "a broken extension prevents zigbee2mqtt from starting", I don't feel this is wrong, this means the construction failed which is not something that should happen. 
- `publishEntityState` can you debug a bit in zigbee/resolveEntity()? Not sure what goes wrong.

sjorge, 2020-04-21:
I update the above a bit with another issue, I was initially  confusing with a bad constructor, as I just had the file open in vim and the .swp was causing z2m to also fail.

Koenkk, 2020-04-21:
`await this.publishEntityState(resolvedEntity.ieeeAddr, state);` should be `await this.publishEntityState(resolvedEntity.device.ieeeAddr, state);` (otherwise it will be null which explains the error).

sjorge, 2020-04-21:
- logger now works, pretty logs <3
- the call to  `this.publishEntityState` now works, I also had to drop the await.

Sadly the call did not do what I though xD... I was hoping it would also send the message to the device.

Enough messing about for today, but let me summarize the above few comments:
- bad constructor prevents z2m startup (working as intended)
- non .js named file in data/extensions causes z2m startup failure (probably not what we want, if I have a file open in vim and restart z2m to test it fails)
- some way to extend the configuration.yaml schema might be useful (https://github.com/Koenkk/zigbee2mqtt/blob/d22e73ebb41f7fc0a72dae08890e0803f16c864b/lib/util/settings.js#L125)
- a generic 'set this property/these properties on device/group' and 'read this property' would be useful

I guess for the last point we could move some of the logic from entityPublish.js into controller ? There may already be a way though, I'll need to take a closer look at homeassistant.js as I think it also publishes states to the device.

Koenkk, 2020-04-21:
To send commands to device this can be used (its a bit hacky but good enough).

```
this.mqtt.emit('message', {topic: "zigbee2mqtt/FRIENDLY_NAME/set`, message: JSON.stringify({state: "ON"})}`

sjorge, 2020-04-21:
I'll try that tomorrow, just need to find a way to get the `zigbee2mqtt` part for the topic.

I also opened https://github.com/Koenkk/zigbee2mqtt/pull/3409 to fix the other crashing issue, the one that shouldn't be fatal.

sjorge, 2020-04-22:
Is there an easy way to get the settings object to fetch the topic:
```${settings.get().mqtt.base_topic}```

I'll see if I can get the basic thing working, the mqtt.emit might work for basic cases and I want to stub out some fancy example extensions first using that. But I think in the long run it would be nice to have a few more 'premade' functions available.

Should I/we create a repo somewhere that hold 'best effort' extensions that people can plug and use by just downloading the .js from the repo?

sjorge, 2020-04-22:
Hmm seeing very weird stuff (this assumes #3410 is already applied)

restore.js + sanitized configuration.yaml
https://gist.github.com/sjorge/5018d939402f35b57a1f9b9849f19d91

- device_options.restore_state seems to be applied (yay)
- desk_lamp applies the color blue as configured also
- bed_lamp also applies the color blue... which it should not  do?

sjorge, 2020-04-22:
Alright I have a minimal working 'restore' plugging that can set state on deviceAnnouncement!

https://gist.github.com/sjorge/5018d939402f35b57a1f9b9849f19d91, I'll look into some of the mentioned things above over the weekend:
- some way to extend the configuration.yaml schema (will need to think on this a bit)
- generic get/set helper than can also be used by lib/extension/publish.js

sjorge, 2020-04-23:
@Koenkk I have an idea on how to extend the schema...
An optional function for the extensions onSettingsSchema() that then additional schema data to merge... although I think that might be a chicken-egg issue as settings is already loaded and passes to the extension right?

Koenkk, 2020-04-23:
@sjorge settings can already be added to `configuration.yaml` without problems. Not sure if adding the schema validation is worth it, it complicates things and does not really add much value?

sjorge, 2020-04-23:
I did notice the validate function was never called, but I guess it would be nice to have it work though?

My example extension does add stuff to device_options or the device config.

Koenkk, 2020-04-24:
I would be a nice too have indeed, but I currently don't see how to easily implement it.

sjorge, 2020-04-24:
I'm not good enough with node to know if something like this could work

- controller
  - load all extensions in lib/extensions and data/extensions and only pass a reference to controller (self)
  - load 'settings', 'logger', 'utils' (or whatever will have some of the util functions eventually, ...
  - for x in extensions: x.onSchemaUpdate() -> update 'settings'
  - settings.validate()siginfo->si_addr
  - for x in extesions: x.onStartup()
  - ...

If the in the extension construct set this.controller = controller, then access the other stuff via
this.controller.getLogger(), this.controller.getSettings(), ...

But not sure how all the ordering would work in node, this works okish in python.

sjorge, 2020-04-25:
So I was already looking at the poll extension and I need a way to do a read that will wait for a return value... is that even possible?

Like read the genOnOff, then return once it is read (ideally with the value)

Koenkk, 2020-04-26:
This should work: `await endpoint.read('genOnOff', ['onOff']);`, bulb will respond where the message will be picked up by `receive.js` which will publish it to MQTT

sjorge, 2020-05-08:
Been a bit busy with other stuff, I'll try to pick this back up soon.

sjorge, 2020-05-15:
OK so I got a failure with a lot of 'leave' requests with:
- all stuff online
- one router online it was still attached by unplugging and re plugging so I saw the device announcement

I didn't get the weird association message this time with just the coordinator (and some sensors on battery I can't reach)

Uploading caps now but using my phone as I turned of the wrong breaker and have no internet. Oops.

sjorge, 2020-05-31:
Looks like devices randomly reannounce them self and that is failing, I will try to grab a capture.

sjorge, 2020-05-31:
https://pkg.blackdot.be/cores/zigbee/errors.pcapng

There is one near packet 330 but a few before that, but they scrolled by to fast.
Note, this is not during pairing... this is just existing devices dropping and failing to rejoin after the device announcement... going to refresh the normal source routing firmware tomorrow morning as nothing is working atm.

closed 2020-07-26 by stale[bot]