Koenkk/zigbee2mqtt #3298: Improve groups handling

magpern, 2020-04-05:
I just bought four new IKEA remotes and a couple of new lights and fyrtur blinds.

To configure these in Z2M is quite a job. The renaming of the devices I did through Zigbee2MqttAssistand, but the group binding, for the remotes to work, is quite an advanced step.

First create a group in configuration.yaml, then restart Z2M for the group to exist, then manually bind all the remotes to the group and them manually add all the devices in each group.

If my father was to do this, he would fail...

So I was thinking. If Z2M would have a `creategroup`, `removegroup`, `listgroups` command, this could be implemented in `zigbee2mqttAssistant` and you would get a graphical interface for this tedious task.

Perhaps this is partly related to #3281?

sjorge, 2020-04-06:
These already exists, see https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html

- zigbee2mqtt/bridge/config/add_group
   > Allows you to add a group, payload should be the name of the group, e.g. my_group.
   >
   > In case you also want to specify the group ID, provide the following payload {"friendly_name": "my_group", "id": 42}.
-  zigbee2mqtt/bridge/config/remove_group
   > Allows you to remove a group, payload should be the name of the group, e.g. my_group.
- zigbee2mqtt/bridge/log
  > can be triggered to publish the whole playload, that includes a `groups` sub key to list them.
Additionally:
-  zigbee2mqtt/bridge/device/[friendly_name]/get_group_membership
   > Returns the list of groups a device is in, and its group capacity.
- zigbee2mqtt/bridge/group/[friendly_name]/(add|remove|remove_all)
   > See https://www.zigbee2mqtt.io/information/groups.html

magpern, 2020-04-06:
@sjorge that was a nice surprise! Thank you for this good answer.

closed 2020-04-06 by magpern

sjorge, 2020-04-06:
Your welcome, sometimes some of the useful stuff is a bit hidding in the mqtt topic and messages page... hopefully after the zigbe2mqtt/bridge refactor, some of this can be improved to to have moving stuff under the zigbee2mqtt/bridge/group topic.