Koenkk/zigbee2mqtt #3302: Support For Tuya TS0121 UK Plug with Power Monitoring

milobt, 2020-04-06:
# Bug Report
## What happened
Need help supporting a new device: Tuya TS0121 UK Plug with Power Monitoring
[https://www.aliexpress.com/item/4000688343995.html?spm=a2g0s.9042311.0.0.11994c4dsnM3Lq](url)

Thanks in advance.

## Debug Info
zigbee2mqtt version: 1.12.0
CC26X2R1 firmware version: 20191106

milobt, 2020-04-08:
Pairs and on/off works - via MQTT command with the following:
`{
        zigbeeModel: ['TS0121'],
        model: 'TS0121',
        vendor: 'Tuyatec',
        description: 'UK Plug with power meter',
        supports: 'on/off, power measurement',
        fromZigbee: [fz.on_off, fz.electrical_measurement_power],
        toZigbee: [tz.on_off],`

Need help figuring out how to get the power stats.

Koenkk, 2020-04-09:
Please try with

```js
{
    zigbeeModel: ['TS0121'],
    model: 'TS0121',
    description: '10A UK Power Monitoring Tuya ZigBee Smart Plug',
    supports: 'on/off, power measurement',
    vendor: 'TuYa',
    fromZigbee: [fz.on_off, fz.electrical_measurement_power],
    toZigbee: [tz.on_off],
    meta: {configureKey: 1},
    configure: async (device, coordinatorEndpoint) => {
        const endpoint = device.getEndpoint(1);
        await bind(endpoint, coordinatorEndpoint, ['genOnOff', 'haElectricalMeasurement']);
        await configureReporting.onOff(endpoint);
        await readEletricalMeasurementPowerConverterAttributes(endpoint);
        await configureReporting.rmsVoltage(endpoint);
        await configureReporting.rmsCurrent(endpoint);
        await configureReporting.activePower(endpoint);
    },
},
```

milobt, 2020-04-11:
So added your suggestion, above to devices.js and  'TS0121': [cfg.switch, cfg.sensor_power], to homeassistant.js and got it to show in the MQTT integrations pane.
I get power state and function and also link quality, but has not been able to configure power monitoring:

`zigbee2mqtt:error 2020-04-11 07:36:51: Failed to configure '0xec1bbdfffe0c1e8f', attempt 3 (Error: ConfigureReporting 0xec1bbdfffe0c1e8f/1 genOnOff([{"attribute":"onOff","minimumReportInterval":0,"maximumReportInterval":3600,"reportableChange":0}], {"timeout":6000,"manufacturerCode":null,"disableDefaultResponse":true}) failed (Error: Timeout - 36326 - 1 - 166 - 6 - 7 after 6000ms)
    at Endpoint.<anonymous> (/zigbee2mqtt-1.12.2/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:326:23)
    at Generator.throw (<anonymous>)
    at rejected (/zigbee2mqtt-1.12.2/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65))

zigbee2mqtt:debug 2020-04-11 07:45:02: Received Zigbee message from '0xec1bbdfffe0c1e8f', type 'attributeReport', cluster 'genBasic', data '{"appVersion":255}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-04-11 07:45:02: No converter available for 'TS0121' with cluster 'genBasic' and type 'attributeReport' and data '{"appVersion":255}'
zigbee2mqtt:debug 2020-04-11 07:45:07: Received Zigbee message from '0xec1bbdfffe0c1e8f', type 'attributeReport', cluster 'genBasic', data '{"appVersion":255}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-04-11 07:45:07: No converter available for 'TS0121' with cluster 'genBasic' and type 'attributeReport' and data '{"appVersion":255}'
zigbee2mqtt:debug 2020-04-11 07:45:08: Received Zigbee message from '0xec1bbdfffe0c1e8f', type 'attributeReport', cluster 'genBasic', data '{"appVersion":255}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-04-11 07:45:08: No converter available for 'TS0121' with cluster 'genBasic' and type 'attributeReport' and data '{"appVersion":255}'
zigbee2mqtt:debug 2020-04-11 07:45:11: Received Zigbee message from '0xec1bbdfffe0c1e8f', type 'attributeReport', cluster 'genBasic', data '{"appVersion":67}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-04-11 07:45:11: No converter available for 'TS0121' with cluster 'genBasic' and type 'attributeReport' and data '{"appVersion":67}'
zigbee2mqtt:debug 2020-04-11 07:45:13: Received Zigbee message from '0xec1bbdfffe0c1e8f', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0,"onTime":0,"offWaitTime":0}' from endpoint 1 with groupID 0`

Koenkk, 2020-04-11:
Please try with

```js
{
    zigbeeModel: ['TS0121'],
    model: 'TS0121',
    description: '10A UK Power Monitoring Tuya ZigBee Smart Plug',
    supports: 'on/off, power measurement',
    vendor: 'TuYa',
    fromZigbee: [fz.on_off, fz.electrical_measurement_power],
    toZigbee: [tz.on_off],
    meta: {configureKey: 1},
    configure: async (device, coordinatorEndpoint) => {
        const endpoint = device.getEndpoint(1);
        await bind(endpoint, coordinatorEndpoint, ['genOnOff', 'haElectricalMeasurement']);
    },
},
```

milobt, 2020-04-11:
Still not getting the power value.

Error:
`zigbee2mqtt:error 2020-04-11 19:43:55: Failed to setup reporting for '0xec1bbdfffe0c1e8f' - Error: ConfigureReporting 0xec1bbdfffe0c1e8f/1 genOnOff([{"attribute":"onOff","minimumReportInterval":0,"maximumReportInterval":300,"reportableChange":0}], {"timeout":6000,"manufacturerCode":null,"disableDefaultResponse":true}) failed (Error: Timeout - 20283 - 1 - 11 - 6 - 7 after 6000ms)
    at Endpoint.<anonymous> (/zigbee2mqtt-1.12.2/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:326:23)
    at Generator.throw (<anonymous>)
    at rejected (/zigbee2mqtt-1.12.2/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65)`

Koenkk, 2020-04-12:
Do you see anything in the zigbee2mqtt debug log related to power measurements when attaching a load to the device?

milobt, 2020-04-12:
On two occasions of unplugging replugging a load I got these two outputs in the logs:

zigbee2mqtt:debug 2020-04-12 12:55:01: Received Zigbee message from '0xec1bbdfffe0c1e8f', type 'readResponse', cluster 'genBasic', data '{"zclVersion":3}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-04-12 12:55:01: No converter available for 'TS0121' with cluster 'genBasic' and type 'readResponse' and data '{"zclVersion":3}'

zigbee2mqtt:debug 2020-04-12 12:56:17: Received Zigbee message from '0xec1bbdfffe0c1e8f', type 'attributeReport', cluster 'genBasic', data '{"appVersion":67}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-04-12 12:56:17: No converter available for 'TS0121' with cluster 'genBasic' and type 'attributeReport' and data '{"appVersion":67}'

Koenkk, 2020-04-12:
Can you try with:

```js
{
    zigbeeModel: ['TS0121'],
    model: 'TS0121',
    description: '10A UK Power Monitoring Tuya ZigBee Smart Plug',
    supports: 'on/off, power measurement',
    vendor: 'TuYa',
    fromZigbee: [fz.on_off, fz.electrical_measurement_power],
    toZigbee: [tz.on_off],
    meta: {configureKey: 1},
    configure: async (device, coordinatorEndpoint) => {
        const endpoint = device.getEndpoint(1);
        await bind(endpoint, coordinatorEndpoint, ['genOnOff', 'haElectricalMeasurement']);
        await readEletricalMeasurementPowerConverterAttributes(endpoint);
        await configureReporting.rmsVoltage(endpoint);
        await configureReporting.rmsCurrent(endpoint);
        await configureReporting.activePower(endpoint);
    },
},
```

If that doesn't work with:

```js
{
    zigbeeModel: ['TS0121'],
    model: 'TS0121',
    description: '10A UK Power Monitoring Tuya ZigBee Smart Plug',
    supports: 'on/off, power measurement',
    vendor: 'TuYa',
    fromZigbee: [fz.on_off, fz.metering_power],
    toZigbee: [tz.on_off],
    meta: {configureKey: 1},
    configure: async (device, coordinatorEndpoint) => {
        const endpoint = device.getEndpoint(1);
        await bind(endpoint, coordinatorEndpoint, ['genOnOff', 'seMetering']);
            await readMeteringPowerConverterAttributes(endpoint);
            await configureReporting.instantaneousDemand(endpoint);
    },
},
```

milobt, 2020-04-14:
Unfortunately, neither worked...
To clarify I have been updating the docker and restarting it (not restarting the plug-in in HASSIO).
I notice that this itself restarts the docker and starts a new log.
Should I need to do anything else to ensure the changes are implemented?

Log output for first suggestions:
`debug 2020-04-14 10:45:44: Setup reporting for '0xec1bbdfffe0c1e8f' - 1 - genOnOff

error 2020-04-14 10:46:00: Failed to setup reporting for '0xec1bbdfffe0c1e8f' - Error: ConfigureReporting 0xec1bbdfffe0c1e8f/1 genOnOff([{"attribute":"onOff","minimumReportInterval":0,"maximumReportInterval":300,"reportableChange":0}], {"timeout":6000,"manufacturerCode":null,"disableDefaultResponse":true}) failed (Error: Timeout - 20283 - 1 - 11 - 6 - 7 after 6000ms)
    at Endpoint.<anonymous> (/zigbee2mqtt-1.12.2/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:326:23)
    at Generator.throw (<anonymous>)
    at rejected (/zigbee2mqtt-1.12.2/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65)`

zigbee2mqtt:debug 2020-04-14 10:50:38: Received Zigbee message from '0xec1bbdfffe0c1e8f', type 'readResponse', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-04-14 10:50:38: MQTT publish: topic 'zigbee2mqtt/0xec1bbdfffe0c1e8f', payload '{"state":"ON","linkquality":129}'

zigbee2mqtt:debug 2020-04-14 10:51:18: Received MQTT message on 'zigbee2mqtt/0xec1bbdfffe0c1e8f/set' with data 'ON'
zigbee2mqtt:debug 2020-04-14 10:51:18: Publishing 'set' 'state' to '0xec1bbdfffe0c1e8f'
zigbee2mqtt:debug 2020-04-14 10:51:18: Received Zigbee message from '0xec1bbdfffe0c1e8f', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-04-14 10:51:18: MQTT publish: topic 'zigbee2mqtt/0xec1bbdfffe0c1e8f', payload '{"state":"ON","linkquality":75}'

Log output for second suggestions:
`debug 2020-04-14 10:59:25: Setup reporting for '0xec1bbdfffe0c1e8f' - 1 - genOnOff
error 2020-04-14 10:59:40: Failed to setup reporting for '0xec1bbdfffe0c1e8f' - Error: ConfigureReporting 0xec1bbdfffe0c1e8f/1 genOnOff([{"attribute":"onOff","minimumReportInterval":0,"maximumReportInterval":300,"reportableChange":0}], {"timeout":6000,"manufacturerCode":null,"disableDefaultResponse":true}) failed (Error: Timeout - 20283 - 1 - 11 - 6 - 7 after 6000ms)
    at Endpoint.<anonymous> (/zigbee2mqtt-1.12.2/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:326:23)
    at Generator.throw (<anonymous>)
    at rejected (/zigbee2mqtt-1.12.2/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65)
debug 2020-04-14 11:00:18: Received MQTT message on 'zigbee2mqtt/0xec1bbdfffe0c1e8f/set' with data 'OFF'
debug 2020-04-14 11:00:18: Publishing 'set' 'state' to '0xec1bbdfffe0c1e8f'
debug 2020-04-14 11:00:18: Received Zigbee message from '0xec1bbdfffe0c1e8f', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
info  2020-04-14 11:00:18: MQTT publish: topic 'zigbee2mqtt/0xec1bbdfffe0c1e8f', payload '{"state":"OFF","linkquality":69}'
info  2020-04-14 11:00:18: MQTT publish: topic 'zigbee2mqtt/0xec1bbdfffe0c1e8f', payload '{"state":"OFF","linkquality":69}'
debug 2020-04-14 11:00:18: Received Zigbee message from '0xec1bbdfffe0c1e8f', type 'readResponse', cluster 'genBasic', data '{"zclVersion":3}' from endpoint 1 with groupID 0
debug 2020-04-14 11:00:18: No converter available for 'M8-EM' with cluster 'genBasic' and type 'readResponse' and data '{"zclVersion":3}
debug 2020-04-14 11:00:19: Received MQTT message on 'zigbee2mqtt/0xec1bbdfffe0c1e8f/set' with data 'ON'
debug 2020-04-14 11:00:19: Publishing 'set' 'state' to '0xec1bbdfffe0c1e8f'
debug 2020-04-14 11:00:19: Received Zigbee message from '0xec1bbdfffe0c1e8f', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
info  2020-04-14 11:00:19: MQTT publish: topic 'zigbee2mqtt/0xec1bbdfffe0c1e8f', payload '{"state":"ON","linkquality":72}'
info  2020-04-14 11:00:19: MQTT publish: topic 'zigbee2mqtt/0xec1bbdfffe0c1e8f', payload '{"state":"ON","linkquality":72}'`

Neither, with a load attached, populates the power sensor with any values.

I did just spot on the back of the switch that it says the model number is M8-EM, but identifies to Z2M as TS0121.
Can't find traces of either on google or Zigbee aliance archive.

Thanks for you help btw in trying to sort this!

Koenkk, 2020-04-14:
Can you share your database.db and disable the report feature in your `configuration.yaml`?

milobt, 2020-04-14:
[https://pastebin.com/gg6YwpPd](https://pastebin.com/gg6YwpPd)

Device: 0xec1bbdfffe0c1e8f

Koenkk, 2020-04-14:
After using the new definitions did you reconfigure the device via: https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigure

milobt, 2020-04-14:
No. I didn’t realise this was necessary. 
I see it reload the add-on after restarting the docker, does this not run a reconfigure at this point?

I will retry the definitions above tomorrow and run the reconfigure command with each. 

Thanks.

milobt, 2020-04-15:
So have tried all the definitions above again with the MQTT configure call and still no luck.

The third definition looked more promising as produced info report, instead of error after docker restart, but still produced errors after reconfigure.

Definition:
```
{
    zigbeeModel: ['TS0121'],
    model: 'TS0121',
    description: '10A UK Power Monitoring Tuya ZigBee Smart Plug',
    supports: 'on/off, power measurement',
    vendor: 'TuYa',
    fromZigbee: [fz.on_off, fz.metering_power],
    toZigbee: [tz.on_off],
    meta: {configureKey: 1},
    configure: async (device, coordinatorEndpoint) => {
        const endpoint = device.getEndpoint(1);
        await bind(endpoint, coordinatorEndpoint, ['genOnOff', 'seMetering']);
            await readMeteringPowerConverterAttributes(endpoint);
            await configureReporting.instantaneousDemand(endpoint);
    },
},
```

Log:
```
zigbee2mqtt:debug 2020-04-15 09:38:01: Received MQTT message on 'zigbee2mqtt/0xec1bbdfffe0c1e8f/set' with data 'OFF'
zigbee2mqtt:debug 2020-04-15 09:38:01: Publishing 'set' 'state' to '0xec1bbdfffe0c1e8f'
zigbee2mqtt:debug 2020-04-15 09:38:01: Received Zigbee message from '0xec1bbdfffe0c1e8f', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-04-15 09:38:01: MQTT publish: topic 'zigbee2mqtt/0xec1bbdfffe0c1e8f', payload '{"state":"OFF","linkquality":75}'
(node:229) UnhandledPromiseRejectionWarning: TypeError: Cannot read property 'includes' of undefined
    at HomeAssistant.onPublishEntityState (/zigbee2mqtt-1.12.2/lib/extension/homeassistant.js:1634:59)
    at EventBus.<anonymous> (/zigbee2mqtt-1.12.2/lib/extension/homeassistant.js:1541:63)
    at EventBus.emit (events.js:223:5)
    at EventBus.emit (/zigbee2mqtt-1.12.2/lib/eventBus.js:15:15)
    at Controller.publishEntityState (/zigbee2mqtt-1.12.2/lib/controller.js:292:23)
    at publish (/zigbee2mqtt-1.12.2/lib/extension/deviceReceive.js:146:22)
    at DeviceReceive.onZigbeeEvent (/zigbee2mqtt-1.12.2/lib/extension/deviceReceive.js:176:13)
    at Controller.callExtensionMethod (/zigbee2mqtt-1.12.2/lib/controller.js:336:44)
    at Controller.onZigbeeEvent (/zigbee2mqtt-1.12.2/lib/controller.js:222:14)
    at Zigbee.emit (events.js:223:5)
(node:229) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 5)

zigbee2mqtt:error 2020-04-15 09:38:44: Failed to configure '0xec1bbdfffe0c1e8f', attempt 1 (Error: Read 0xec1bbdfffe0c1e8f/1 seMetering(["multiplier","divisor"], {"timeout":6000,"manufacturerCode":null,"disableDefaultResponse":true}) failed (Error: Status 'UNSUPPORTED_ATTRIBUTE')
    at Endpoint.<anonymous> (/zigbee2mqtt-1.12.2/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:182:23)
    at Generator.next (<anonymous>)
    at fulfilled (/zigbee2mqtt-1.12.2/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:5:58))

zigbee2mqtt:error 2020-04-15 09:40:37: Failed to setup reporting for '0xec1bbdfffe0c1e8f' - Error: Bind 0xec1bbdfffe0c1e8f/1 genOnOff from 'undefined' failed (TypeError: Cannot read property 'toString' of undefined)
    at Endpoint.<anonymous> (/zigbee2mqtt-1.12.2/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:244:23)
    at Generator.throw (<anonymous>)
    at rejected (/zigbee2mqtt-1.12.2/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65)
```

milobt, 2020-04-15:
FYI, this is the result of the other two definitions:

Definition:
```
{
    zigbeeModel: ['TS0121'],
    model: 'TS0121',
    description: '10A UK Power Monitoring Tuya ZigBee Smart Plug',
    supports: 'on/off, power measurement',
    vendor: 'TuYa',
    fromZigbee: [fz.on_off, fz.electrical_measurement_power],
    toZigbee: [tz.on_off],
    meta: {configureKey: 1},
    configure: async (device, coordinatorEndpoint) => {
        const endpoint = device.getEndpoint(1);
        await bind(endpoint, coordinatorEndpoint, ['genOnOff', 'haElectricalMeasurement']);
    },
},
```

Log:

```
zigbee2mqtt:info  2020-04-15 09:27:42: MQTT publish: topic 'zigbee2mqtt/0xec1bbdfffe0c1e8f', payload '{"state":"ON","linkquality":78}'
(node:229) UnhandledPromiseRejectionWarning: TypeError: Cannot read property 'includes' of undefined
    at HomeAssistant.onPublishEntityState (/zigbee2mqtt-1.12.2/lib/extension/homeassistant.js:1634:59)
    at EventBus.<anonymous> (/zigbee2mqtt-1.12.2/lib/extension/homeassistant.js:1541:63)
    at EventBus.emit (events.js:223:5)
    at EventBus.emit (/zigbee2mqtt-1.12.2/lib/eventBus.js:15:15)
    at Controller.publishEntityState (/zigbee2mqtt-1.12.2/lib/controller.js:292:23)
    at publish (/zigbee2mqtt-1.12.2/lib/extension/deviceReceive.js:146:22)
    at DeviceReceive.onZigbeeEvent (/zigbee2mqtt-1.12.2/lib/extension/deviceReceive.js:176:13)
    at Controller.callExtensionMethod (/zigbee2mqtt-1.12.2/lib/controller.js:336:44)
    at Controller.onZigbeeEvent (/zigbee2mqtt-1.12.2/lib/controller.js:222:14)
    at Zigbee.emit (events.js:223:5)
(node:229) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 16)
(node:229) UnhandledPromiseRejectionWarning: TypeError: Cannot read property 'includes' of undefined
    at HomeAssistant.onPublishEntityState (/zigbee2mqtt-1.12.2/lib/extension/homeassistant.js:1634:59)
    at EventBus.<anonymous> (/zigbee2mqtt-1.12.2/lib/extension/homeassistant.js:1541:63)
    at EventBus.emit (events.js:223:5)
    at EventBus.emit (/zigbee2mqtt-1.12.2/lib/eventBus.js:15:15)
    at Controller.publishEntityState (/zigbee2mqtt-1.12.2/lib/controller.js:292:23)
    at EntityPublish.onMQTTMessage (/zigbee2mqtt-1.12.2/lib/extension/entityPublish.js:194:30)
    at async Controller.callExtensionMethod (/zigbee2mqtt-1.12.2/lib/controller.js:336:21)
(node:229) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 17)
```

---

Definition:

```
{
    zigbeeModel: ['TS0121'],
    model: 'TS0121',
    description: '10A UK Power Monitoring Tuya ZigBee Smart Plug',
    supports: 'on/off, power measurement',
    vendor: 'TuYa',
    fromZigbee: [fz.on_off, fz.electrical_measurement_power],
    toZigbee: [tz.on_off],
    meta: {configureKey: 1},
    configure: async (device, coordinatorEndpoint) => {
        const endpoint = device.getEndpoint(1);
        await bind(endpoint, coordinatorEndpoint, ['genOnOff', 'haElectricalMeasurement']);
        await readEletricalMeasurementPowerConverterAttributes(endpoint);
        await configureReporting.rmsVoltage(endpoint);
        await configureReporting.rmsCurrent(endpoint);
        await configureReporting.activePower(endpoint);
    },
},
```

Log:

```
zigbee2mqtt:debug 2020-04-15 09:34:36: Received MQTT message on 'zigbee2mqtt/bridge/configure' with data '0xec1bbdfffe0c1e8f'
zigbee2mqtt:info  2020-04-15 09:34:36: Configuring '0xec1bbdfffe0c1e8f'
zigbee2mqtt:debug 2020-04-15 09:34:36: Received Zigbee message from '0xec1bbdfffe0c1e8f', type 'readResponse', cluster 'haElectricalMeasurement', data '{}' from endpoint 1 with groupID 0
zigbee2mqtt:error 2020-04-15 09:34:36: Failed to configure '0xec1bbdfffe0c1e8f', attempt 1 (Error: Read 0xec1bbdfffe0c1e8f/1 haElectricalMeasurement(["acVoltageMultiplier","acVoltageDivisor","acCurrentMultiplier"], {"timeout":6000,"manufacturerCode":null,"disableDefaultResponse":true}) failed (Error: Status 'UNSUPPORTED_ATTRIBUTE')
    at Endpoint.<anonymous> (/zigbee2mqtt-1.12.2/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:182:23)
    at Generator.next (<anonymous>)
    at fulfilled (/zigbee2mqtt-1.12.2/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:5:58))
```

Koenkk, 2020-04-15:
Can you try with the latest dev branch and:

```js
{
    zigbeeModel: ['TS0121'],
    model: 'TS0121',
    description: '10A UK Power Monitoring Tuya ZigBee Smart Plug',
    supports: 'on/off, power measurement',
    vendor: 'TuYa',
    fromZigbee: [fz.on_off, fz.metering_power],
    toZigbee: [tz.on_off],
    meta: {configureKey: 1},
    configure: async (device, coordinatorEndpoint) => {
        const endpoint = device.getEndpoint(1);
        await bind(endpoint, coordinatorEndpoint, ['genOnOff', 'seMetering']);
            await configureReporting.instantaneousDemand(endpoint);
    },
},
```

milobt, 2020-04-16:
What's the best way of using dev branch with modified devices?
Does the dev branch use the dame docker and same devices.js file or is there another to edit?
Or am I better using the devices.js file in the share/zigbee2mqtt folder and setting zigbee_shepherd_devices: true in the config?

Koenkk, 2020-04-16:
It uses the same, the `zigbee_shepherd_devices` is the recommended way.

milobt, 2020-04-17:
Everything seems to have screwed up.
I switched to edge and and enabled zigbee-shepherd_devices and it didn't load.
I think it may have had the same pan_id - does this matter?
My pan_id was 6755 and I changed it to 6756 - does the format matter?
Should I/can I swapped back to the old Pan_id?

Tried to swap back to the normal branch and it didn't want to start up.
Tried changing the pan_id and it wiped my database. Luckily I had a backup, but that only allows it to recognise devices, but nothing seems to be reconnecting - everything is unavailable in HASSIO.

Seems to be sending topics to devices, but LQIs are failing.

milobt, 2020-04-17:
Everything seems to have screwed up.
I switched to edge and and enabled zigbee-shepherd_devices and it didn't load.
I think it may have had the same pan_id - does this matter?
My pan_id was 6755 and I changed it to 6756 - does the format matter?
Should I/can I swapped back to the old Pan_id?

Tried to swap back to the normal branch and it didn't want to start up.
Tried changing the pan_id and it wiped my database. Luckily I had a backup, but that only allows it to recognise devices, but nothing seems to be reconnecting - everything is unavailable in HASSIO.

I enabled joining and that looked like it worked, but then all the devices disconnect again.

Tries to be sending topics to devices, but fails, as does LQIs.

Log: [https://pastebin.com/6LaL6Rnm](https://pastebin.com/6LaL6Rnm)

milobt, 2020-04-17:
I have attempted to repair devices, which worked for a short period and now they have disconnected again.

I'm tempted to try a snapshot backup, but is from 3 weeks ago and have already tried a restore of just the z2m add-ons without success.
Is it likely that my coordinator firmware has become corrupt?
(I have a CC26X2R1, not the standard CC2531)

Koenkk, 2020-04-17:
Yes the config is the same, but I think you also need to copy the configuration from the web interface from the standard to edge addon. If the pan_id, network_key or channel is different zigbee2mqtt will reset itself.

milobt, 2020-04-22:
I'm on edge version and still will not configure the Tuya TS0121 device.
Only on/off controllable.

Does the edge add-on files update separately to the standard add-on?
Otherwise, how do I know if there is an update to the files?

Definition tried:

```
{
    zigbeeModel: ['TS0121'],
    model: 'TS0121',
    description: '10A UK Power Monitoring Tuya ZigBee Smart Plug',
    supports: 'on/off, power measurement',
    vendor: 'TuYa',
    fromZigbee: [fz.on_off, fz.metering_power],
    toZigbee: [tz.on_off],
    meta: {configureKey: 1},
    configure: async (device, coordinatorEndpoint) => {
        const endpoint = device.getEndpoint(1);
        await bind(endpoint, coordinatorEndpoint, ['genOnOff', 'seMetering']);
            await configureReporting.instantaneousDemand(endpoint);
    },
},
```

Error:

```
error 2020-04-22 10:16:41: Failed to configure '0xec1bbdfffe0c1e8f', attempt 1 (Error: ConfigureReporting 0xec1bbdfffe0c1e8f/1 seMetering([{"attribute":"instantaneousDemand","minimumReportInterval":0,"maximumReportInterval":3600,"reportableChange":1}], {"timeout":10000,"manufacturerCode":null,"disableDefaultResponse":true}) failed (Error: Timeout - 39688 - 1 - 6 - 1794 - 7 after 10000ms)
    at Endpoint.<anonymous> (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:326:23)
    at Generator.throw (<anonymous>)
    at rejected (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65))
debug 2020-04-22 10:16:41: Received Zigbee message from '0xec1bbdfffe0c1e8f', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
info  2020-04-22 10:16:41: MQTT publish: topic 'zigbee2mqtt/0xec1bbdfffe0c1e8f', payload '{"state":"OFF","linkquality":75}'
info  2020-04-22 10:16:41: Configuring '0xec1bbdfffe0c1e8f'
info  2020-04-22 10:16:41: MQTT publish: topic 'zigbee2mqtt/0xec1bbdfffe0c1e8f', payload '{"state":"OFF","linkquality":75}'
error 2020-04-22 10:17:04: Failed to configure '0xec1bbdfffe0c1e8f', attempt 2 (Error: ConfigureReporting 0xec1bbdfffe0c1e8f/1 seMetering([{"attribute":"instantaneousDemand","minimumReportInterval":0,"maximumReportInterval":3600,"reportableChange":1}], {"timeout":10000,"manufacturerCode":null,"disableDefaultResponse":true}) failed (Error: Timeout - 39688 - 1 - 11 - 1794 - 7 after 10000ms)
    at Endpoint.<anonymous> (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:326:23)
    at Generator.throw (<anonymous>)
    at rejected (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65))
```

Koenkk, 2020-04-22:
In this case I don't know how to setup the power reporting for this device. Probably something non-standard is used (which is common for TuYa devices). We would need to sniff the traffic with the original gateway in order to figure this out: https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

milobt, 2020-04-23:
Yesterday I paired it with my Echo Plus with built-in zigbee hub and could only control on/off; it didn't offer any power readings, but I am not sure whether this is the case with other smart switches with power metre support?

I have emailed the seller to check that these plugs do support power metre - they explicitly specify that they do on the seller's listing.

For now I am tempted to just setup the plugs with on/off support as that is all I really require, power metre is a nice to have.
But would be nice to get it working in the future, as this is a feature they should have.

I might get on and buy a Tuya hub (~£15) and then can try sniffing the packets. Will need your help in deciphering what it all means though.

Can I ask for you to implement a release with the support for on/off for now?

homeassistant.js:
`'TS0121': [cfg.switch],`

Devices.js:
```
{
        zigbeeModel: ['TS0121'],
        model: 'TS0121',
        description: '10A UK Power Monitoring Tuya ZigBee Smart Plug',
        supports: 'on/off, power measurement',
        vendor: 'TuYa',
        fromZigbee: [fz.on_off, fz.metering_power],
        toZigbee: [tz.on_off],
        // meta: {configureKey: 1},
        // configure: async (device, coordinatorEndpoint) => {
        //     const endpoint = device.getEndpoint(1);
        //     await bind(endpoint, coordinatorEndpoint, ['genOnOff', 'seMetering']);
        //         await configureReporting.instantaneousDemand(endpoint);
        // },
    },
```

Koenkk, 2020-04-23:
Done, power sensor can be added later without problems.

milobt, 2020-04-27:
Have had confirmation from the seller of these switches (on Ali Express) that they do not support power monitoring!!! So, unfortunately, this will never be the case.

So, thanks for your support and sorry for wasting your time.

Is the change in the dev branch?
Does this update automatically in HASSIO add-on or do I need to reinstall?
(I don't ever seem to get a prompt to update the dev branch).

Koenkk, 2020-04-27:
Its already in, to update to latest edge remove and reinstall. Can we close this issue?

closed 2020-04-27 by milobt

mabosch74, 2020-06-27:
> Have had confirmation from the seller of these switches (on Ali Express) that they do not support power monitoring!!! So, unfortunately, this will never be the case.

that can't be true. BlitzWolf annouced it with power metering, see here in my log:
`debug 2020-06-27 09:38:10: Received Zigbee message from '0x842e14fffe51c575', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,11]}' from endpoint 1 with groupID 0
debug 2020-06-27 09:38:10: No converter available for 'TS0121' with cluster 'seMetering' and type 'attributeReport' and data '{"currentSummDelivered":[0,11]}'
debug 2020-06-27 09:38:10: Received Zigbee message from '0x842e14fffe51c575', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,11]}' from endpoint 1 with groupID 0
debug 2020-06-27 09:38:10: No converter available for 'TS0121' with cluster 'seMetering' and type 'attributeReport' and data '{"currentSummDelivered":[0,11]}'
debug 2020-06-27 09:38:10: Received Zigbee message from '0x842e14fffe51c575', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,11]}' from endpoint 1 with groupID 0
debug 2020-06-27 09:38:10: No converter available for 'TS0121' with cluster 'seMetering' and type 'attributeReport' and data '{"currentSummDelivered":[0,11]}'
debug 2020-06-27 09:38:10: Received Zigbee message from '0x842e14fffe51c575', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,11]}' from endpoint 1 with groupID 0
debug 2020-06-27 09:38:10: No converter available for 'TS0121' with cluster 'seMetering' and type 'attributeReport' and data '{"currentSummDelivered":[0,11]}'
debug 2020-06-27 09:38:11: debug 2020-06-27 09:38:11: Received Zigbee message from '0x842e14fffe51c575', type 'read', cluster 'genTime', data '["localTime"]' from endpoint 1 with groupID 0
debug 2020-06-27 09:38:15: Received Zigbee message from '0x842e14fffe51c575', type 'attributeReport', cluster 'genBasic', data '{"65506":30,"appVersion":64}' from endpoint 1 with groupID 0
debug 2020-06-27 09:38:15: No converter available for 'TS0121' with cluster 'genBasic' and type 'attributeReport' and data '{"65506":30,"appVersion":64}'
debug 2020-06-27 09:38:15: Received Zigbee message from '0x842e14fffe51c575', type 'attributeReport', cluster 'genBasic', data '{"65503":"�ω&\u0013�ω&\u0012"}' from endpoint 1 with groupID 0
debug 2020-06-27 09:38:15: No converter available for 'TS0121' with cluster 'genBasic' and type 'attributeReport' and data '{"65503":"�ω&\u0013�ω&\u0012"}'`

milobt, 2020-06-27:
Do you have power metering working?
If so, using what definition?

mabosch74, 2020-06-27:
not yet. I take a look on it.

Hedda, 2021-01-28:
FYI, another relatively new TS0121 variant is this Lonsonho branded version which also features power/energy-metering support.

https://zigbee.blakadder.com/Lonsonho_TS0121.html

Hope to see full support for these Tuya smart plugs as there are few inexpensive Zigbee plugs featuring power/energy-metering.

Really like these two because power/energy-metering support and fact that they are both inexpensive and very small.

[image]

[image]

Hedda, 2021-01-28:
And this one from "Rehentele" looks to be yet another Tuya TS0121 variant with power-metering and energy-metering support:

https://www.aliexpress.com/item/4000449939329.html

Note that Rehentele sells a TS0121 version with power-metering as well as a TS0111 version without power-metering support.
[image]

[image]

[image]

Creo2005, 2021-04-23:
The same, but $14.39 - https://www.aliexpress.com/item/1005001929382211.html