Koenkk/zigbee2mqtt #3304: after updating E1524 remotes, battery dead in 24h

Edzilla2000, 2020-04-06:
# Bug Report
I updated several Ikea E1524 remotes. Since then, new batteries in the remote last about 24 hours.
I have tried changing the batteries in the remotes, with several brands and it still happens.

The firmware version ended up being 1.2.223 in one and 1.2.214 in the other.
They are both hardware version "1"

## Debug Info
zigbee2mqtt version: docker image koenkk/zigbee2mqtt:latest-dev
CC26X2R1 firmware version: 20200312

Fabiancrg, 2020-04-07:
Hello,

I also have an IKEA Tradfri E1524 but not too sure about the version, it should be the latest.
(no upgrade availble when I check).
I don't see where I canfind the current version, in debug mode I get this:
Got OTA request '{"fieldControl":1,"manufacturerCode":4476,"imageType":4545,"fileVersion":587286065}'

Is it the same version as yours ?

With this version I don't have any battery issue.

Edzilla2000, 2020-04-08:
For the version I'm using the value that zigbee2mqttAssistant returns.
The thing is, I have issues with only those remotes that I have updated. Every other is just fine and does not have massive battery usage

Edzilla2000, 2020-04-24:
After removing the two remotes and pairing them with the original Tradfri bridge, it says there was no update available.
I repaired it to Z2mqtt and I haven't had any issue since then.

closed 2020-04-24 by Edzilla2000