Koenkk/zigbee2mqtt #3306: Support EigenStone Thermostat S2

slydiman, 2020-04-07:
Could you add support the device EigenStone Thermostat S2 (KTWKQ03ES), please.
http://docs.opencloud.aqara.com/en/guideline/product-discription/#thermostat
I can provide any necessary information and help with testing/debugging.
Thanks in advance.

slydiman, 2020-04-07:
{"id":3,
"type":"Router",
"ieeeAddr":"0x00158d000XXXXXXX",
"nwkAddr":48239,
"manufId":4151,
"manufName":"LUMI",
"powerSource":"DC Source",
"modelId":"lumi.airrtc.tcpecn02",
"epList":[1],
"endpoints":{
"1":{"profId":260,
"epId":1,
"devId":257,
"inClusterList":[0,3,1,2,4,5,10,513,514,6,1024,1026,1029,1030,12],
"outClusterList":[25,10],
"clusters":{
"genBasic":{
"attributes":{"modelId":"lumi.airrtc.tcpecn02",
"manufacturerName":"LUMI",
"powerSource":4,
"zclVersion":0,
"appVersion":20,
"stackVersion":2,
"hwVersion":52,
"dateCode":"05-15-2019"}}},
"binds":[]}},
"appVersion":20,
"stackVersion":2,
"hwVersion":52,
"dateCode":"05-15-2019",
"zclVersion":0,
"interviewCompleted":true,
"meta":{},
"lastSeen":1586242356531}

slydiman, 2020-04-07:
I read https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
But I'm using Zigbee2mqtt Hass.io Add-on.
I have no idea where is the docker run command for Zigbee2mqtt and how to edit it.
Where can I find and how to edit 
node_modules/zigbee-herdsman-converters/devices.js
and
node_modules/zigbee-herdsman-converters/converters/fromZigbee.js
?
Please advise.

slydiman, 2020-04-30:
Power Off:

```
debug 2020-04-30 18:59:03: No converter available for 'KTWKQ03ES' with cluster 'genBasic' and type 'attributeReport' and data '{"65281":{"3":30,"5":58,"7":"0x0000000000000000","8":13332,"9":2816,"100":305140579,"105":0,"106":0,"108":1,"110":65534,"111":0,"155":0}}'
...
debug 2020-04-30 18:59:06: No converter available for 'KTWKQ03ES' with cluster 'genBasic' and type 'attributeReport' and data '{"65520":{"type":"Buffer","data":[170,128,5,209,135,0,134,32,0]}}'
debug 2020-04-30 18:59:06: No converter available for 'KTWKQ03ES' with cluster 'genBasic' and type 'attributeReport' and data '{"65520":{"type":"Buffer","data":[170,128,8,206,135,0,133,35,96,19,48,2]}}'
debug 2020-04-30 18:59:06: No converter available for 'KTWKQ03ES' with cluster 'genBasic' and type 'attributeReport' and data '{"61440":64168448}'
```
Power On:

```
debug 2020-04-30 18:59:24: No converter available for 'KTWKQ03ES' with cluster 'genBasic' and type 'attributeReport' and data '{"65520":{"type":"Buffer","data":[170,128,5,209,135,0,134,32,0]}}'
debug 2020-04-30 18:59:24: No converter available for 'KTWKQ03ES' with cluster 'genBasic' and type 'attributeReport' and data '{"65520":{"type":"Buffer","data":[170,128,8,206,135,0,133,35,99,19,32,17]}}'
debug 2020-04-30 18:59:24: No converter available for 'KTWKQ03ES' with cluster 'genBasic' and type 'attributeReport' and data '{"61440":64168448}'
...
debug 2020-04-30 18:59:29: No converter available for 'KTWKQ03ES' with cluster 'genBasic' and type 'attributeReport' and data '{"65520":{"type":"Buffer","data":[170,128,8,206,135,0,133,35,99,19,48,18]}}'
debug 2020-04-30 18:59:29: No converter available for 'KTWKQ03ES' with cluster 'genBasic' and type 'attributeReport' and data '{"61440":64168448}'
```

Change heat temperature from 19C to 20C:
(current room temperature is 24C)

```
debug 2020-04-30 18:59:40: No converter available for 'KTWKQ03ES' with cluster 'genTime' and type 'read' and data '["time"]'
debug 2020-04-30 18:59:40: No converter available for 'KTWKQ03ES' with cluster 'genBasic' and type 'attributeReport' and data '{"65520":{"type":"Buffer","data":[170,128,5,209,135,0,134,32,2]}}'
debug 2020-04-30 18:59:41: No converter available for 'KTWKQ03ES' with cluster 'genBasic' and type 'attributeReport' and data '{"65520":{"type":"Buffer","data":[170,128,5,209,135,0,134,32,2]}}'
debug 2020-04-30 18:59:45: No converter available for 'KTWKQ03ES' with cluster 'genBasic' and type 'attributeReport' and data '{"65520":{"type":"Buffer","data":[170,128,8,206,135,0,133,35,99,20,48,18]}}'
debug 2020-04-30 18:59:45: No converter available for 'KTWKQ03ES' with cluster 'genBasic' and type 'attributeReport' and data '{"61440":64168448}'
```

I see changes 19 > 20, but there are too much other unknown parameters.

Here is 
[Chinese documentation for a configuration menu](https://drive.google.com/drive/folders/1AdK2pJzL1aVZpxdHiORPDm-hOiVDpK_H)

kvvoff, 2020-05-19:
this device will arrive soon too, I would like to get support in z2m.

bluefoxlee, 2020-05-28:
Would love to see Zigbee2mqtt support although I am using S1 version. Will try to do some test when I get back to my house (working outside at moment.) 

BTW, S2 just got HomeKit support through Aqara gateway. Perhaps it could be a new solution for home automation (depends on platform).

slydiman, 2020-05-28:
> BTW, S2 just got HomeKit support through Aqara gateway. 
Aqara gateway does not support S2 at all, especially HomeKit version. S2 works with the Xiaomi gateway v2 only.

bluefoxlee, 2020-05-28:
> > BTW, S2 just got HomeKit support through Aqara gateway.
> 
> Aqara gateway does not support S2 at all, especially HomeKit version. S2 works with the Xiaomi gateway v2 only.

You might want to take  a look of the news and release note. Not sure which Aqara gateway are you referring to since there are just too many different version of Xiaomi/Aqara gateway. 

https://homekitnews.com/2020/05/27/aqara-hub-firmware-update-adds-homekit-functionality-for-more-devices/

slydiman, 2020-05-28:
I see, it is a new firmware. I tried Aqara hub 6 months ago and it did not support S2.

But let's discuss how to add S2 support to Zigbee2mqtt.

bluefoxlee, 2020-05-28:
> But let's discuss how to add S2 support to Zigbee2mqtt.

Yes. Would love to see S2 supported, and believe it will be same for S1 and CO2 version as well.
Thanks for your hard work. I can only test my S1 until few weeks later

Adminiuga, 2020-05-31:
have you tried dev version of HA Core?

slydiman, 2020-05-31:
No. Is dev version of HA Core already supported it somehow?

Adminiuga, 2020-05-31:
Doh, was posting to wrong issue. Disregard above comments

closed 2020-07-26 by stale[bot]

rfvermut, 2021-05-18:
> Here is [Chinese documentation for a configuration menu](https://drive.google.com/drive/folders/1AdK2pJzL1aVZpxdHiORPDm-hOiVDpK_H)

I tried my best to provide translation for that via Google Translate https://gist.github.com/rfvermut/ca72fbacd459d1d1e9cab253afab4bd9

dennyxiao, 2021-09-11:
> Could you add support the device EigenStone Thermostat S2 (KTWKQ03ES), please.
> http://docs.opencloud.aqara.com/en/guideline/product-discription/#thermostat
> I can provide any necessary information and help with testing/debugging.
> Thanks in advance.

I need this device work with Z2M. Any further support  possible ?

dennyxiao, 2021-09-24:
Any further information.

santiacho-jt, 2024-08-15:
Any update on this?