Koenkk/zigbee2mqtt #3308: dispositivo Blaupunkt SCM-S1

zhapwey, 2020-04-07:
# Bug Report
## What happened
I have a device like this, it works correctly but the network map returns it to me as end device and not as rooter being a device connected to the network.
nor do I get when I run the motor by the external buttons, to send me an update of the value of the blind.
## What did you expect to happen

Send status update with the use of external controls

## How to reproduce it (minimal and precise)
Use external controls with to device 
## Debug Info
zigbee2mqtt version: 1.11.0
CC253X firmware version: :20190608

IVI053, 2020-04-07:
It's not only in the network map and the reason is, that it ist not operating as a router afaik

zhapwey, 2020-04-08:
And that does not update values ​​when using external buttons does not have a solution either?

Because it is a shame it works very well but that when you raise the blind from the button it is not updated it is a problem

Best Regards

Koenkk, 2020-04-09:
Try updating the definition inside `devices.js` to

```js
{
    zigbeeModel: ['SCM-2_00.00.03.15', 'SCM-R_00.00.03.15TC', 'SCM_00.00.03.14TC', 'SCM_00.00.03.05TC'],
    model: 'SCM-S1',
    vendor: 'Blaupunkt',
    description: 'Roller shutter',
    supports: 'open/close',
    fromZigbee: [fz.cover_position_via_brightness, fz.cover_state_via_onoff],
    toZigbee: [tz.cover_position_via_brightness, tz.cover_open_close_via_brightness],
    meta: {configureKey: 1},
    configure: async (device, coordinatorEndpoint) => {
        const endpoint = device.getEndpoint(1);
        await bind(endpoint, coordinatorEndpoint, ['genLevelCtrl']);
        await configureReporting.brightness(endpoint);
    },
},
```

How to get to devices.js: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

zhapwey, 2020-04-13:
I have tested the configuration that you send me and it works perfectly, now it reports when orders are sent to it by zigbee or by the keypad. Great job!!!

Can the library be updated with this device version? I don't know how to do a PR for you, and I have another team I wish I could add to your excellent work.

Best regards

Koenkk, 2020-04-13:
Done, will be available in the dev branch in a few hours. I assume this can be closed now.

closed 2020-04-13 by Koenkk