Koenkk/zigbee2mqtt #3309: Xiaomi Honeywell gas detector JTQJ-BF-01LM/BW measure units?

jrhbcn, 2020-04-07:
Hi,

I recently paired a Xiaomi gas detector [JTQJ-BF-01LM/BW](https://www.zigbee2mqtt.io/devices/JTQJ-BF-01LM_BW.html#xiaomi-jtqj-bf-01lmbw) to zigbee2mqtt. Everything went super-smoothly as always. Again, thanks for this very nice software.

Does anybody know the measure units that the gas_density topic reports in and if there are any low-high limits to these values? 

Regards,

McGiverGim, 2020-04-07:
I don't know if this will help you, but I have the Smoke detector, that is very similar to yours and the smoke_density moves from 0 to 255 (that are 8 bits, so maybe this is how it has been designed). I have never seen more than 255.
I can't say nothing about the units...

jrhbcn, 2020-04-07:
Thanks for the info @McGiverGim ,

In that case it makes more sense I think to report vales in `%`. I can do that in the mqtt sensor definition I guess something like:

```
    unit_of_measurement: "%"
    value_template: "{{ ( value_json.gas_density | float / 255 * 100 ) | round(0) }}"
```

Is it possible to do something similar using the mqtt automatic discovery?

closed 2020-06-13 by stale[bot]