Koenkk/zigbee2mqtt #3310: Request support for eWlink SA-003 Zigbee AC inline switch

netweaver1970, 2020-04-07:
Hi,
Can you provide support for this simple AC inline Zigbee switch?

https://www.aliexpress.com/item/32982997140.html

Thanks,
Geert

`zigbee2mqtt:info  2020-04-07 11:25:05: Device '0x00124b001cd60814' joined
zigbee2mqtt:info  2020-04-07 11:25:05: Starting interview of '0x00124b001cd60814'
zigbee2mqtt:info  2020-04-07 11:25:05: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x00124b001cd60814"}}'
zigbee2mqtt:info  2020-04-07 11:25:05: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x00124b001cd60814"}}'
zigbee2mqtt:debug 2020-04-07 11:25:06: Received Zigbee message from '0x00124b001cd60814', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-04-07 11:25:06: Skipping message, modelID is undefined and still interviewing
zigbee2mqtt:debug 2020-04-07 11:25:06: Received Zigbee message from '0x00124b001cd60814', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-04-07 11:25:06: Skipping message, modelID is undefined and still interviewing
zigbee2mqtt:debug 2020-04-07 11:25:06: Received Zigbee message from '0x00124b001cd60814', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-04-07 11:25:06: Skipping message, modelID is undefined and still interviewing
zigbee2mqtt:debug 2020-04-07 11:25:06: Received Zigbee message from '0x00124b001cd60814', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-04-07 11:25:06: Skipping message, modelID is undefined and still interviewing
zigbee2mqtt:debug 2020-04-07 11:25:06: Received Zigbee message from '0x00124b001cd60814', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-04-07 11:25:06: Skipping message, modelID is undefined and still interviewing
zigbee2mqtt:debug 2020-04-07 11:25:06: Received Zigbee message from '0x00124b001cd60814', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-04-07 11:25:06: Skipping message, modelID is undefined and still interviewing
zigbee2mqtt:debug 2020-04-07 11:25:06: Received Zigbee message from '0x00124b001cd60814', type 'readResponse', cluster 'genBasic', data '{"modelId":"SA-003-Zigbee","manufacturerName":"eWeLink"}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-04-07 11:25:06: No converter available for 'SA-003-Zigbee' with cluster 'genBasic' and type 'readResponse' and data '{"modelId":"SA-003-Zigbee","manufacturerName":"eWeLink"}'
zigbee2mqtt:debug 2020-04-07 11:25:06: Setup reporting for '0x00124b001cd60814' - 1 - genOnOff
zigbee2mqtt:info  2020-04-07 11:25:06: MQTT publish: topic 'homeassistant/switch/0x00124b001cd60814/switch/config', payload '{"payload_off":"OFF","payload_on":"ON","value_template":"{{ value_json.state }}","command_topic":"zigbee2mqtt/0x00124b001cd60814/set","state_topic":"zigbee2mqtt/0x00124b001cd60814","json_attributes_topic":"zigbee2mqtt/0x00124b001cd60814","name":"0x00124b001cd60814_switch","unique_id":"0x00124b001cd60814_switch_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x00124b001cd60814"],"name":"0x00124b001cd60814","sw_version":"Zigbee2mqtt 1.12.0-dev","model":"Zigbee smart plug (SA-003-Zigbee)","manufacturer":"eWeLink"},"availability_topic":"zigbee2mqtt/bridge/state"}'
zigbee2mqtt:info  2020-04-07 11:25:06: MQTT publish: topic 'homeassistant/sensor/0x00124b001cd60814/linkquality/config', payload '{"icon":"mdi:signal","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}","state_topic":"zigbee2mqtt/0x00124b001cd60814","json_attributes_topic":"zigbee2mqtt/0x00124b001cd60814","name":"0x00124b001cd60814_linkquality","unique_id":"0x00124b001cd60814_linkquality_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x00124b001cd60814"],"name":"0x00124b001cd60814","sw_version":"Zigbee2mqtt 1.12.0-dev","model":"Zigbee smart plug (SA-003-Zigbee)","manufacturer":"eWeLink"},"availability_topic":"zigbee2mqtt/bridge/state"}'
zigbee2mqtt:debug 2020-04-07 11:25:06: Device '0x00124b001cd60814' announced itself
zigbee2mqtt:info  2020-04-07 11:25:06: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"0x00124b001cd60814"}}'
zigbee2mqtt:debug 2020-04-07 11:25:06: Received Zigbee message from '0x00124b001cd60814', type 'readResponse', cluster 'genBasic', data '{"powerSource":1,"zclVersion":1}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-04-07 11:25:06: No converter available for 'SA-003-Zigbee' with cluster 'genBasic' and type 'readResponse' and data '{"powerSource":1,"zclVersion":1}'
zigbee2mqtt:error 2020-04-07 11:25:21: Failed to setup reporting for '0x00124b001cd60814' - Error: ConfigureReporting 0x00124b001cd60814/1 genOnOff([{"attribute":"onOff","minimumReportInterval":0,"maximumReportInterval":300,"reportableChange":0}], {"timeout":6000,"manufacturerCode":null,"disableDefaultResponse":true}) failed (Error: Timeout - 59031 - 1 - 15 - 6 - 7 after 6000ms)
    at Endpoint.<anonymous> (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:326:23)
    at Generator.throw (<anonymous>)
    at rejected (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65)
zigbee2mqtt:debug 2020-04-07 11:25:21: Received Zigbee message from '0x00124b001cd60814', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-04-07 11:25:21: No converter available for 'SA-003-Zigbee' with cluster 'genBasic' and type 'readResponse' and data '{}'
zigbee2mqtt:error 2020-04-07 11:25:21: Failed to interview '0x00124b001cd60814', device has not successfully been paired
zigbee2mqtt:info  2020-04-07 11:25:21: Configuring '0x00124b001cd60814'
zigbee2mqtt:info  2020-04-07 11:25:21: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_failed","meta":{"friendly_name":"0x00124b001cd60814"}}'
zigbee2mqtt:info  2020-04-07 11:25:21: Successfully configured '0x00124b001cd60814'
zigbee2mqtt:debug 2020-04-07 11:29:07: Saving state to file /app/data/state.json
zigbee2mqtt:debug 2020-04-07 11:30:05: Received Zigbee message from '0x00124b001cd60814', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-04-07 11:30:05: MQTT publish: topic 'zigbee2mqtt/0x00124b001cd60814', payload '{"state":"ON","linkquality":212}'
zigbee2mqtt:debug 2020-04-07 11:30:05: Received Zigbee message from '0x00124b001cd60814', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-04-07 11:30:05: Received Zigbee message from '0x00124b001cd60814', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-04-07 11:30:05: Received Zigbee message from '0x00124b001cd60814', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-04-07 11:30:05: Received Zigbee message from '0x00124b001cd60814', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-04-07 11:30:05: Received Zigbee message from '0x00124b001cd60814', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0`

Koenkk, 2020-04-07:
Its already supported?

netweaver1970, 2020-04-07:
Indeed I get the LQI and Switch entities in HA but why do I get Interview failed errors (and others) in the log:

`endpoint 1 with groupID 0 zigbee2mqtt:debug 2020-04-07 11:25:21: No converter available for 'SA-003-Zigbee' with cluster 'genBasic' and type 'readResponse' and data '{}' zigbee2mqtt:error 2020-04-07 11:25:21: Failed to interview '0x00124b001cd60814', device has not successfully been paired zigbee2mqtt:info 2020-04-07 11:25:21: Configuring '0x00124b001cd60814' zigbee2mqtt:info 2020-04-07 11:25:21: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_failed","meta":{"friendly_name":"0x00124b001cd60814"}}' zigbee2mqtt:info 2020-04-07 11:25:21: Successfully configured '0x00124b001cd60814'`

Is it just a visual issue? At least it's confusing.

Thanks.

Koenkk, 2020-04-08:
Regarding interview failed, can you update to zigbee2mqtt 1.12.1 and provide the logging when pairing the device and running with herdsman debug logging?

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

netweaver1970, 2020-04-09:
log Details on trying to remove the Switch attached. I heard the relay
click a few times when executing the removal request via Z2M Assistant but
the Z2MA GUI gave an error.
It does seem that *something* is removed concerning the switch but I still
see a trace of it in configuration.yaml and in the Z2MA GUI.
in HomeAssistant I can still control the switch as before, so even the
removal didn't work fine. I think I need a clean state before I should
capture the re-joining.

btw, I have retain on for all devices (on header level).
The machine with Z2M and HA on, is a lubuntu VM in virtualbox, running on
Win7 (my main work laptop OS). All linux components (HA, Z2M, Z2MA,
Node-Red etc...) are orchestrated via docker-compose.

Thanks for looking into it
Error. An error occurred while processing your request.

*Request ID:* |9a54806f-43894bd3d8d9d029.
Development Mode

Swapping to *Development* environment will display more detailed
information about the error that occurred.

*The Development environment shouldn't be enabled for deployed
applications.* It can result in displaying sensitive information from
exceptions to end users. For local debugging, enable the *Development*
environment by setting the *ASPNETCORE_ENVIRONMENT* environment variable to
*Development* and restarting the app.
[image: image.png]

netweaver1970, 2020-04-09:
I'm on z2m dev channel, so after later fix you made things seems to be better.
Closing the issue.

Thanks.

zigbee2mqtt:info  2020-04-09 16:32:08: Device 'eWeLink_14' joined
zigbee2mqtt:info  2020-04-09 16:32:08: Starting interview of 'eWeLink_14'
zigbee2mqtt:info  2020-04-09 16:32:08: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"eWeLink_14"}}'
zigbee2mqtt:info  2020-04-09 16:32:08: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"eWeLink_14"}}'
zigbee2mqtt:debug 2020-04-09 16:32:09: Received Zigbee message from 'eWeLink_14', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-04-09 16:32:09: Skipping message, modelID is undefined and still interviewing
zigbee2mqtt:debug 2020-04-09 16:32:09: Received Zigbee message from 'eWeLink_14', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-04-09 16:32:09: Skipping message, modelID is undefined and still interviewing
zigbee2mqtt:debug 2020-04-09 16:32:09: Received Zigbee message from 'eWeLink_14', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-04-09 16:32:09: Skipping message, modelID is undefined and still interviewing
zigbee2mqtt:debug 2020-04-09 16:32:09: Received Zigbee message from 'eWeLink_14', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-04-09 16:32:09: Skipping message, modelID is undefined and still interviewing
zigbee2mqtt:debug 2020-04-09 16:32:09: Received Zigbee message from 'eWeLink_14', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-04-09 16:32:09: Skipping message, modelID is undefined and still interviewing
zigbee2mqtt:debug 2020-04-09 16:32:09: Received Zigbee message from 'eWeLink_14', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-04-09 16:32:09: Skipping message, modelID is undefined and still interviewing
zigbee2mqtt:debug 2020-04-09 16:32:09: Device 'eWeLink_14' announced itself
zigbee2mqtt:info  2020-04-09 16:32:09: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"eWeLink_14"}}'
zigbee2mqtt:debug 2020-04-09 16:32:09: Received Zigbee message from 'eWeLink_14', type 'readResponse', cluster 'genBasic', data '{"modelId":"SA-003-Zigbee"}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-04-09 16:32:09: No converter available for 'SA-003-Zigbee' with cluster 'genBasic' and type 'readResponse' and data '{"modelId":"SA-003-Zigbee"}'
zigbee2mqtt:debug 2020-04-09 16:32:09: Received Zigbee message from 'eWeLink_14', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"eWeLink"}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-04-09 16:32:09: No converter available for 'SA-003-Zigbee' with cluster 'genBasic' and type 'readResponse' and data '{"manufacturerName":"eWeLink"}'
zigbee2mqtt:debug 2020-04-09 16:32:09: Received Zigbee message from 'eWeLink_14', type 'readResponse', cluster 'genBasic', data '{"powerSource":1}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-04-09 16:32:09: No converter available for 'SA-003-Zigbee' with cluster 'genBasic' and type 'readResponse' and data '{"powerSource":1}'
zigbee2mqtt:debug 2020-04-09 16:32:09: Received Zigbee message from 'eWeLink_14', type 'readResponse', cluster 'genBasic', data '{"zclVersion":1}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-04-09 16:32:09: No converter available for 'SA-003-Zigbee' with cluster 'genBasic' and type 'readResponse' and data '{"zclVersion":1}'
zigbee2mqtt:debug 2020-04-09 16:32:10: Received Zigbee message from 'eWeLink_14', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-04-09 16:32:10: No converter available for 'SA-003-Zigbee' with cluster 'genBasic' and type 'readResponse' and data '{}'
zigbee2mqtt:debug 2020-04-09 16:32:10: Received Zigbee message from 'eWeLink_14', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-04-09 16:32:10: No converter available for 'SA-003-Zigbee' with cluster 'genBasic' and type 'readResponse' and data '{}'
zigbee2mqtt:debug 2020-04-09 16:32:10: Received Zigbee message from 'eWeLink_14', type 'readResponse', cluster 'genBasic', data '{"hwVersion":1}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-04-09 16:32:10: No converter available for 'SA-003-Zigbee' with cluster 'genBasic' and type 'readResponse' and data '{"hwVersion":1}'
zigbee2mqtt:debug 2020-04-09 16:32:10: Received Zigbee message from 'eWeLink_14', type 'readResponse', cluster 'genBasic', data '{"dateCode":"20190907-V102"}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-04-09 16:32:10: No converter available for 'SA-003-Zigbee' with cluster 'genBasic' and type 'readResponse' and data '{"dateCode":"20190907-V102"}'
zigbee2mqtt:debug 2020-04-09 16:32:10: Received Zigbee message from 'eWeLink_14', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
[... cut ...]
zigbee2mqtt:info  2020-04-09 16:32:10: Configuring 'eWeLink_14'
zigbee2mqtt:info  2020-04-09 16:32:10: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"eWeLink_14","model":"SA-003-Zigbee","vendor":"eWeLink","description":"Zigbee smart plug","supported":true}}'
zigbee2mqtt:info  2020-04-09 16:32:10: Successfully configured 'eWeLink_14'
zigbee2mqtt:debug 2020-04-09 16:32:12: Saving state to file /app/data/state.json

closed 2020-04-09 by netweaver1970