Koenkk/zigbee2mqtt #3311: Request support for Lonsonho/"CR Smart Home" Motorized Valve TS0001

netweaver1970, 2020-04-07:
Hi,
Can you complete the support for the below valve?
The proper Entities in HomeAssistant are available already for LQI and Switch, also the control of the valve works fine through HA, although the on/off seems reversed as closed/open.
But as per log, maybe some other things extra need setting up to complete the support.

https://www.aliexpress.com/item/4000033754819.html

Thanks,
Geert

`zigbee2mqtt:info  2020-04-07 12:13:59: Device '0x086bd7fffefb662d' joined

zigbee2mqtt:info  2020-04-07 12:13:59: Starting interview of '0x086bd7fffefb662d'

zigbee2mqtt:info  2020-04-07 12:13:59: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x086bd7fffefb662d"}}'

zigbee2mqtt:info  2020-04-07 12:13:59: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x086bd7fffefb662d"}}'

zigbee2mqtt:debug 2020-04-07 12:14:00: Device '0x086bd7fffefb662d' announced itself

zigbee2mqtt:info  2020-04-07 12:14:00: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"0x086bd7fffefb662d"}}'

zigbee2mqtt:debug 2020-04-07 12:14:00: Received Zigbee message from '0x086bd7fffefb662d', type 'readResponse', cluster 'genBasic', data '{"modelId":"TS0001","manufacturerName":"TUYATEC-pf3wdnnk"}' from endpoint 1 with groupID 0

zigbee2mqtt:debug 2020-04-07 12:14:00: No converter available for 'TS0001' with cluster 'genBasic' and type 'readResponse' and data '{"modelId":"TS0001","manufacturerName":"TUYATEC-pf3wdnnk"}'

zigbee2mqtt:debug 2020-04-07 12:14:00: Setup reporting for '0x086bd7fffefb662d' - 1 - genOnOff

zigbee2mqtt:info  2020-04-07 12:14:00: MQTT publish: topic 'homeassistant/switch/0x086bd7fffefb662d/switch/config', payload '{"payload_off":"OFF","payload_on":"ON","value_template":"{{ value_json.state }}","command_topic":"zigbee2mqtt/0x086bd7fffefb662d/set","state_topic":"zigbee2mqtt/0x086bd7fffefb662d","json_attributes_topic":"zigbee2mqtt/0x086bd7fffefb662d","name":"0x086bd7fffefb662d_switch","unique_id":"0x086bd7fffefb662d_switch_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x086bd7fffefb662d"],"name":"0x086bd7fffefb662d","sw_version":"Zigbee2mqtt 1.12.0-dev","model":"Valve control (TS0001)","manufacturer":"CR Smart Home"},"availability_topic":"zigbee2mqtt/bridge/state"}'

zigbee2mqtt:info  2020-04-07 12:14:00: MQTT publish: topic 'homeassistant/sensor/0x086bd7fffefb662d/linkquality/config', payload '{"icon":"mdi:signal","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}","state_topic":"zigbee2mqtt/0x086bd7fffefb662d","json_attributes_topic":"zigbee2mqtt/0x086bd7fffefb662d","name":"0x086bd7fffefb662d_linkquality","unique_id":"0x086bd7fffefb662d_linkquality_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x086bd7fffefb662d"],"name":"0x086bd7fffefb662d","sw_version":"Zigbee2mqtt 1.12.0-dev","model":"Valve control (TS0001)","manufacturer":"CR Smart Home"},"availability_topic":"zigbee2mqtt/bridge/state"}'

zigbee2mqtt:debug 2020-04-07 12:14:01: Received Zigbee message from '0x086bd7fffefb662d', type 'readResponse', cluster 'genBasic', data '{"powerSource":1,"zclVersion":3}' from endpoint 1 with groupID 0

zigbee2mqtt:debug 2020-04-07 12:14:01: No converter available for 'TS0001' with cluster 'genBasic' and type 'readResponse' and data '{"powerSource":1,"zclVersion":3}'

zigbee2mqtt:debug 2020-04-07 12:14:07: Saving state to file /app/data/state.json

zigbee2mqtt:error 2020-04-07 12:14:16: Failed to setup reporting for '0x086bd7fffefb662d' - Error: ConfigureReporting 0x086bd7fffefb662d/1 genOnOff([{"attribute":"onOff","minimumReportInterval":0,"maximumReportInterval":300,"reportableChange":0}], {"timeout":6000,"manufacturerCode":null,"disableDefaultResponse":true}) failed (Error: Timeout - 21627 - 1 - 19 - 6 - 7 after 6000ms)

    at Endpoint.<anonymous> (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:326:23)

    at Generator.throw (<anonymous>)

    at rejected (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65)

zigbee2mqtt:debug 2020-04-07 12:14:16: Received Zigbee message from '0x086bd7fffefb662d', type 'readResponse', cluster 'genBasic', data '{"appVersion":64}' from endpoint 1 with groupID 0

zigbee2mqtt:debug 2020-04-07 12:14:16: No converter available for 'TS0001' with cluster 'genBasic' and type 'readResponse' and data '{"appVersion":64}'

zigbee2mqtt:error 2020-04-07 12:14:16: Failed to interview '0x086bd7fffefb662d', device has not successfully been paired

zigbee2mqtt:info  2020-04-07 12:14:16: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_failed","meta":{"friendly_name":"0x086bd7fffefb662d"}}'`

Koenkk, 2020-04-08:
Seems that it doesn't support reporting. To confirm can you set `report: false` in your `configuration.yaml`, if it works I can prevent this device from reporting being setup.

netweaver1970, 2020-04-08:
Sure, I upgraded to 1.12.1, remarked the reporting in configuration.yaml, removed the valve actuator, restarted z2m and re-paired:

The dump on the reporting request seems to be gone. So far so good. It clearly doesn't support it. Still 'interview failed' messages though.

The valve actuator itself seemingly works fine, it just looks messy in the logs when pairing and when having reporting on.

zigbee2mqtt:info  2020-04-08 18:19:22: Device '0x086bd7fffefb662d' joined
zigbee2mqtt:info  2020-04-08 18:19:22: Starting interview of '0x086bd7fffefb662d'
zigbee2mqtt:info  2020-04-08 18:19:22: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x086bd7fffefb662d"}}'
zigbee2mqtt:info  2020-04-08 18:19:22: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x086bd7fffefb662d"}}'
zigbee2mqtt:debug 2020-04-08 18:19:23: Device '0x086bd7fffefb662d' announced itself
zigbee2mqtt:info  2020-04-08 18:19:23: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"0x086bd7fffefb662d"}}'
zigbee2mqtt:debug 2020-04-08 18:19:23: Received Zigbee message from '0x086bd7fffefb662d', type 'readResponse', cluster 'genBasic', data '{"modelId":"TS0001","manufacturerName":"TUYATEC-pf3wdnnk"}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-04-08 18:19:23: No converter available for 'TS0001' with cluster 'genBasic' and type 'readResponse' and data '{"modelId":"TS0001","manufacturerName":"TUYATEC-pf3wdnnk"}'
zigbee2mqtt:debug 2020-04-08 18:19:23: Received Zigbee message from '0x086bd7fffefb662d', type 'readResponse', cluster 'genBasic', data '{"powerSource":1,"zclVersion":3}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-04-08 18:19:23: No converter available for 'TS0001' with cluster 'genBasic' and type 'readResponse' and data '{"powerSource":1,"zclVersion":3}'
zigbee2mqtt:debug 2020-04-08 18:19:23: Received Zigbee message from '0x086bd7fffefb662d', type 'readResponse', cluster 'genBasic', data '{"appVersion":64}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-04-08 18:19:23: No converter available for 'TS0001' with cluster 'genBasic' and type 'readResponse' and data '{"appVersion":64}'
zigbee2mqtt:error 2020-04-08 18:19:23: Failed to interview '0x086bd7fffefb662d', device has not successfully been paired
zigbee2mqtt:info  2020-04-08 18:19:23: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_failed","meta":{"friendly_name":"0x086bd7fffefb662d"}}'

Koenkk, 2020-04-09:
@netweaver1970 could you provide the herdsman debug logging when pairing the device?

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

netweaver1970, 2020-04-09:
also this pairing went smooth now.
Closing the issue.
Thanks

zigbee2mqtt:info  2020-04-09 16:37:55: Device '0x086bd7fffefb662d' joined
zigbee2mqtt:info  2020-04-09 16:37:55: Starting interview of '0x086bd7fffefb662d'
zigbee2mqtt:info  2020-04-09 16:37:55: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x086bd7fffefb662d"}}'
zigbee2mqtt:info  2020-04-09 16:37:55: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x086bd7fffefb662d"}}'
zigbee2mqtt:debug 2020-04-09 16:37:56: Device '0x086bd7fffefb662d' announced itself
zigbee2mqtt:info  2020-04-09 16:37:56: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"0x086bd7fffefb662d"}}'
zigbee2mqtt:debug 2020-04-09 16:37:56: Received Zigbee message from '0x086bd7fffefb662d', type 'readResponse', cluster 'genBasic', data '{"modelId":"TS0001"}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-04-09 16:37:56: No converter available for 'TS0001' with cluster 'genBasic' and type 'readResponse' and data '{"modelId":"TS0001"}'
zigbee2mqtt:debug 2020-04-09 16:37:56: Received Zigbee message from '0x086bd7fffefb662d', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"TUYATEC-pf3wdnnk"}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-04-09 16:37:56: No converter available for 'TS0001' with cluster 'genBasic' and type 'readResponse' and data '{"manufacturerName":"TUYATEC-pf3wdnnk"}'
zigbee2mqtt:debug 2020-04-09 16:37:56: Received Zigbee message from '0x086bd7fffefb662d', type 'readResponse', cluster 'genBasic', data '{"powerSource":1}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-04-09 16:37:56: No converter available for 'TS0001' with cluster 'genBasic' and type 'readResponse' and data '{"powerSource":1}'
zigbee2mqtt:debug 2020-04-09 16:37:56: Received Zigbee message from '0x086bd7fffefb662d', type 'readResponse', cluster 'genBasic', data '{"zclVersion":3}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-04-09 16:37:56: No converter available for 'TS0001' with cluster 'genBasic' and type 'readResponse' and data '{"zclVersion":3}'
zigbee2mqtt:debug 2020-04-09 16:37:56: Received Zigbee message from '0x086bd7fffefb662d', type 'readResponse', cluster 'genBasic', data '{"appVersion":64}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-04-09 16:37:56: No converter available for 'TS0001' with cluster 'genBasic' and type 'readResponse' and data '{"appVersion":64}'
zigbee2mqtt:debug 2020-04-09 16:37:57: Received Zigbee message from '0x086bd7fffefb662d', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-04-09 16:37:57: No converter available for 'TS0001' with cluster 'genBasic' and type 'readResponse' and data '{}'
zigbee2mqtt:debug 2020-04-09 16:37:57: Received Zigbee message from '0x086bd7fffefb662d', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-04-09 16:37:57: No converter available for 'TS0001' with cluster 'genBasic' and type 'readResponse' and data '{}'
zigbee2mqtt:debug 2020-04-09 16:37:57: Received Zigbee message from '0x086bd7fffefb662d', type 'readResponse', cluster 'genBasic', data '{"dateCode":"20181219"}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-04-09 16:37:57: No converter available for 'TS0001' with cluster 'genBasic' and type 'readResponse' and data '{"dateCode":"20181219"}'
zigbee2mqtt:debug 2020-04-09 16:37:57: Received Zigbee message from '0x086bd7fffefb662d', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-04-09 16:37:57: No converter available for 'TS0001' with cluster 'genBasic' and type 'readResponse' and data '{}'
zigbee2mqtt:info  2020-04-09 16:37:57: Successfully interviewed '0x086bd7fffefb662d', device has successfully been paired
zigbee2mqtt:info  2020-04-09 16:37:57: Device '0x086bd7fffefb662d' is supported, identified as: CR Smart Home Valve control (TS0001)
zigbee2mqtt:info  2020-04-09 16:37:57: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"0x086bd7fffefb662d","model":"TS0001","vendor":"CR Smart Home","description":"Valve control","supported":true}}'

closed 2020-04-09 by netweaver1970

netweaver1970, 2020-04-09:
PS. can you still make sure this device doesn't complain when having reporting on ?
Thanks.

Koenkk, 2020-04-09:
Done (in latest dev branch).