Koenkk/zigbee2mqtt #3314: Zigbee2mqtt unresponsive after a few hours.

Huppeltje16, 2020-04-07:
Hi,

After running Zigbee2mqtt 1.4.0 with a CC2531, 5 OSRAM Smart+ plugs and 3 sunricher LED dimmers for about a year without any problems, I upgraded my Domoticz install and had to upgrade to a newer version.
I upgraded to version 1.12.0 with the latest firmware version on the CC2531, and had to reset my zigbee network. 
Now Zigbee2mqtt becomes unresponsive after a few hours. Reading the network still work, according to the log and by publishing a zigbee2mqtt/bridge/networkmap message and the log shows pings and traffic, but set/state messages have no effect.
Rebooting the pi does not help.
The only way to solve this is by stopping Zigbee2mqtt, unplugging the CC2531, plugging it in again and then starting Zigbee2mqtt, 
I've done a DEBUG=* npm start during the time the network didn't respond. Part of the output is below.

## Debug Info
zigbee2mqtt version: 1.12.0
CC253X firmware version:  Coordinator firmware version: '{"type":"zStack30x","meta":{"transportrev":2,"product":2,"majorrel":2,"minorrel":7,"maintrel":2,"revision":20190425}}'

[errlog.txt](https://github.com/Koenkk/zigbee2mqtt/files/4446385/errlog.txt)

Koenkk, 2020-04-08:
It seems that the adapter is disconnected from your system: `serialport/stream binding.read error Error: Port is not open`. Could it be a power problem? I would also suggest updating to the latest firmware: https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing

Huppeltje16, 2020-04-09:
Hi Koen,
Thanks for the reply. I will install this firmware. I thought I already had the latest, but this is in another directory.
Can it be that the adapter gets unresponsive now because of a scheduled reboot of my Pi, where the USB doesn't get a real powerdown because my RFLink keeps it powered through its own powersupply? Or is there a greater probability that it gets unresponsive and doesn't reset because the power stays on?

Kind regards, 
Petro Boef

Koenkk, 2020-04-10:
Does it keep working without these reboots?

Huppeltje16, 2020-04-10:
Hi Koen,

It’s better, but not completely stable.
I flashed the coordinator, which created new instabilities. Now I have problems reaching some devices and get LQI errors when I try to graph my network with Graphviz.
Initially all devices were unreachable, after deleting the database and repairing them, two devices keep giving these errors.

What can this be?

Kind Regards,

Petro Boef

De informatie in dit bericht is uitsluitend bestemd voor de geadresseerde. Aan dit bericht en de bijlagen kunnen geen rechten worden ontleend. Heeft u deze e-mail onbedoeld ontvangen? Dan verzoeken wij u het te vernietigen en de afzender te informeren. Openbaar maken, kopiëren en verspreiden van deze e-mail of informatie uit deze e-mail is alleen toegestaan met voorafgaande schriftelijke toestemming van de afzender. Het Radboudumc staat geregistreerd bij de Kamer van Koophandel in het handelsregister onder nummer 41055629.

The content of this message is intended solely for the addressee. No rights can be derived from this message or its attachments. If you are not the intended recipient, we kindly request you to delete the message and inform the sender. It is strictly prohibited to disclose, copy or distribute this email or the information inside it, without a written consent from the sender. Radboud university medical center is registered with the Dutch Chamber of Commerce trade register with number 41055629.

Huppeltje16, 2020-04-10:
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/4463112/log.txt)
Hi Koen,
Today started relatively good with my implementation.
The devices I added yesterday were still responsive, so I added the rest of them.
Everything seemed to be working, but this evening none of the devices respond to any commands.
Graphing shows 6 of the 8 devices, but none of them responds to any of the commands.
I added the log from today.

Koenkk, 2020-04-11:
I see that you are using the zstack 3 firwmware, note that this is not a recommended firmware (https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator#im-a-zigbee2mqtt-user-which-one-should-i-use). I would suggest using the 1.2 source routing firmware: https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing.  Switching to this firmware requires repairing all your devices.

In your log I see issues like 

```
debug 2020-04-10 21:26:17: Received MQTT message on 'zigbee2mqtt/0x000d6f0013f05169/set/state' with data 'on'
info  2020-04-10 21:26:17: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"entity_not_found","message":{"friendly_name":"0x000d6f0013f05169"}}'
error 2020-04-10 21:26:17: Entity '0x000d6f0013f05169' is unknown
```

These are caused when the device is not in your `configuration.yaml`.

Huppeltje16, 2020-04-11:
Hi Koen,

Thanks that you're actively monitoring this issue.
I've just replaced the firmware with the SR version of the 1.2 Home stack. I selected the SR because you suggested it, although I only have 8 devices and am not planning to expand my zigbee network. 
I'll see if my devices keep responding the coming days. I'll let you know what happens.

Kind Regards,

Petro

Huppeltje16, 2020-04-11:
Hi Koen,
Trouble again. One of the Osram Smart+ plugs (the one I use for my garden lighting) functioned perfectly upto just a few minutes ago. 
I tried switching it off, which succeeded, but switching it on didn't work anymore. The log shows messages about timeouts from then on.
I especially placed another Smart+ plug somewhere in the middle between this plug and the coördinator because I had my doubts about the range of these devices. The maximum distance is about 6 metres.
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/4465243/log.txt)

Koenkk, 2020-04-12:
Look like interference/range issues.

- Make sure the CC2531 is connected through a USB extension cable.
- Try to reduce wifi interference

More information: https://www.zigbee2mqtt.io/how_tos/how_to_improve_network_range.html

fcasarsa, 2020-04-12:
I am experiencing the same issue.

First I was using a CC2531 directly connected to a PC, then I moved it to a raspberry 1 and connect to it with ser2net, then using a rpi3, last i put a USB extension with a ferritis to reduce interference.

It happens quite often while restarting zigbee2mqtt on the pc. On the RPI the device is still plugged but unresponsive.

Could be?

closed 2020-06-20 by stale[bot]