Koenkk/zigbee2mqtt #3317: Unable to pair Bosch ISW-ZPR1-WP13 anymore

Fabiancrg, 2020-04-08:
# Bug Report
## What happened
Since migration from CC2531 to CC2652R1 I have some issues with this device, it worked during a couple of days and suddenly was stucked with "occupancy":true.
To fix this I had to remove the device and add it again.
Today, I upgraded to Z2M 1.12.1 and as the occupency sensor was again stucked, I removed it and tried to pair it again. But now I can't pair it anymore.

## What did you expect to happen

Device pairing process without any issue and device reporting the correct occupency status.

## How to reproduce it (minimal and precise)

Remove device and try to pair it.

## Debug Info
zigbee2mqtt version: 1.12.1-dev (commit #3f058c3)
CC253X firmware version: zStack 3 with firmware 20191106 

Debug log is available there --> [https://pastebin.com/RETYmtBS](url)

Device address is 0x00155f00b44d48c7

Error message in the log is:  Failed to interview '0x00155f00b44d48c7', device has not successfully been paired`

After that I see a message saying Successfully configured '0x00155f00b44d48c7'

but device is not reporting anythinbg in Z2M when motion is detected.

Fabiancrg, 2020-04-09:
I did another test today, I have a lab with a CC2531 with zStack 3 firmwrare 20190425 and Z2M 1.11.
In this lab I have two routers made of nRF52840 devboard and dongle. 
I was able to pair the device without any issue.
From there, I removed the device and upgraded to Z2M 1.12.1, now I can't pair the device anymore, same issue that with my prod installation.

Fabiancrg, 2020-04-09:
@Koenkk  can this be an issue with the latest version ? I see there was an update related to the device interview [#3276 ](https://github.com/Koenkk/zigbee2mqtt/issues/3276)

Koenkk, 2020-04-09:
This was indeed an issue in the latest version. 

As not only this device, but also other device like Xiaomi Power plug was affected by this I decided to make a new zigbee2mqtt release. Fix is included in Zigbee2mqtt 1.12.2.

Thanks for reporting & providing good logging of the issue.

closed 2020-04-09 by Koenkk

Fabiancrg, 2020-04-09:
@Koenkk Thank you very much for this quick fix !
I confirm everything is working as before now.