Koenkk/zigbee2mqtt #3318: CC2531 identified as EndDevice

lealoureiro, 2020-04-08:
Hello,

First of all thanks for all great work.

I'm having an issue pairing a CC2531 with Router firmware (**router1**) that should be paired to another CC2531 Router (**router2**) which this one identified properly, and then connected to Coordinator.

XiaomiPowerPlug -> router1 -> router2 -> Coordinator 

The router1 is also identified as EndDevice, which should not be the case and it says is using battery but in fact is plugged to an iPhone USB charger.

You can see the messages below:
```json
{
	"state": true,
	"led_state": false,
	"linkquality": 57,
	"device": {
		"friendlyName": "router2",
		"model": "CC2530.ROUTER",
		"ieeeAddr": "0x00124b001938xxxx",
		"networkAddress": 27394,
		"type": "Router",
		"manufacturerID": 4447,
		"manufacturerName": "LUMI",
		"powerSource": "Mains (single phase)",
		"zclVersion": 1,
		"hardwareVersion": 1,
		"dateCode": "2018-09-26"
	}
}
```
Message 2:
```json
{
	"state": true,
	"led_state": false,
	"linkquality": 57,
	"description": "49266/0x00158D0002ECxxxxx",
	"type": "CHILD_RFD",
	"rssi": 170,
	"device": {
		"friendlyName": "router1",
		"model": "CC2530.ROUTER",
		"ieeeAddr": "0x00124b001936xxxx",
		"networkAddress": 1464,
		"type": "EndDevice",
		"manufacturerID": 4151,
		"manufacturerName": "LUMI",
		"powerSource": "Battery",
		"zclVersion": 1,
		"hardwareVersion": 1,
		"dateCode": "2018-09-26"
	}
}
```

And with this i'm having issues pairing or connecting a Xiaomi Power Plug as it is very far from Coordinator thats why I'm trying to have 2 routers in the middle.

Shall i try pair the Power Plug holding it next to neighbor router or next to the coordinator and them move it to the right place next to the router.

## Debug Info
zigbee2mqtt version: 1.12.0
CC253X firmware version: 
- [I flashed both routers using this firmware](https://github.com/Koenkk/Z-Stack-firmware/blob/master/router/CC2531/bin/CC2531_router_2019_02.zip)
- [Coordinator Firmware](https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/Z-Stack_Home_1.2/bin/default/CC2531_DEFAULT_20190608.zipl)

Thank you!

Regards,
Leandro

Koenkk, 2020-04-09:
This issue probably occurred because the interface did not fully complete (probably due to bad range). Removing and repairing it should fix it (https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigremove). Note that this does not actually affect the functionality as a router, its just a cosmetic issue.

lealoureiro, 2020-04-09:
I tried to repair twice next to coordinator still same issue :(

I will try assemble the CC2530 router as it has more power

closed 2020-06-17 by stale[bot]