Koenkk/zigbee2mqtt #3319: Implement proper support for xbee router

alekslyse, 2020-04-09:
This has been up a couple of times, but never really been solved. Xbees are very useful to use as routers in the zigbee mesh, but the interview of them fails. In reality it should behave as an ikea repeater, and in deconx it works as a regular signal repeater.

Could someone with some more debug knowledge take a look on the wireshark and the log and see if its any signs why the interview fails completely and if ever anything we can do to register it correctly as a router?

To make things easier for debugging its used the standard keys on z2m and the edge z2m version

[xbee.zip](https://github.com/Koenkk/zigbee2mqtt/files/4455219/xbee.zip)

alekslyse, 2020-04-09:
After some testing it seems like xbees are giving some strange naming. In deconz they specified xbee is using DIN Zigbee profile. Is this somehthing z2m would be able to use? Yes, the router "works", but would like it register as an xbee. Even if I put a router device with 0x0001 (I see the sniffer sees that also as device id) nothing happens. Maybe someone with a bit more xbee knowledge could make a PK for xbee?

// Digi Drop-In-Networking (DIN) ZigBee Profile, used by the XBee.
#define DIN_PROFILE_ID                      0xC105 // Digi Drop-In-Networking
#define DIN_DDO_ENDPOINT                    0xE6   // Digi Device Object endpoint
#define DIN_DDM_ENDPOINT                    0xE8   // Digi Data Management endpoint
#define DEV_ID_DIN_XBEE                     0x0001 // Device ID used by the XBee

Koenkk, 2020-04-10:
- Looking at the log, the device joins fine in the Zigbee network. Which means it will already act as a Router.
- Regarding interview failed, it seems that the XBEE doesn't respond to the node descriptor requests (one of the things needed by zigbee-herdsman to properly identify a device). To interview it successfully the XBEE should run a firmware which responds to these requests. These commands are mandatory in the ZCL specification.

alekslyse, 2020-04-11:
1. It seems to respond somehow as deconz manage to get info. What report information are z2m looking for? Can I overcome the standard issues by writing custom commands due to ref https://www.digi.com/resources/documentation/Digidocs/90002002/Reference/r_send_ZCL_cmds.htm main goals just to identify the router correctly with firmware info

2. The zha integration of home assistant wrote quirks for xbee https://github.com/zigpy/zha-device-handlers/blob/ea6232ec87191a671c76d39fb519eda73c20f7d7/zhaquirks/xbee/ is z2m capable of doing the same by accessing the xbee like they do. Would be a bit interesting having access to gpio etc

Koenkk, 2020-04-11:
1. I'm not sure how they did it, it would require a sniff to see how they do it.
2. It looks that this setup is done before the device joins the network? I guess this make sense as the XBEE first has to be initialised

alekslyse, 2020-04-12:
Yes it need to join the network first, then it should be able to add sub switching etc. Not sure how this will be possible in z2m though.

I see the main issue is the lack of the attributes report, is attributes something the ZigBee device is asked for from z2m during the interview or does the device send it's attributes on a regular interval?

I think the most important is to just be able to identify it's a xbee so I can write custom handler for it, the question is just how to identify it when it's no attribute report sent.

I tried to sniff deconz and there it's just hard coded so that didn't give much info. Is it anything you see in the log in the first post we can use to identify it as an xbee so it can be linked to a custom handler? 

If it get identified I guess the custom handler give me access/freedom to send custom zcl commands.

alekslyse, 2020-04-12:
I see the only repeatable id is manufacturerID: 4126 - can the manufactorID be used somehow to create a custom handler for xbee to be able to followup with custom commands for the rest of the interview?

alekslyse, 2020-04-12:
I see its a test where you identify a device with just the manufactor id - can we do the same thing with xbee?

`it('Identify xiaomi device without manufacturerName', () => {
            const device = {type: 'Router', manufacturerID: 4447};
            expect(true).toBe(utils.isXiaomiDevice(device));
        });`

Koenkk, 2020-04-12:
In case the device sends the modelID attribute we can add something similar to this without having the XBEE respond to node descriptor requests. To do this just add an entry here: https://github.com/Koenkk/zigbee-herdsman/blob/master/src/controller/model/device.ts#L365

alekslyse, 2020-04-12:
Hmm I added it to that file, but when running interview it still wont touch that quirk. As it could be some npm cache stuff it wont work here. Is it possible to set a custom path to the herdmanns repos in the config so the dev is easier?

. Maybe you can add it to the repo  - should be corret info at least. Any other places I need to add other than that for the interview process to go through?

 ` 'XBEE-XB24C': {
                type: 'Router', manufacturerID: 4126, manufacturerName: 'XBEE', powerSource: 'Mains (single phase)'
            }`

alekslyse, 2020-04-12:
Sent in a pull request. Is it any other places I need to change for z2m to use the quirks? I assume with that I can make a custom device named XBEE-XB24C and follow regular device add process?

Koenkk, 2020-04-13:
@alekslyse I've checked the attached sniffer log of the OP, but this device does not send the modelID so this fix won't help. Based on the sniff, I currently do not see any information which can identify this is an XBEE device.

I'm very interested in how ZHA or Deconz do this, as a next step I would recommend sniffing the traffic when pairing with that.

alekslyse, 2020-04-13:
Where is the 4126 coming from then? Is it a default is z2m is putting in? I see everytime someone try with an xbee that model is present in the z2m logs

alekslyse, 2020-04-13:
And are you sure? It sends 0x0000101e thats = 4126 

[image]

Koenkk, 2020-04-13:
This log is not the same as you provided in the OP, there it doesn't respond to the node descriptor requests (no node descriptor response in the log). 

- Can you provide this sniff and herdsman debug log of this new log?
- What did you change between the OP and this?

alekslyse, 2020-04-13:
Nothing changed at all

wireshark line 51 it responds to the id
wireshark line 93 it responds Profile: Maxstream (0xc105)

The 0xc105 is a xbee default profile, that is not picked up by z2m

[z2m.zip](https://github.com/Koenkk/zigbee2mqtt/files/4470239/z2m.zip)

Koenkk, 2020-04-13:
Where does the screenshot from https://github.com/Koenkk/zigbee2mqtt/issues/3319#issuecomment-612896896 come from? Because with the information for there we can implement a workaround.

alekslyse, 2020-04-13:
Look at the line 51 in the attached wireshark log in the post above (z2m.zip) - same response

Koenkk, 2020-04-13:
Looks that the device pairs correctly now, only thing is a missing modelID which we can work around. Can you provide your `database.db` entry of this device?

alekslyse, 2020-04-13:
Here is the database - should be the second on the list. Only thing thats filled is the model, nothing else.

[database.db.zip](https://github.com/Koenkk/zigbee2mqtt/files/4470431/database.db.zip)

Koenkk, 2020-04-13:
Should be supported in the latest dev branch now, can you check?

It's identified via a new method called "fingerprint": https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices.js#L3216

alekslyse, 2020-04-16:
It seems to be working. I will testest with clean install.

How would the flow be to pull out info from the endpoint like firmware? I linked to an API I hope I can use if not I have to build it myself. 

So. 1 where should unit code be at load for more info like firmware etc.

2. Say the xbee support gpio, where should the code for those be?

Koenkk, 2020-04-16:
You can now write toZigbee converters to send zigbee commands to the device (https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/converters/toZigbee.js). But the firmware on the device should also be able to act on these commands.

closed 2020-06-23 by stale[bot]