Koenkk/zigbee2mqtt #3320: ptvo_switch_uart   value ep1 published in all messages

pdeprost, 2020-04-09:
Hello,

First of all thanks for the new UART support of PTVO router.

How to adapt the converter fromZigbee.js to avoid the last UART message to be repeated in the next MQTT messages (I need it only once, when received by the CC2530 serial port) 

     ptvo_switch_uart: {
         cluster: 'genMultistateValue',
         type: ['attributeReport', 'readResponse'],
         convert: (model, msg, publish, options, meta) => {
              return {'ep1': msg.data['stateText']};
         },
    },

Thank you!

Regards,

Pierre.

Koenkk, 2020-04-09:
@ptvoinfo could you help with this?

ptvoinfo, 2020-04-10:
@pdeprost, @Koenkk 
I don't know how to disable saving of the last state for a single value if the "cache_state" configuration option is active in "configuration.yaml" in zigbee2mqtt.

pdeprost, 2020-04-10:
@Koenkk @ptvoinfo 
Yes but, if  ep1 is managed like an 'input'  button?
example
"click":"bottom_left_double"
appears only once in the message  (when the button is fired).

ptvoinfo, 2020-04-10:
@pdeprost Yes, "click" is marked as "do not cache". But it is not possible to do the same for a custom value.

pdeprost, 2020-04-11:
@Koenkk @ptvoinfo 
Ok, Thank you.  So I will add an incremental index number to the serial data to be able to distinguish a new message form the older repeated from the cache. 
Another question: 
How to send data from Z2M to CC2530 UART's output

ptvoinfo, 2020-04-11:
@Koenkk Could I ask to add a feature to ignore caching for a single value for an individual device? Somehow, specify it in the returned response.

pdeprost, 2020-04-11:
I find this a great idea!

Another question:
How to send data from Z2M to CC2530 UART's output?
mqtt topic "zigbee2mqtt/..../ep1/set" isn't working.

ptvoinfo, 2020-04-11:
@pdeprost  zigbee2mqtt/[FRIENDLY_NAME]/set/ep1

Koenkk, 2020-04-11:
I think the code should be changed form `return {'ep1': msg.data['stateText']};` to `return {'action': msg.data['stateText']};`. This will prevent the message from being cached.

pdeprost, 2020-04-11:
Nice! It's working correctly! 
and to keep compatibility I changed it to : 
		return {'ep1': msg.data['stateText'], 'action': msg.data['stateText']};
Thank you

Koenkk, 2020-04-12:
@ptvoinfo what do you think, does it make sense?

ptvoinfo, 2020-04-12:
@Koenkk It is applicable in this particular case. But the meaning of the value is confusing.

Koenkk, 2020-04-13:
I wil close this as the pull request has been merged. Thanks!

closed 2020-04-13 by Koenkk