Koenkk/zigbee2mqtt #3322: Feature: Zigbee Green Power support

Koenkk, 2020-04-09:
Zigbee Green Power support has now been added. This enables support for e.g. the Philips Hue Tap and Friends of Hue switches.

 I've created this issue in order to discuss this functionallity.

- Requires latest dev branch
- [Documentation](https://github.com/Koenkk/zigbee2mqtt.io/blob/develop/docs/devices/GreenPower_On_Off_Switch.md) (**important**)

electrofloat, 2020-04-09:
So If I'm reading the docs right, this won't work without a Philips Hue device, am I right?

So even If I bought a Hue Tap only (without any bridges, other Hue devices, etc.) I wouldn't be able to use it.

Koenkk, 2020-04-09:
@electrofloat correct

electrofloat, 2020-04-09:
> @electrofloat correct (and it needs to be in range)

That's a shame.. I'd have liked to use it in itself (without any other devices).
But great improvement anyways.
Maybe in the future some miracle happens and makes this doable :)

Koenkk, 2020-04-09:
@electrofloat maybe in the future an additional CC2652R router can also be used for this. But for sure a "special device" is needed to make it work.

alekslyse, 2020-04-09:
Have you tried the green power firmware for xbee? I did see yesterday when I was trying to get xbee working with z2m they had a dedicated ZigBee green power firmware. Anything you want me to try?

TheNetStriker, 2020-04-10:
@Koenkk I've just tested this and I managed to get my two Hue Tap switches to pair. They are now in the database and I assigned a friendly name, but I cannot see anything in the debug log after the pairing when I push the buttons. There are also no mqtt messages on zigbee2mqtt/friendlyname.

Are there any additional steps needed to get this to work or what am I doing wrong?

By the way, I managed to get the devices paired on channel 11 by long clicking on the big button with the one dot that surrounds the smaller buttons. (The documentation states that only channel 15, 20 and 25 can be used)

Koenkk, 2020-04-11:
@TheNetStriker thanks for testing this, any chance you could sniff the traffic when joining the device and pressing buttons afterwards? https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

TheNetStriker, 2020-04-11:
@Koenkk I already uploaded a packet capture in the first thread about the Hue Tap. Here is the link to the pcap file: https://www.dropbox.com/s/pk1jt5h5yvtkb36/HueTap.pcapng?dl=0

Or do you need a fresh packet capture for this?

I also found something about the Hue Tap in the log:
```
zigbee2mqtt:error 2020-04-11 13:12:59: Failed to execute LQI for 'huetap_1'
  zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - mgmtLqiReq - {"dstaddr":28743,"startindex":0} +10s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,3,37,49,71,112,0,32] +10s
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,49,0,85] +10s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,49,0,85] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 49 - [0] - 85 +0ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - mgmtLqiReq - {"status":0} +10s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
```

And in the state.json this is the only property that the device has:
"linkquality": 0

Koenkk, 2020-04-11:
Green power devices don't respond to these LQI requests, so I've excluded them from the network scan (fixed in latest dev branch)

Regarding the sniff, it doesn't contain useful information. I need a sniff in were they are also successfully paired with zigbee2mqtt.

ndfred, 2020-04-11:
Congratulations on getting this working! Should the docs also specify that a Zigbee 3.0 firmware is needed for this to work, or did you manage to get it to work with Zigbee 1.2 as well?

Koenkk, 2020-04-12:
@ndfred it also works with 1.2

ndfred, 2020-04-12:
Just tried it with zStack12 on channel 20, worked wonderfully! A few nits that might be worth calling out in the documentation:

* binding doesn't work (I expected that, but always good to be explicit)
* the vendor might need some polish, I would have expected it to be Philips (not a big deal relly)

Specifically this is the configuration entry for the device (mine is a [Philips Hue Tap](https://www2.meethue.com/en-us/p/hue-tap-switch/046677473365) switch which I bought in France ages ago):

```
{
  "ieeeAddr": "0x0000000000408e0f",
  "type": "GreenPower",
  "networkAddress": 36367,
  "model": "GreenPower_On_Off_Switch",
  "vendor": "GreenPower",
  "description": "On/off switch",
  "friendly_name": "bedroom_fred_bedside_remote",
  "manufacturerID": null,
  "modelID": "GreenPower_2",
  "lastSeen": 1586699144251
}
```

Koenkk, 2020-04-12:
@ndfred thanks, added this to the docs.

We cannot distinguish between e.g. a Hue Tap and a Friends of Hue Niko switch, Green Power device don't share such specific information.

TheNetStriker, 2020-04-13:
@Koenkk I've just uploaded a new packet capture to my Dropbox: https://www.dropbox.com/s/andvjbcpr78p8jk/HueTap.pcapng?dl=0

Koenkk, 2020-04-13:
@TheNetStriker I need your network key in order to decrypt te sniff, please send them to me on telegram (@koenkk).

Zibeez, 2020-04-15:
This is fantastic! This was the only thing keeping me from migrating away from my Hue bridge :)

I tried this with a Niko Friends of Hue switch and a couple of Philips Hue bulbs, with a CC2531 USB stick.

Some notes:
* Is it possible to clarify which actions are supported by which devices? Looking at https://github.com/Koenkk/zigbee-herdsman-converters/blob/21ebddc40f13df79eeb3bd47adbace614826f93b/converters/fromZigbee.js#L4503, a lot of actions are listed, but I find that for the switch I tested only 'recall_scene_x', 'press_x_of_2' and 'release_x_of_2' seem to be supported.
* For the switch I tested, these names are also misleading. Better would be: 'recall_scene_[0-3]' -> 'button_press_[0-3]' and 'recall_scene_[4-7]' -> 'button_release_[0-3]'
* I sometimes experience a (large) delay. At this point I am unsure whether the delay is caused by the fact that the CC2531 is underpowered, a zigbee2mqtt performance issue, or a performance issue in the automation code processing the mqtt messages (I use openhab on a raspberry pi). With the Hue Bridge, response was always instantaneously.

ndfred, 2020-04-15:
Sharing my experience with it, hoping it might be helpful to you, I have a Hue Tap and not a Niko one:

* the actions I've been able to capture are `toggle`, `recall_scene_0`, `recall_scene_1` and `recall_scene_2` respectively for the buttons with 1, 2, 3 and 4 dots
* I haven't seen any delay, how long is "large" in your experience? A few things worth checking:
  * if you tail the ZigBeeToMQTT logs while you are taping the buttons, are you seeing the delay? If not then maybe OpenHAB is introducing it.
  * are you seeing a lot of traffic when tailing the ZigBeeToMQTT log? If so you might see contention on the network.
  * did you change your ZigBee channel from the value you had on the Hue bridge? You might now see interference you didn't see before
  * can you share the link quality of the bulbs around the switch, and have a look at your network map to if the routing makes sense? It might take a bit for your new network to adapt.
  * are you running your CC2531 USB stick from a USB extension cord, and do you have a PCB antenna or an actual antenna? The shielding might be worse than the Hue Bridge.
  * do you still have the Hue bridge running? That could cause ZigBee interference if they are on the same channel (though ZigBee is not that chatty).
  * how large is your ZigBee network (number of devices)? CC2531 is performance issues are only apparent past a certain threshold

I have 17 devices including 6 remotes and one motion detector so really 10 devices that are always on the network, which is not a lot for the CC2531. My Hue Tap remote link quality is 10, which is less than other devices but I guess that is down to its low power budget.

Koenkk, 2020-04-15:
> Is it possible to clarify which actions are supported by which devices? Looking at

1. Hard to do, the main problem is that we cannot identify which device it exactly is, unlike non green power devices, they don't share an exact model ID.

 > For the switch I tested, these names are also misleading. Better would be: 'recall_scene_[0-3]' -> 'button_press_[0-3]' and 'recall_scene_[4-7]' -> 'button_release_[0-3]'

Because of point 1, we cannot further specify the actions, it make work for e.g. the hue tap but not for another device, currently I made the action match against the green power specification.

Zibeez, 2020-04-15:
@ndfred 
Thanks for the quick responses!

> I haven't seen any delay, how long is "large" in your experience? 

Multiple seconds - up to 5.

>    if you tail the ZigBeeToMQTT logs while you are taping the buttons, are you seeing the delay? If not then maybe OpenHAB is introducing it.
>    are you seeing a lot of traffic when tailing the ZigBeeToMQTT log? If so you might see contention on the network.

I will do some more experiments in the next days.

>    did you change your ZigBee channel from the value you had on the Hue bridge? You might now see interference you didn't see before

Yes, my Hue bridge was running on a different channel (I had 2 zigbee networks running in parallel for the past year - one for the Hue bridge, one for zigbee2mqtt). It could be that the Hue Bridge channel was more robust, I will check that if none of the other suggestions help.

>   can you share the link quality of the bulbs around the switch, and have a look at your network map to if the routing makes sense? It might take a bit for your new network to adapt.

Good points, will check that later this week.

> are you running your CC2531 USB stick from a USB extension cord, and do you have a PCB antenna or an actual antenna? The shielding might be worse than the Hue Bridge.

Yes, I run the USB stick from an USB extension cord. I do not use any PCB or antenna.

>  do you still have the Hue bridge running? That could cause ZigBee interference if they are on the same channel (though ZigBee is not that chatty).

No, I shut it down. As mentioned above, it was running fine in parallel with zigbee2mqtt for the past year.

>  how large is your ZigBee network (number of devices)? CC2531 is performance issues are only apparent past a certain threshold

I have 29 devices (and 3 groups). 
Note that I use the Niko switch to control groups of 3 bulbs using the 'groups' configuration, which works very well (they always go on/off in sync, albeit sometimes with that delay). 

I will do more tests during the weekend. Thanks for the help!

@Koenkk 
> Because of point 1, we cannot further specify the actions, it make work for e.g. the hue tap but not for another device, currently I made the action match against the green power specification.

Ah, that helps! Might be a good idea to clarify that in the documentation.

Zibeez, 2020-04-19:
>>    if you tail the ZigBeeToMQTT logs while you are taping the buttons, are you seeing the delay? If not then maybe OpenHAB is introducing it.
    are you seeing a lot of traffic when tailing the ZigBeeToMQTT log? If so you might see contention on the network.

> I will do some more experiments in the next days.

Additional testing pointed indeed to Openhab and its MQTT broker. After optimizing those, zigbee2mqtt + Green Power support now works similar to the old Hue bridge :-)

presslab-us, 2020-05-04:
Does anyone have experience using Z-Stack 3.0.x Router as a GPP?  I have it working with Zigbee2mqtt but there are some problems with Z-Stack.  I opened a thread over on the TI forum.
https://e2e.ti.com/support/wireless-connectivity/zigbee-and-thread/f/158/t/902398

mshli, 2020-05-06:
@Koenkk: For your Information: The Philips Hue Tap can also join the Network on Channel 11 by pressing the "One-Dot-Button" for 10 Seconds. Works without any problems. Please add this to the docs.

thibaultmol, 2020-05-06:
@Koenkk so no special zigbee hub/usb stick is needed to be 'compatible with' zigbee green power, correct?

sjorge, 2020-05-06:
> @Koenkk so no special zigbee hub/usb stick is needed to be 'compatible with' zigbee green power, correct?

I think the only requirement is a GP enabled router... so a hue bulb.

Koenkk, 2020-05-06:
@mshli added, thanks

@thibaultmol , @sjorge  is right (no special adapter is required).

juamiso, 2020-05-07:
just tried iobroker.zigbee adatper since the last update stated that green energy was working (link to this thread) I tried to pair a NodOn friends of hue switch but it does not recognise the device. (I had it previously working in the philips hue bridge). 

Can I fix this locally or do I have to wait for some updates?

```
Error: Device 0x000000000171a1a7 "GreenPower_2" not described in statesMapping.. undefined
New device joined '0x000000000171a1a7' model GreenPower_2
Error: Device 0x000000000171a1a7 "GreenPower_2" not described in statesMapping.. undefined
Interview successful: {"friendly_name":"0x000000000171a1a7","model":"GreenPower_On_Off_Switch","vendor":"GreenPower","description":"On/off switch","supported":true}
```
Device information
model:GreenPower_On_Off_Switch
vendor:GreenPower
description:On/off switch
supports:action
model:GreenPower_2
type:GreenPower
nwk:41383
manuf id:null
interviewed:true
configured:false

Nice info here about the pairing proc. on different Zigbee Channels:
[here](https://phoscon.de/en/support#pairing-friends-of-hue-switch)

jkrenzer, 2020-07-07:
> > @Koenkk so no special zigbee hub/usb stick is needed to be 'compatible with' zigbee green power, correct?
> 
> I think the only requirement is a GP enabled router... so a hue bulb.

Just one question: Does it need to be specific a hue bulb or would a normal Zigbee 3.0 compatible bulb do? I read some conflicting reports of GP being part of the Zigbee 3.0 core specification? *head scratch*

Koenkk, 2020-07-07:
@jkrenzer good question, according to https://training.ti.com/sites/default/files/zigbee-green-power-techday2019.pdf slide 13  all Zigbee 3.0 routing devices must implement the green power proxy. So theoretically a Zigbee 3.0 router should be enough.

jkrenzer, 2020-07-07:
That would be good news for us, wouldn't it? I have to see if I can acquire a GP remote and test with my tint (Müller Licht) bulbs which are Zigbee 3.0.

swannes, 2020-07-18:
> @jkrenzer good question, according to https://training.ti.com/sites/default/files/zigbee-green-power-techday2019.pdf slide 13 all Zigbee 3.0 routing devices must implement the green power proxy. So theoretically a Zigbee 3.0 router should be enough.

I tried paring a Niko switch https://www.niko.eu/nl-be/producten/niko-home-control/producten-voor-traditionele-bekabeling/geconnecteerde-schakelbare-wandcontactdoos-voor-niko-home-control-met-penaarde-en-productmodel-niko-aa84a653-36c4-55de-b6d1-d2ddfc6326dc
It shows up as a router / connected socket outlet in zigbee2mqttassistant, but did not show op in Home Assistant. No HUE bulbs in my zigbee-mesh wich act as a router.

sjorge, 2020-07-18:
Those don't like like GP based end devices but actual routers, I don't think z2m supports those yet.

Also in the market for a zigbee based switch (but it needs to support decoupling), I'll go over the tech doc later this week as my current alternative is expensive.

XRyu, 2020-09-03:
Can we re-open this? I have an HueTap paired here. No other Hue Products besides an Bridge which i migrated from (months ago) and two Motion Sensors which are still paired to this bridge.
In my Z2M Mesh there is no other Hue device. I am running the latest firmware on a CC1352P2. Was able to pair with Z2M easily. When pushing buttons in 1 of 4 times it works and i get toggle/recall_scene_(0/1/2), when it does not i get this Error in the Logs:

```
(node:16) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 16),
(node:16) UnhandledPromiseRejectionWarning: TypeError [ERR_INVALID_ARG_TYPE]: The "data" argument must be of type string or an instance of Buffer, TypedArray, or DataView. Received undefined,
    at Cipheriv.update (internal/crypto/cipher.js:155:11),
    at GreenPower.encryptSecurityKey (/app/node_modules/zigbee-herdsman/dist/controller/greenPower.js:62:34),
    at GreenPower.<anonymous> (/app/node_modules/zigbee-herdsman/dist/controller/greenPower.js:69:34),
    at /app/node_modules/zigbee-herdsman/dist/controller/greenPower.js:27:71,
    at Generator.next (<anonymous>),
    at new Promise (<anonymous>),
    at __awaiter (/app/node_modules/zigbee-herdsman/dist/controller/greenPower.js:23:12),
    at GreenPower.onZclGreenPowerData (/app/node_modules/zigbee-herdsman/dist/controller/greenPower.js:66:16),
    at Controller.<anonymous> (/app/node_modules/zigbee-herdsman/dist/controller/controller.js:432:37),
(node:16) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 17)
```

Koenkk, 2020-09-04:
@XRyu please provide the herdsmans debug logging of this issue

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

GbolaG, 2020-09-05:
@Koenkk 

If we are using the ZZH CC2652 stick as our Z2MQTT coordinator, do we still need a separate 'special' device such as a Hue bulb?

Thanks for all your work on this project.

Koenkk, 2020-09-05:
@islanderx yes

XRyu, 2020-09-05:
Btw i am having some Müller Licht Zigbee 3.0 Lamps running in my mesh if this helps :)

Attached the log, where i tapped the hueTap several times, sometimes the lights turned on / off, but most times nothing happened :) Hope the Log helps.
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/5178527/log.txt)

Koenkk, 2020-09-07:
@XRyu I expected that you will publish the full herdsman debug log of [this](https://github.com/Koenkk/zigbee2mqtt/issues/3322#issuecomment-686716478) error but I don't see it.

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

XRyu, 2020-09-07:
> @XRyu I expected that you will publish the full herdsman debug log of [this](https://github.com/Koenkk/zigbee2mqtt/issues/3322#issuecomment-686716478) error but I don't see it.
> 
> To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

Oh, i thought its in there but it seems to not be logged to file. I have seen these entries in the Console Logs of Docker running through. As i am using the docker-Version, is it possible the herdsman debug log is saved somewhere else than the z2m debug log? I dont see the herdsman-Debug output in the generated log(x).txt files

Koenkk, 2020-09-07:
Unfortunately it's only logged to stdout (not to zigbee2mqtt log file). However I only need the error + some lines above that.

XRyu, 2020-09-07:
alright! I think i got it now :)

```

2020-09-07T20:33:44.256Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [],
(node:18) UnhandledPromiseRejectionWarning: TypeError [ERR_INVALID_ARG_TYPE]: The "data" argument must be of type string or an instance of Buffer, TypedArray, or DataView. Received undefined,
    at GreenPower.encryptSecurityKey (/app/node_modules/zigbee-herdsman/dist/controller/greenPower.js:62:34),
    at Cipheriv.update (internal/crypto/cipher.js:155:11),
    at GreenPower.<anonymous> (/app/node_modules/zigbee-herdsman/dist/controller/greenPower.js:69:34),
    at Generator.next (<anonymous>),
    at /app/node_modules/zigbee-herdsman/dist/controller/greenPower.js:27:71,
    at Controller.<anonymous> (/app/node_modules/zigbee-herdsman/dist/controller/controller.js:432:37),
    at GreenPower.onZclGreenPowerData (/app/node_modules/zigbee-herdsman/dist/controller/greenPower.js:66:16),
    at new Promise (<anonymous>),
    at __awaiter (/app/node_modules/zigbee-herdsman/dist/controller/greenPower.js:23:12),
(node:18) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 18),
2020-09-07T20:33:44.257Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,27,68,129,0,0,6,0,92,220,1,1,0,84,0,124,252,170,0,0,7,8,115,10,0,0,16,0,100,34,27,29],
2020-09-07T20:33:44.257Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,27,68,129,0,0,6,0,92,220,1,1,0,84,0,124,252,170,0,0,7,8,115,10,0,0,16,0,100,34,27,29],
2020-09-07T20:33:44.257Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 27 - 2 - 4 - 129 - [0,0,6,0,92,220,1,1,0,84,0,124,252,170,0,0,7,8,115,10,0,0,16,0,100,34,27] - 29,
2020-09-07T20:33:44.257Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":6,"srcaddr":56412,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":84,"securityuse":0,"timestamp":11205756,"transseqnumber":0,"len":7,"data":{"type":"Buffer","data":[8,115,10,0,0,16,0]}},
2020-09-07T20:33:44.258Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":115,"manufacturerCode":null,"commandIdentifier":10},"Payload":[{"attrId":0,"dataType":16,"attrData":0}]},"address":56412,"endpoint":1,"linkquality":84,"groupID":0}',
2020-09-07T20:33:44.256Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,15,36,1,130,99,242,1,33,0,174,0,30,5,24,194,11,4,0,121],
2020-09-07T20:33:44.255Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":25474,"destendpoint":242,"srcendpoint":1,"clusterid":33,"transid":174,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[24,194,11,4,0]}},
2020-09-07T20:33:44.255Z zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0x0000000000446382:25474/242 (0,0),
2020-09-07T20:33:44.253Z zigbee-herdsman:controller:endpoint DefaultResponse 0x0000000000446382/242 33(4, {"timeout":10000,"disableResponse":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}),
2020-09-07T20:33:44.251Z zigbee-herdsman:controller:log Skipping command 'commisioningNotification' because it is missing from the lookup,
2020-09-07T20:33:44.240Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":194,"manufacturerCode":null,"commandIdentifier":4},"Payload":{"options":2848,"srcID":4481922,"frameCounter":2267,"commandID":16,"payloadSize":0,"commandFrame":{"raw":{"type":"Buffer","data":[138,180,91,60,109,214,110]}}}},"address":25474,"endpoint":242,"linkquality":51,"groupID":0}',
2020-09-07T20:33:44.239Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":33,"srcaddr":25474,"srcendpoint":242,"dstendpoint":242,"wasbroadcast":1,"linkquality":51,"securityuse":0,"timestamp":11204331,"transseqnumber":0,"len":22,"data":{"type":"Buffer","data":[1,194,4,32,11,130,99,68,0,219,8,0,0,16,0,138,180,91,60,109,214,110]}},
2020-09-07T20:33:44.239Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 42 - 2 - 4 - 129 - [0,0,33,0,130,99,242,242,1,51,0,235,246,170,0,0,22,1,194,4,32,11,130,99,68,0,219,8,0,0,16,0,138,180,91,60,109,214,110,138,180,29] - 153,
2020-09-07T20:33:44.239Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,42,68,129,0,0,33,0,130,99,242,242,1,51,0,235,246,170,0,0,22,1,194,4,32,11,130,99,68,0,219,8,0,0,16,0,138,180,91,60,109,214,110,138,180,29,153],
2020-09-07T20:33:44.237Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,42,68,129],
2020-09-07T20:33:44.238Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [0,0,33,0,130,99,242,242,1,51,0,235,246,170,0,0,22,1,194,4,32,11,130,99,68,0,219,8,0,0,16,0,138,180,91,60,109,214,110,138,180,29,153],
2020-09-07T20:33:44.236Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":25474,"relaycount":2,"relaylist":[23660,28386]},
[... cut ...]
2020-09-07T20:33:41.730Z zigbee-herdsman:controller:endpoint CommandResponse 0x00158d000284e09a/1 genOta.queryNextImageResponse({"status":152}, {"timeout":10000,"disableResponse":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}),
Zigbee2MQTT:debug 2020-09-07 22:33:41: Received Zigbee message from 'Licht - Outdoor Spots C', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":655368546,"imageType":1795,"manufacturerCode":4454}' from endpoint 1 with groupID 0,
2020-09-07T20:33:41.730Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
```

Koenkk, 2020-09-08:
Issue should be fixed in latest dev branch

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

XRyu, 2020-10-10:
Works now :)

closed 2020-11-18 by github-actions[bot]

wb441, 2020-12-25:
Good afternoon!
I have a couple of Niko Friends of Hue switches.
The doc allowed me to pair them successfully.
So far, I could only get them to do anything useful by going via NodeRED to "pair" the button actions to send a MQTT command back to z2M.

Ideally, I would like the Niko Friends of Hue switch to influence a nearby Ubisys D1.  That last one supports Zigbee Green Power, but is there in fact now way for them to work together unless a layer like NodeRED/Home Assistant/OpenHAB is used?

Thanks for the clarification

vandenbogerd, 2021-05-24:
I just managed to get the "GreenPower_On_Off_Switch" in Zigbee2MQTT.
But I don't get how to get the 4 buttons active in HA?

Anything I'm missing here?

davidjb, 2021-08-01:
@vandenbogerd The four buttons appear as the state of the relevant Tap switch's sensor in Home Assistant. So, in other words if you have a Tap switch with the friendly name of `tap_switch_foyer` in zigbee2mqtt, it'll appear as `sensor.tap_switch_foyer_action` in HA and its state will momentarily be one of `toggle` (button 1), `recall_scene_0` (button 2), `recall_scene_1` (button 3) and `recall_scene_2` (button 4) after a press occurs.

From there, you can use the state of the sensor to create triggers for automations like so.  This example listens to presses from different buttons on two different switches to perform the same action but you could just as easily listen to just a single trigger:

```yaml
- id: switch_foyer_light
  alias: Toggle Foyer Light from foyer switch and lounge switch
  initial_state: true
  trigger:
    - platform: state
      entity_id: sensor.tap_switch_foyer_action
      to: toggle
    - platform: state
      entity_id: sensor.tap_switch_lounge_action
      to: recall_scene_2
  action:
    - service: light.toggle
      entity_id: light.foyer_light
```

phasperhoven, 2021-12-21:
I'm in the process of moving my devices from deConz to Zigbee2mqtt. After pairing one of my Greenpower switches I noticed this combination doesn't seem to support a hold (long press) action. I just see press and release actions. deConz sent different codes for short and long press. Unfortunately I was using some of these long presses, so I'd like them supported:-). Is that something that is coming, or has someone found some sort of solution/workaround, either in Z2M or in Homeassistant?

juamiso, 2021-12-21:
> I'm in the process of moving my devices from deConz to Zigbee2mqtt. After pairing one of my Greenpower switches I noticed this combination doesn't seem to support a hold (long press) action. I just [...]

Consider staying with deconz. I went from zigbee2mqtt to deconz and the robustness has improved dramatically. At least in my case it has been rock solid (using green zigbee devices).

phasperhoven, 2021-12-21:
> Consider staying with deconz. I went from zigbee2mqtt to deconz and the robustness has improved dramatically. At least in my case it has been rock solid (using green zigbee devices).

Thanks, that obviously was not what I wanted to hear :-) :-)
But of course switching back may alway be an option.
I have two reasons for switching:
1) device compatibility where Z2M supports more types of devices
2) terrible performance of deConz at the moment: when I switch on a light it may last several seconds before it responds.

nickagian, 2022-01-25:
I've read through the thread here and I'm still unsure how to use Green Power devices with Z2M. It is written that a prerequisite is that a Hue device has to be in range, but no further details are given.

Will any Hue device work?
Do I need a Hue gateway?
Do I need to do anything special or just power the Hue device and keep it in range?

wb441, 2022-01-27:
Dear nickagian,

Indeed, I found this somewhat surprising as well.
Although many devices talk about Zigbee Green Power, in fact,
if I want the Friends of Hue batteryless button, I need to have a Hue device present in the installation. In my case, I use the Philips Hue plug.
If I unplug it, the Green Power Device doesn't work anymore.

I am sure that the Friends of Hue devices need a Hue device which is powered to 'decode' the messages.
I have that working well on multiple locations. I also would find it great that this 'translation' could be extracted and added to the zigbee herdsman/z2m implementation, but today I believe it is not yet taken up by someone.

If Zigbee Green Power devices from other manufacturers have enough with the implemention on your Zigbee coordinator, that I can not confirm or deny. Perhaps you can post a question about it with a more experienced member? I would be interested to find out more.

wb441, 2022-01-27:
The doc also states that binding is not supported. However, z2m knows which device the command is coming from. So I wonder what the technical reason is that binding is not supported for Green Power devices...

akentner, 2022-01-29:
> I'm in the process of moving my devices from deConz to Zigbee2mqtt. After pairing one of my Greenpower switches I noticed this combination doesn't seem to support a hold (long press) action. I just [...]

I also just switched from deCONZ to z2m and sadly realized, that PTM215Z (in my case Scenic Friends of Hue) does not support "long" press. I only get `recall_scene_x` and `press|release_x_of_2`.

I'm not familiar with external converters, yet. Perhaps, it is possible to "fake" a long press. Like, if the time between `recall_scene_0` and `recall_scene_4` is 0-500ms => short, if 500ms-1500ms =>, longer than 1500ms => hold.
I don't know, if that possible. Anybody ideas on that?

phasperhoven, 2022-01-31:
I (sort of) implemented the long press in Homeassistant: pressing a button triggers an automation that sets a boolean to true, waits a second and switches it off again. Releasing triggers an automation that chooses between two different actions depending on the state of the boolean.

Downside: the long press-action is not triggered on hold, but on release after hold. Takes a little getting used to,

MortenVinding, 2022-02-02:
> Indeed, I found this somewhat surprising as well. Although many devices talk about Zigbee Green Power, in fact, if I want the Friends of Hue batteryless button, I need to have a Hue device present [...]

I only have a SunRicher SR-ZGP2801K4-DIM remote.
Pairing to Z2M is cumbersome but doable. I only have 2 Hue bulb's in my zigbee network, so I tried to disconnect power to both of them, but the remote continued to work.

I believe that the IKEA bulbs with newer firmware also works as Green Power proxies.

> Consider staying with deconz. I went from zigbee2mqtt to deconz and the robustness has improved dramatically. At least in my case it has been rock solid (using green zigbee devices).

Is that only with green power devices? I don't like green power because I prefer to bind the remotes to the bulb's.
but my experience with Z2M is rock solid! I simply can't remember the last time it missed a beat.
even with most of the house fitted with cheap IKEA bulb's it still super stable.

and that correlates fine with the experience I hear from other places, where many has switched from Deconz to Z2M.

> 

> I also just switched from deCONZ to z2m and sadly realized, that PTM215Z (in my case Scenic Friends of Hue) does not support "long" press. I only get `recall_scene_x` and `press|release_x_of_2`.
> 
> I'm not familiar with external converters, yet. Perhaps, it is possible to "fake" a long press. Like, if the time between `recall_scene_0` and `recall_scene_4` is 0-500ms => `press_1_short`, if 500ms-1500ms =>`press_1_long`, longer than 1500ms => `hold_1`.

That is interesting. Do we really miss some info from these devices, or is the hold action from Deconz actually just a simulated event? @Koenkk ?

gulli1986, 2022-02-21:
Same here, coming from Deconz to Zigbee2MQTT and missing the long_press action. Is is something that could be implemented like it was done with the [Hue Dimmer Switch](https://github.com/Koenkk/zigbee2mqtt/issues/585)?

Skorfulose, 2022-02-21:
As per the [device specification](https://fccid.io/SZV-PTM215Z/User-Manual/usersmanual-2073658.pdf), I am now quite sure it only sends press and release events.

> 1.1 Basic Functionality
> […]
>  It is therefore possible to distinguish between radio telegrams sent when the energy bar was pushed and radio telegrams sent when the energy bar was released.
>  By identifying these different telegrams types and measuring the time between pushing and releasing of the energy bar, it is possible to distinguish between “Long” and “Short” push button presses. This enables simple implementation of applications such as dimming control or blinds control including slat action.

gulli1986, 2022-02-21:
I am really interested to see an automation either in Node Red or HA if someone managed to simulate a long press!

MortenVinding, 2022-02-21:
> I am really interested to see an automation either in Node Red or HA if someone managed to simulate a long press!

I have had success by using the blueprint:
Controller - EnOcean PTM 215Z (Friends of Hue) switch
Controller automation for executing press/hold/release actions triggered by EnOcean PTM 215Z (Friends of Hue) switch.

it has 3 actions for each of the 4 buttons:
Pressed,
Held and
Released

I use it with the brightness_move command in Zigbee2MQTT, so the "Held" action sends a brightness_move command, and the "Released" action sends a brightness_stop command.
works almost as good as direct binding, even on my small RPi3b.

I have also tried with the new version of Tasmota with the new "Dimmer >", "Dimmer !" and "Dimmer !" commands.
(brightness increase, stop and decrease.

I use this code for Tasmota:

```
service: mqtt.publish
data:
  topic: cmnd/tasmota_CCEAC0/Backlog
  payload: Dimmer1 >
```

not as accurate as zigbee control, but usable.

MortenVinding, 2022-02-21:
> > That is interesting. Do we really miss some info from these devices, or is the hold action from Deconz actually just a simulated event? @Koenkk ?
> 
> I guess it does, you can see when the device sends a message by pressing button while debug logging is activated.
> 
> See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

Thanks a lot @Koenkk!
From looking at some of the other GitHub issues I now see there is no held command from this switches, so I guess the blueprint implementation I have found is the best way to do this.

I really appreciate the crazy amount of work you put in to this project! I have been using Zigbee2MQTT for a few years now (first with OpenHAB and now with Home Assistant) and it just keeps getting better and better!
I love the technology agnostic nature of it (changing from OpenHAB to Home Assistant was just a matter of pointing my new Home Assistant to the existing MQTT broker!), and in my opinion none of the other zigbee implementations does get anywhere close to the functionality and stability of Zigbee2MQTT!

May I recommend every user to do what I have done: DONATE!
Donate if Zigbee2MQTT is as important in your smart home setup as it is in mine. We really want this project to continue developing as much as it already has don't we? 😉
[https://www.buymeacoffee.com/koenkk](https://www.buymeacoffee.com/koenkk)

mikkelsiggaard, 2022-07-27:
I've just spend some time reading about Zigbee Green Power. 
Sources:
https://csa-iot.org/wp-content/uploads/2022/01/docs-14-0563-18-batt-Green-Power-Basic-specification-v1.1.1.pdf
https://www.silabs.com/documents/public/user-guides/ug103-15-green-power-fundamentals.pdf
https://www.ti.com/lit/an/swra615a/swra615a.pdf
https://hometoys.com/the-power-of-zigbee-30-all-about-the-new-and-improved-zigbee-30/

It is feature set included in Zigbee 3.0 and it enables battery-less devices to communicate, by using the energy harvested by pressing buttons. 
I think this is why there are no "hold" actions sent; because no energy can be harvested by simply holding a button. 
This can only be done by pressing a button. 
A device with a battery can use Zigbee Green Power to achieve lower battery consumption, but sacrifices bi-directional communication (beware of exceptions). 
To further reduce battery consumption Zigbee Green Power devices uses a smaller packet size, compared to good ol' regular ZigBee. 
[image]

The best technical, but still relatively high-level description of the requirements to run a Zigbee Green Power is found in the document from Texas Instruments:
[image]
From this image, I think that the Coordinator can act as the "Sink" (I think they also call it "source"). 
What we normally think of as a Zigbee Router, which now is called a "proxy" (and sometimes a "node"), is used to convert the smaller Green Power packet to a normal Zigbee packet, that is then sent to the Coordinator/sink/source - possibly along one or more Routers/proxies/nodes (normal Zigbee mesh stuff). 

I think this explains the behaviour people in this thread are experiencing. Namely: 

1. A Router/proxy/node that conforms to The Zigbee Green Power specification v 1.1.1 has to be between the Zigbee Green Power device and the Coordinator/sink/source. I.E. a Philips Hue bulb/plug. 
2. Since the Coordinator/sink/source doesn't "know" that a packet came from a ZigBee Green Power device, because it was translated by the Router/proxy/node, I don't think a Coordinator/sink/source has any problems receiving packets from a Zigbee Green Power device - and hence there shouldn't be any modifications made too Zigbee2MQTT (AFAIK) since it already has been handled by https://github.com/Koenkk/zigbee-herdsman/pull/92. 

So, it is up to the rest of the manufacturers of mains-powered Zigbee 3.0 (sometimes also called "Zigbee PRO") to implement support for the Zigbee Green Power specification v.1.1.1. 

Alternatively (and I don't know if this is up to spec, instead of waiting for the lazy manufacturers, could it be possible to implement the Zigbee Green Power specification v.1.1.1 in the Coordinator firmware? I realise this would circumvent the whole mesh functionality of Zigbee, but it might work? Just maybe.. ? 

BTW: A nice breakdown has also been made here (https://github.com/zigpy/zigpy/issues/341) but with ZHA in mind. 

A quick note: I hate the way the Zigbee standard has evolved. Standards within standards, confusing naming schemes and exception upon exceptions. I've spared you all the nasty details, to be able to provide as clear and precise a description as possible. I think they had pretty good intentions, but too much freedom was left for the manufacturers - in my opinion, that is.

[4 more replies]