Koenkk/zigbee2mqtt #3323: Old Ikea e1524 button always report battery status 21

sanchosk, 2020-04-09:
# Bug Report
## What happened
I have multiple Ikea Tradfri buttons. Today I obtained an old version of E1524. I updated the firmware to the latest version, now zigbee2mqtt reports firmware update false.
Everything works, but the battery is always reported as 21.
No matter if I power it from 3V, 3.1V or 2.4V.
I can see the button is struggling - when the voltage is below ~2.5V, it blinks red LED after every button press, but reports always 21.

I guess this is button firmware problem as the zigbee2mqtt just "relays" the message...

## What did you expect to happen
I expected it to report the same percentage values as other TRADFRI buttons, e.g. E1810.

## How to reproduce it (minimal and precise)
Update the firmware of E1524
Power the device from desk power supply with 2.4V.
Observe the LED indicator blinking to show low power.
zigbee message will be
{"linkquality":63,"update_available":false,"battery":21,"action":"toggle"}
Increase the voltage to 3V.
Observe the same report from the device:
{"linkquality":60,"update_available":false,"battery":21,"action":"toggle"}

## Debug Info
zigbee2mqtt version: zigbee2mqtt@1.12.0
CC253X firmware version: zStack12, dateCode 20190608

Koenkk, 2020-04-11:
The new battery status is not immediately reported, this can take up to 24 hours. If after that it still doesn't work you can try to reconfigure the device: https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigure (make sure to wake it up right before doing that by pressing a button on it).

myradon, 2020-04-13:
Have the same with my  IKEA E1524/E1810. Zigbee2mqtt states the device is "unsupported". Battery was 47% so OTA-update didn't work. Switche battery 47%?! Connected 3.3Volts from Powersupply : 47%. Next day 47%! I tossed the remote control in the corner.

sanchosk, 2020-04-13:
I plugged both to 3.3V lab bench power supply and managed to do the upgrade.
Did not help :(
I've kept it on the power supply for 48 hours, still reports the same.
Funny thing - I also have the E1810, worked OK before (reported battery level 75%). I've updated it yesterday (powered also by power supply during the activity, not the batter), put the battery back and it now reports 17 :)

Koenkk, 2020-04-14:
@myradon is this wit zigbee2mqtt 1.12.2? Because there should be a fix included for this there. After updating make sure to wait at least 24 hours for a new battery report to come in.

sanchosk, 2020-04-14:
This is what the startup log said in journalctl:
`2020-04-06 02:36:15: Starting zigbee2mqtt version 1.12.0 (commit #840b9d9)`

sanchosk, 2020-04-14:
I now forced git reset and pulled the latest master. But the differences were not related to this problem.
Now the journal reports
`2020-04-14 09:09:37: Starting zigbee2mqtt version 1.12.2 (commit #911cd58)`

closed 2020-06-20 by stale[bot]

jempo, 2021-06-07:
Hi, i've the same problem. Changed the battery from 2 days ago but is still 21%.
I've the version 1.19.1.