Koenkk/zigbee2mqtt #3324: Adding/removing from group doesnt work

andreypopov, 2020-04-10:
ver 1.12.2:
Trying to add/remove device from group. No response on valid group/device names.   Please, check.

[image]

Koenkk, 2020-04-11:
Can you provide the debug logging and your `configuration.yaml`?

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

andreypopov, 2020-04-17:
[image]
`info  2020-04-17 07:46:29: Adding 'door1' to 'qwe1'
info  2020-04-17 07:46:31: Adding 'door2' to 'qwe1'
error 2020-04-17 07:46:34: Failed to call 'Groups' 'onMQTTMessage' (Error: Command 0x00158d000239cc0a/1 genGroups.add({"groupid":2,"groupname":""}, {"timeout":6000,"manufacturerCode":null,"disableDefaultResponse":true}) failed (Error: AREQ - AF - dataConfirm after 5000ms)
    at Endpoint.<anonymous> (/mnt/data/root/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:349:23)
    at Generator.throw (<anonymous>)
    at rejected (/mnt/data/root/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65))
error 2020-04-17 07:46:36: Failed to call 'Groups' 'onMQTTMessage' (Error: Command 0x00158d0002b774d8/1 genGroups.add({"groupid":2,"groupname":""}, {"timeout":6000,"manufacturerCode":null,"disableDefaultResponse":true}) failed (Error: AREQ - AF - dataConfirm after 5000ms)
    at Endpoint.<anonymous> (/mnt/data/root/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:349:23)
    at Generator.throw (<anonymous>)
    at rejected (/mnt/data/root/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65))
error 2020-04-17 07:46:57: Device 'q4' does not exist
info  2020-04-17 07:46:57: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_group_add_failed","message":{"friendly_name":"q4","group":"test","error":"entity doesn't exists"}}'
`

andreypopov, 2020-04-17:
[image]

andreypopov, 2020-04-21:
any news here?

Koenkk, 2020-04-21:
- Can you check with latest dev branch?
- What device is `0x00158d000239cc0a`?

andreypopov, 2020-04-22:
**1.12.2-dev**
0x00158d000239cc0a: lumi.sensor_magnet

nothing changed with dev

Koenkk, 2020-04-22:
This device does not supports groups

andreypopov, 2020-04-22:
Can you add log with error? no messages right now

andreypopov, 2020-04-23:
Also there is no error, when device is turned off. 

[image]

Koenkk, 2020-04-23:
- There is currently no list of devices that support group. Generally these are switches and bulbs
- The MQTT feedback will be fixed here #3281

closed 2020-07-02 by stale[bot]