Koenkk/zigbee2mqtt #3325: adding orvibo T11W1LZ single switch

ARoozitalab, 2020-04-10:
hi i want to add orvibo t11w1lz single switch to zigbee2mqtt

in debug mode i can see :
`zigbee2mqtt:debug 2020-04-10 11:55:04: Received Zigbee message from '0x00124b000adca903', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 2 with groupID 0
zigbee2mqtt:warn  2020-04-10 11:55:04: Received message from unsupported device with Zigbee model '374b2e283a7147609956d4412615afc0'
zigbee2mqtt:warn  2020-04-10 11:55:04: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
zigbee2mqtt:debug 2020-04-10 11:55:23: Device '0x00124b000adca903' announced itself
zigbee2mqtt:info  2020-04-10 11:55:23: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"0x00124b000adca903"}}'
zigbee2mqtt:debug 2020-04-10 11:55:29: Received Zigbee message from '0x00124b000adca903', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 2 with groupID 0
zigbee2mqtt:warn  2020-04-10 11:55:29: Received message from unsupported device with Zigbee model '374b2e283a7147609956d4412615afc0'
zigbee2mqtt:warn  2020-04-10 11:55:29: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.`

i added follow codes to device.js

`{
    zigbeeModel: ['374b2e283a7147609956d4412615afc0'],
    model: 'T11W1LZ',
    vendor: 'Orvibo',
    description: 'swith 1 state', 
    supports: 'on/off', 
    fromZigbee: [], 
    toZigbee: [tz.on_off],
        ep: (device) => {
            return {'1': 1};
        }
},`

and added follow command to hassio configuration.yaml

`switch:
  - platform: "mqtt"
    state_topic: "zigbee2mqtt/0x00124b000adca903"
    availability_topic: "zigbee2mqtt/bridge/state"
    payload_off: "OFF"
    payload_on: "ON"
    value_template: "{{ value_json.state }}"
    command_topic: "zigbee2mqtt/0x00124b000adca903/1/set"

sensor:
  - platform: "mqtt"
    state_topic: "zigbee2mqtt/0x00124b000adca903"
    availability_topic: "zigbee2mqtt/bridge/state"
    icon: "mdi:signal"
    unit_of_measurement: "lqi"
    value_template: "{{ value_json.linkquality }}"`

but I have this error in my logs
`zigbee2mqtt:error 2020-04-10 13:34:36: Publish 'set' 'state' to '0x00124b000adca903' failed: 'TypeError: Cannot read property 'command' of undefined'
`
I am new with zigbee2mqtt and I read this page 
https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
but I can't understand how should I solve my problem

thanks

Koenkk, 2020-04-10:
Try with

```js
{
    zigbeeModel: ['374b2e283a7147609956d4412615afc0'],
    model: 'T11W1LZ',
    vendor: 'Orvibo',
    description: 'swith 1 state', 
    supports: 'on/off', 
    fromZigbee: [fz.on_off], 
    toZigbee: [tz.on_off],
},
```

and control it by sending to `zigbee2mqtt/0x00124b000adca903/set` payload `{"state": "ON"}`.

ARoozitalab, 2020-04-25:
thanks now device change correctly and I have status correctly
but I have a new problem
after each home assistance reboot device.js  back to default without my new config.
and after config , although everything works correctly but I have these logs:
```
zigbee2mqtt:info  2020-04-25 17:30:47: MQTT publish: topic 'zigbee2mqtt/0x00124b000adca903', payload '{"state":"OFF","linkquality":78}'
(node:229) UnhandledPromiseRejectionWarning: TypeError: Cannot read property 'includes' of undefined
    at HomeAssistant.onPublishEntityState (/zigbee2mqtt-1.12.2/lib/extension/homeassistant.js:1633:59)
    at EventBus.<anonymous> (/zigbee2mqtt-1.12.2/lib/extension/homeassistant.js:1540:63)
    at EventBus.emit (events.js:223:5)
    at EventBus.emit (/zigbee2mqtt-1.12.2/lib/eventBus.js:15:15)
    at Controller.publishEntityState (/zigbee2mqtt-1.12.2/lib/controller.js:292:23)
    at publish (/zigbee2mqtt-1.12.2/lib/extension/deviceReceive.js:146:22)
    at DeviceReceive.onZigbeeEvent (/zigbee2mqtt-1.12.2/lib/extension/deviceReceive.js:176:13)
    at Controller.callExtensionMethod (/zigbee2mqtt-1.12.2/lib/controller.js:336:44)
    at Controller.onZigbeeEvent (/zigbee2mqtt-1.12.2/lib/controller.js:222:14)
    at Zigbee.emit (events.js:223:5)
(node:229) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 53)
```

Koenkk, 2020-04-26:
@ARoozitalab this bug has already been fixed in the latest dev branch. (edge for hassio).

ARoozitalab, 2020-05-01:
thanks. I install zigbee2mqtt-edge but I have the same problem. my new config on device.js back to default after hassio reboot or power off-on

ARoozitalab, 2020-05-06:
hi, I installed hassio and zigbee2mqtt-edge addon again, but I have the same problem.  device.js back to default after hassio reboot or power off-on and each time I should add my device config on device.js again.

Koenkk, 2020-05-06:
Does it work correctly with the following config?

```js
{
    zigbeeModel: ['374b2e283a7147609956d4412615afc0'],
    model: 'T11W1LZ',
    vendor: 'Orvibo',
    description: 'swith 1 state', 
    supports: 'on/off', 
    fromZigbee: [fz.on_off], 
    toZigbee: [tz.on_off],
},
```

If yes, I will make it supported by default.

closed 2020-07-12 by stale[bot]