Koenkk/zigbee2mqtt #3326: Can not create group for IKEA switch (and light bulb)

brean, 2020-04-10:
# Bug Report
I wanted to setup a group of my IKEA light switch and bulb, but it fails at startup.
I can however see and control light bulb and switch in home assistant using zigbee2mqtt

## What happened
When starting I see this error message:
`zigbee2mqtt:error 2020-04-10 16:49:04: Failed to call 'Groups' 'onZigbeeStarted' (Error: Command 0xd0c.../1 genGroups.add({"groupid":1,"groupname":""}, {"timeout":6000,"manufacturerCode":null,"disableDefaultResponse":true}) failed (Error: AREQ - AF - dataConfirm after 5000ms)`

## What did you expect to happen
Both devices should end up in a group, so when I press the switch the light should toggle.

## How to reproduce it (minimal and precise)
```
homeassistant: true
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://mqtt'
serial:
  port: /dev/ttyACM0
advanced:
  log_level: debug
devices:
  '0xd0c...':
    friendly_name: 'living_room_switch'
  '0x90...':
    friendly_name: 'living_room_light_back'
    
groups:
  # ID, each group should have a different numerical ID
  '1':
    # Required: Name which will be used to control the group
    friendly_name: group_1
    # Required: Retain messages (true/false) (default: false)
    devices:
      - '0xd0c...'
      - '0x90...'
```

Docker-compose file and additional config at https://github.com/brean/docker-compose-hass

## Debug Info
zigbee2mqtt version: 1.12.2 (commit #7d27a54) (using docker)
CC253X firmware version: {"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}}'

Koenkk, 2020-04-11:
Can you try with the latest dev branch, in case it doesn't work please provide the herdsman debug logging.

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

brean, 2020-04-11:
Thanks for the quick response,
with the "latest-dev"-docker image its nearly the same, I see this in the logs:
`
zigbee2mqtt       | zigbee2mqtt:error 2020-04-11 16:06:32: Failed to call 'Groups' 'onZigbeeStarted' (Error: Command 0xd0cf.../1 genGroups.add({"groupid":1,"groupname":""}, {"timeout":10000,"manufacturerCode":null,"disableDefaultResponse":true}) failed (Error: Timeout - 45496 - 1 - 2 - 4 - 0 after 10000ms)
`
full herdsman log here: https://pastebin.com/4hJThvHm

Koenkk, 2020-04-12:
> I wanted to setup a group of my IKEA light switch and bulb, but it fails at startup.

What you probably want to do is setup the IKEA remote to directly control the bulb. This is not done by putting them in the same group but with binding: https://www.zigbee2mqtt.io/information/binding.html

brean, 2020-04-12:
You are right! Binding is what I want but binding also doesn't work:
> zigbee2mqtt       | zigbee2mqtt:error 2020-04-12 10:10:27: Failed to bind cluster 'genOnOff' from 'living_room_switch' to 'living_room_light_back' (Error: Bind 0xd0cf.../1 genOnOff from '0x90f.../1' failed (Error: AREQ - ZDO - bindRsp after 10000ms))

Full herdsman log here: https://pastebin.com/ND1dGUeb

Koenkk, 2020-04-12:
Did you wake up the device right before binding it? (by pressing some buttons on it)

brean, 2020-04-12:
Yes, I pressed some buttons before.
When I remove the battery of the switch for a quick moment and run the mqtt-binding command again I don't see any errors in the log but also the buttons don't work (controlling the lights in home assistant works however).

Koenkk, 2020-04-12:
I forgot that this device does not support binding to a device but only a group. 
It also requires a recent firmware.

- https://www.zigbee2mqtt.io/devices/E1524_E1810.html#binding
- https://www.zigbee2mqtt.io/information/ota_updates.html

brean, 2020-04-12:
Binding to a group also doesn't work. I don't know the firmware version of my switch but when I try the OTA-update I am getting 2 errors:
> zigbee2mqtt       | (node:16) UnhandledPromiseRejectionWarning: Error: Timeout - 45496 - 1 - null - 25 - 1 after 30000ms
zigbee2mqtt       |     at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
zigbee2mqtt       |     at listOnTimeout (internal/timers.js:549:17)
zigbee2mqtt       |     at processTimers (internal/timers.js:492:7)
zigbee2mqtt       | (node:16) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)

and a bit later 

> zigbee2mqtt       | zigbee2mqtt:error 2020-04-12 10:40:55: Failed to check if update available for 'living_room_switch' (Device didn't respond to OTA request)
zigbee2mqtt       | zigbee2mqtt:info  2020-04-12 10:40:55: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"ota_update","message":"Failed to check if update available for 'living_room_switch' (Device didn't respond to OTA request)","meta":{"status":"check_failed","device":"living_room_switch"}}'

brean, 2020-04-12:
I just re-read https://www.zigbee2mqtt.io/devices/E1524_E1810.html
the workaround for older firmware worked!

But I got some exceptions, don't know if you are interested in investigating:
> zigbee2mqtt       | 2020-04-12T08:49:47.175Z zigbee-herdsman:adapter:zStack:znp:error Error while parsing to ZpiObject 'Error: CommandID '184' from subsystem '5' not found
zigbee2mqtt       |     at Function.fromUnpiFrame (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/zpiObject.js:44:19)
zigbee2mqtt       |     at Znp.onUnpiParsed (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:76:48)
zigbee2mqtt       |     at Parser.emit (events.js:310:20)
zigbee2mqtt       |     at Parser.parseNext (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/unpi/parser.js:47:26)
zigbee2mqtt       |     at Parser._transform (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/unpi/parser.js:26:14)
zigbee2mqtt       |     at Parser.Transform._read (_stream_transform.js:191:10)
zigbee2mqtt       |     at Parser.Transform._write (_stream_transform.js:179:12)
zigbee2mqtt       |     at doWrite (_stream_writable.js:442:12)
zigbee2mqtt       |     at writeOrBuffer (_stream_writable.js:426:5)
zigbee2mqtt       |     at Parser.Writable.write (_stream_writable.js:317:11)'

and 

zigbee2mqtt       | 2020-04-12T08:49:57.141Z zigbee-herdsman:controller:endpoint Command 0xd0c.../1 genGroups.add({"groupid":901,"groupname":""}, {"timeout":10000,"manufacturerCode":null,"disableDefaultResponse":true}) failed (Error: Timeout - 45496 - 1 - 4 - 4 - 0 after 10000ms)
zigbee2mqtt       | zigbee2mqtt:error 2020-04-12 10:49:57: Failed to call 'Groups' 'onZigbeeStarted' (Error: Command 0xd0cf5efffe14677b/1 genGroups.add({"groupid":901,"groupname":""}, {"timeout":10000,"manufacturerCode":null,"disableDefaultResponse":true}) failed (Error: Timeout - 45496 - 1 - 4 - 4 - 0 after 10000ms)
zigbee2mqtt       |     at Endpoint.<anonymous> (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:349:23)
zigbee2mqtt       |     at Generator.throw (<anonymous>)
zigbee2mqtt       |     at rejected (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65))
zigbee2mqtt       | 2020-04-12T08:49:57.150Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100]
zigbee2mqtt       | 2020-04-12T08:49:57.151Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100]
zigbee2mqtt       | 2020-04-12T08:49:57.151Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100
zigbee2mqtt       | 2020-04-12T08:49:57.151Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0}
zigbee2mqtt       | 2020-04-12T08:49:57.152Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []

brean, 2020-04-12:
Interesting side-effect: when I press the switch now the light changes, I see that zigbee2mqtt publishes the toggle action but home-assistant still thinks the light bulb is off... Is there a known way around this or do I have to hack something that tricks home assistant in updating the status? 😅😉

brean, 2020-04-12:
I just looked into the Ikea-App, according to that my remote should have the 2.3.014-firmware...

Koenkk, 2020-04-12:
For status reports use the report feature: https://www.zigbee2mqtt.io/information/report.html

brean, 2020-04-13:
Works very nice! Thank you so much!

closed 2020-04-13 by brean