Koenkk/zigbee2mqtt #3328: CC-2531

zambujal, 2020-04-10:
# Bug Report
## What happened

## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug Info
zigbee2mqtt version:
CC253X firmware version:

jperrin31, 2020-04-12:
Hi,
It is the same for me with the upgrade.
Device is created in /dev but module zigbee2mqtt don’t start and there are no logs
Cool I am not alone !

s-hutter, 2020-04-14:
Had the same problem here. This worked for me:
Copy your zigbee2mqtt configuration into a text file, uninstall the Zigbee2MQTT add-on, then reinstall it. Paste your old configuration, click save and finally start the addon.
Problem seems to be fixed now.

jperrin31, 2020-04-14:
You are right, I uninstalled / reinstalled zigbee2mqtt keeping a backup of zigbee2mqtt folder and it is Ok
Thanks for your help 
Regards

zambujal, 2020-04-15:
I will try that! Many thanks!

zambujal, 2020-04-17:
It's working now... I lost allmoust everything...I made several attempts and mistakes... I thought it was the cc2531 malfunction... it's working now and I have 3 more cc 2531 in hand... :D .what hapenned don't know... I disabled the update on... this is working and very well now... Thank you Koenkk !

zambujal, 2020-04-17:
I have an central station in my aquarium (mostly arduino sensores), they send the data via mqtt (with an shield esp 8266). the esp 8266 communicates with the arduino via COM ... the raspberry colects it all (the mosquito mqtt server) and then to the home assistant. The only sensor that I made that doesn't work is the pH too much electricity in the water... The aquarium has 1000 liters , 600W of leds (controlled via the arduino) made by me... there is security sensors that I put there and it's the homeassistant that manages them... like... water in the floor... In the house I had put mostly zigbee sensors...  I have more than 20 zigbee sensors in the total...I think .

closed 2020-06-26 by stale[bot]