Koenkk/zigbee2mqtt #3330: Ikea LED1738G7 get recognized as LED1733G7

swissbyte, 2020-04-10:
#3172 Bug Report
## What happened
I have connected, or tried to connect, a Ikea Tradfri LED1738G7 Bulb. 
The bulb gets recognized as LED1733G7 and is not working. 
The bulb keeps pulsing after the interview of the CC2531
The Bulb is placed 5cm away from the ZNP

Could be related to #3144 

## What did you expect to happen
I expected to get the Bulb recognized correctly

## How to reproduce it (minimal and precise)

Maybe someone has an idea, what could happened here. 

## Debug Info
`zigbee2mqtt:info  2020-04-10 21:29:02: Device '0x086bd7fffe2f1b93' joined
zigbee2mqtt:info  2020-04-10 21:29:02: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x086bd7fffe2f1b93"}}'
zigbee2mqtt:info  2020-04-10 21:29:02: Starting interview of '0x086bd7fffe2f1b93'
zigbee2mqtt:info  2020-04-10 21:29:02: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x086bd7fffe2f1b93"}}'
zigbee2mqtt:info  2020-04-10 21:29:05: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"0x086bd7fffe2f1b93"}}'
zigbee2mqtt:info  2020-04-10 21:29:06: MQTT publish: topic 'homeassistant/light/0x086bd7fffe2f1b93/light/config', payload '{"brightness":true,"color_temp":true,"schema":"json","command_topic":"zigbee2mqtt/0x086bd7fffe2f1b93/set","state_topic":"zigbee2mqtt/0x086bd7fffe2f1b93","json_attributes_topic":"zigbee2mqtt/0x086bd7fffe2f1b93","name":"0x086bd7fffe2f1b93_light","unique_id":"0x086bd7fffe2f1b93_light_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x086bd7fffe2f1b93"],"name":"0x086bd7fffe2f1b93","sw_version":"Zigbee2mqtt 1.12.0","model":"TRADFRI LED bulb E14 600 lumen, dimmable, white spectrum, opal white (LED1733G7)","manufacturer":"IKEA"},"availability_topic":"zigbee2mqtt/bridge/state"}'
zigbee2mqtt:info  2020-04-10 21:29:06: MQTT publish: topic 'homeassistant/sensor/0x086bd7fffe2f1b93/linkquality/config', payload '{"icon":"mdi:signal","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}","state_topic":"zigbee2mqtt/0x086bd7fffe2f1b93","json_attributes_topic":"zigbee2mqtt/0x086bd7fffe2f1b93","name":"0x086bd7fffe2f1b93_linkquality","unique_id":"0x086bd7fffe2f1b93_linkquality_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x086bd7fffe2f1b93"],"name":"0x086bd7fffe2f1b93","sw_version":"Zigbee2mqtt 1.12.0","model":"TRADFRI LED bulb E14 600 lumen, dimmable, white spectrum, opal white (LED1733G7)","manufacturer":"IKEA"},"availability_topic":"zigbee2mqtt/bridge/state"}'
zigbee2mqtt:info  2020-04-10 21:29:06: MQTT publish: topic 'homeassistant/binary_sensor/0x086bd7fffe2f1b93/update_available/config', payload '{"payload_on":true,"payload_off":false,"value_template":"{{ value_json.update_available}}","state_topic":"zigbee2mqtt/0x086bd7fffe2f1b93","json_attributes_topic":"zigbee2mqtt/0x086bd7fffe2f1b93","name":"0x086bd7fffe2f1b93_update_available","unique_id":"0x086bd7fffe2f1b93_update_available_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x086bd7fffe2f1b93"],"name":"0x086bd7fffe2f1b93","sw_version":"Zigbee2mqtt 1.12.0","model":"TRADFRI LED bulb E14 600 lumen, dimmable, white spectrum, opal white (LED1733G7)","manufacturer":"IKEA"},"availability_topic":"zigbee2mqtt/bridge/state"}'
zigbee2mqtt:error 2020-04-10 21:29:31: Failed to interview '0x086bd7fffe2f1b93', device has not successfully been paired`

Koenkk, 2020-04-11:
- Is the CC2531 connected through a USB extension cable? 
- Can you try the following firmware: https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing

swissbyte, 2020-04-12:
Hi Koenkk 

Thank you so much for your answer. First of all i want to say thank you for all these hours you have invested into this project. 

I have flashed my CC2531 with the above mentioned firmware. The CC2531 is not connected through an extension cable. It is connected directly. 

The Bulb now works perfectly. But... it gets detected as LED1733G7 instead of LED1738G7
https://pastebin.com/1a4yWujR
What is the difference or what has changed in the above mentioned firmware?

Just as a sidenote. 

By the way, i saw that there is an upcoming support for the CC2538. 
Would you reccommend to switch to an CC2538 or CC2652?

Again, thank you so much for your engagement. I highly appreciate it.

Koenkk, 2020-04-12:
I would recommend the CC2652 in case you are having stability issues and have a larger network 25/30+ devices.

Can you share your `database.db`? I think this is just a cosmetic issue and can easily be fixed.

swissbyte, 2020-04-14:
Sure. Sorry for the delay. 
But im not exactly sure where the database.db is located in my hass.io zigbee2mqtt docker container
I will look for it. But the HW/SW is now working since 2 days.

Koenkk, 2020-04-14:
@swissbyte I don't use hassio but it should have a data directory, if this doesn't help please ask this question here: https://github.com/danielwelch/hassio-zigbee2mqtt

closed 2020-04-23 by swissbyte