Koenkk/zigbee2mqtt #3333: Light group brightness 1% after turn on

sti0, 2020-04-11:
# Bug Report
## What happened
On triggering a simple light.turn_on service without brightness setting sometimes the lights which are grouped in zigbee2mqtt are turned on with 1% brightness. But the brightness was never this low in a former state. If this happens e.g. to Hue bulbs it seems that they off while they on.

## What did you expect to happen
Former brightness level

## How to reproduce it (minimal and precise)
I can't reproduce this behavior. It works most of the time but sometimes it doesn't.
## Debug Info
zigbee2mqtt version: 1.12.2
CC26X2R1 firmware version: https://github.com/Koenkk/zigbee-herdsman-converters/issues/715#issuecomment-605680466

groenmarsmannetje, 2020-04-13:
Do you use transition? I experienced this earlier too. Had to turn off transition. Evasie it was very annoying.

AngelFreak, 2020-04-15:
I have the same experience on my GLEDOPTO GL-B-007Z, on zigbee2mqtt version: 1.12.2

sti0, 2020-04-15:
> Do you use transition? I experienced this earlier too. Had to turn off transition. Because it was very annoying.

Thanks for the hint. I'm using the transition setting on all the affected bulbs and/or the affected groups. I will try to remove this setting and will report if the problem is solved - or let's say the bug is found...

Koenkk, 2020-04-15:
Please share herdsman debug logging

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

sti0, 2020-04-15:
Added the configuration entry for debugging. I will report if the issue happens again.

MrSabel, 2020-04-16:
See attached logging
[logging.txt](https://github.com/Koenkk/zigbee2mqtt/files/4486170/logging.txt)

Scenario:
- "werkkamer" (Group) has a `brightness` of `178`
- 11:29:46 set `brightness` to `255`
- 11:29:47 set `state` to `on`
- "werkkamer" lights where set to `178`

Group "werkkamer" contains 3 lights, "werkkamer1", "werkkamer2", "werkkamer3"
In this case, the brightness was not set to `1` but `178`.

Koenkk, 2020-04-17:
@MrSabel I see the issue but cannot reproduce it.

- Please reproduce the issue after removing the group from `data/state.json` and a fresh start of zigbee2mqtt 
- Make sure you are on the latest zigbee2mqtt dev branch
- Logging with `DEBUG=zigbee-herdsman:controller*` is enough.

marrold, 2020-04-19:
I experience the same issue, using groups of Tradfri devices bound to a Tradfri switch.

If I turn the lights on / off with HA and sending brightness, everything is fine. If I turn them off with HA, then _on_ with the switch, they come on at 0% brightness.

I don't really know when it started as I did a big update with a few versions of HA and Z2M 

I tried running the dev branch (edge add-on in hassio) but it wasn't working correctly, like it was ignoring my existing configuration.yaml. I am on the latest stable versions of hassio and zigbee2mqtt.

Koenkk, 2020-04-20:
@marrold do you turn them off with a transition?

marrold, 2020-04-21:
No, just `light.turn_off` 

So, the bulb is on, HA sends `light.turn_off` to the _group_ with no attributes and the light turns off. If I then use a switch bound to the group, they come back on with brightness 1.

Thanks

sti0, 2020-04-21:
It happend to me yesterday again:

```
debug 2020-04-20 06:05:29: Received Zigbee message from 'lgt_buro_schreibtischlampe', type 'readResponse', cluster 'genBasic', data '{"zclVersion":1}' from endpoint 11 with groupID 0
debug 2020-04-20 06:05:29: Successfully pinged 'lgt_buro_schreibtischlampe'
debug 2020-04-20 06:05:41: Received Zigbee message from 'lgt_buro_stehlampe', type 'readResponse', cluster 'genBasic', data '{"zclVersion":1}' from endpoint 11 with groupID 0
debug 2020-04-20 06:05:41: Successfully pinged 'lgt_buro_stehlampe'
debug 2020-04-20 06:05:44: Received Zigbee message from 'rmt_buro', type 'commandToggle', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 8336
info  2020-04-20 06:05:44: MQTT publish: topic 'zigbee2mqtt/rmt_buro/action', payload 'toggle'
info  2020-04-20 06:05:44: MQTT publish: topic 'zigbee2mqtt/rmt_buro', payload '{"linkquality":123,"last_seen":1587355544393,"battery":47,"update_available":true,"action":"toggle"}'
info  2020-04-20 06:05:44: MQTT publish: topic 'zigbee2mqtt/rmt_buro', payload '{"linkquality":123,"last_seen":1587355544393,"battery":47,"update_available":true,"action":""}'
debug 2020-04-20 06:05:44: Received MQTT message on 'zigbee2mqtt/grp_buro/set' with data '{"state": "ON"}'
debug 2020-04-20 06:05:44: Publishing 'set' 'state' to 'grp_buro'
info  2020-04-20 06:05:44: MQTT publish: topic 'zigbee2mqtt/lgt_buro_schreibtischlampe', payload '{"state":"ON","last_seen":1587355529453,"brightness":1,"color_temp":366,"linkquality":96,"update_available":false}'
info  2020-04-20 06:05:44: MQTT publish: topic 'zigbee2mqtt/grp_buro', payload '{"state":"ON","brightness":1,"color_temp":366}'
info  2020-04-20 06:05:44: MQTT publish: topic 'zigbee2mqtt/lgt_buro_stehlampe', payload '{"state":"ON","last_seen":1587355541623,"brightness":1,"color_temp":366,"linkquality":126,"update_available":false}'
info  2020-04-20 06:05:44: MQTT publish: topic 'zigbee2mqtt/lgt_buro_strahler', payload '{"state":"ON","last_seen":1587355472379,"brightness":1,"color_temp":366,"linkquality":108,"update_available":false}'
debug 2020-04-20 06:05:47: Received Zigbee message from 'rmt_buro', type 'commandToggle', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 8336
info  2020-04-20 06:05:47: MQTT publish: topic 'zigbee2mqtt/rmt_buro/action', payload 'toggle'
info  2020-04-20 06:05:47: MQTT publish: topic 'zigbee2mqtt/rmt_buro', payload '{"linkquality":126,"last_seen":1587355547073,"battery":47,"update_available":true,"action":"toggle"}'
info  2020-04-20 06:05:47: MQTT publish: topic 'zigbee2mqtt/rmt_buro', payload '{"linkquality":126,"last_seen":1587355547073,"battery":47,"update_available":true,"action":""}'
debug 2020-04-20 06:05:51: Received MQTT message on 'zigbee2mqtt/grp_buro/set' with data '{"state": "OFF"}'
debug 2020-04-20 06:05:51: Publishing 'set' 'state' to 'grp_buro'
info  2020-04-20 06:05:51: MQTT publish: topic 'zigbee2mqtt/lgt_buro_schreibtischlampe', payload '{"state":"OFF","last_seen":1587355529453,"brightness":1,"color_temp":366,"linkquality":96,"update_available":false}'
info  2020-04-20 06:05:51: MQTT publish: topic 'zigbee2mqtt/grp_buro', payload '{"state":"OFF","brightness":1,"color_temp":366}'
info  2020-04-20 06:05:51: MQTT publish: topic 'zigbee2mqtt/lgt_buro_stehlampe', payload '{"state":"OFF","last_seen":1587355541623,"brightness":1,"color_temp":366,"linkquality":126,"update_available":false}'
info  2020-04-20 06:05:51: MQTT publish: topic 'zigbee2mqtt/lgt_buro_strahler', payload '{"state":"OFF","last_seen":1587355472379,"brightness":1,"color_temp":366,"linkquality":108,"update_available":false}'
debug 2020-04-20 06:05:59: Received Zigbee message from 'rmt_buro', type 'commandToggle', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 8336
info  2020-04-20 06:05:59: MQTT publish: topic 'zigbee2mqtt/rmt_buro/action', payload 'toggle'
info  2020-04-20 06:05:59: MQTT publish: topic 'zigbee2mqtt/rmt_buro', payload '{"linkquality":126,"last_seen":1587355559512,"battery":47,"update_available":true,"action":"toggle"}'
info  2020-04-20 06:05:59: MQTT publish: topic 'zigbee2mqtt/rmt_buro', payload '{"linkquality":126,"last_seen":1587355559512,"battery":47,"update_available":true,"action":""}'
debug 2020-04-20 06:05:59: Received MQTT message on 'zigbee2mqtt/grp_buro/set' with data '{"state": "ON"}'
debug 2020-04-20 06:05:59: Publishing 'set' 'state' to 'grp_buro'
info  2020-04-20 06:05:59: MQTT publish: topic 'zigbee2mqtt/lgt_buro_schreibtischlampe', payload '{"state":"ON","last_seen":1587355529453,"brightness":1,"color_temp":366,"linkquality":96,"update_available":false}'
info  2020-04-20 06:05:59: MQTT publish: topic 'zigbee2mqtt/grp_buro', payload '{"state":"ON","brightness":1,"color_temp":366}'
info  2020-04-20 06:05:59: MQTT publish: topic 'zigbee2mqtt/lgt_buro_stehlampe', payload '{"state":"ON","last_seen":1587355541623,"brightness":1,"color_temp":366,"linkquality":126,"update_available":false}'
info  2020-04-20 06:05:59: MQTT publish: topic 'zigbee2mqtt/lgt_buro_strahler', payload '{"state":"ON","last_seen":1587355472379,"brightness":1,"color_temp":366,"linkquality":108,"update_available":false}'
debug 2020-04-20 06:06:04: Received MQTT message on 'zigbee2mqtt/lgt_buro_schreibtischlampe/set' with data '{"state": "ON", "brightness": 136}'
debug 2020-04-20 06:06:04: Publishing 'set' 'brightness' to 'lgt_buro_schreibtischlampe'
[... cut ...]
info  2020-04-20 06:06:08: MQTT publish: topic 'zigbee2mqtt/lgt_buro_stehlampe', payload '{"state":"ON","last_seen":1587355568606,"brightness":136,"color_temp":366,"linkquality":126,"update_available":false}'
```

I pressed the toggle button on an IKEA remote and which triggers a `light.turn_on` action in Home Assistant. The affected z2m group consists by three Hue white ambiance bulbs. After turning on, the lights are triggered with brightness 1. After some on and off I changed the brightness in HA. At z2m configuration I had set a transition on both the device and group config (`transition: 0.4`)

Koenkk, 2020-04-21:
@marrold @sti0 please try to provide the following; https://github.com/Koenkk/zigbee2mqtt/issues/3333#issuecomment-615386832

chris-jennings, 2020-04-30:
Has this started happening since people are completed OTA updates for their bulbs? That is what I have noticed, although this required me to update zigbee2mqtt so I can't be sure.

groenmarsmannetje, 2020-04-30:
I had the issue after the transition on turning off/on was introduced. Workaround for me then was to remove all transitions. But since version 1.12 it came back, while I still do not use transitions. No clear reason why it happens, because it is not all the time. The IKEA bulbs are all updated who have this behavior, but I also have some Trust bulbs in same group and these do not turn on while the IKEA bulbs turn on at 1% when the whole group is turned on.

marrold, 2020-05-03:
For what it's worth I have not intentionally performed an OTA update on any of my affected bulbs.

marrold, 2020-06-02:
@Koenkk Please could you let me know how to run zigbee2mqtt with `DEBUG=zigbee-herdsman:controller*` when using Hass.io / supervised ? 

Thanks

marrold, 2020-06-02:
Someone on IRC helped with the above.

lounge_light is a group of Tradfri LED1732G11 bulbs ((0x000d6ffffe5d683d, 0x14b457fffe3d85a8) bound to a E1743 switch (0x14b457fffed4ca69)

If switched off with HA and switched on with the bound switch, the group switches back on at 1% brightness.

23:35:10 - lounge_light switched off via HA overview
23:35:50 - lounge_light switched on via HA overview
23:36:20 - lounge_light switched off via HA overview
23:36:40 - lounge_light switched on via HA lounge switch 1 (0x14b457fffed4ca69), brightness is 1%

I've attached the log, however it's not showing as much debug output as the console. Hopefully it helps.

[z2m_log.txt](https://github.com/Koenkk/zigbee2mqtt/files/4720083/z2m_log.txt)

Thanks!

marrold, 2020-06-04:
Config below. I've experimented with transition enabled / disabled (Both at group level and triggered from HA) but the end result was the same:

configuration.yaml

```
{
  "devices": "devices.yaml",
  "groups": "groups.yaml",
  "homeassistant": true,
  "permit_join": false,
  "mqtt": {
    "base_topic": "zigbee2mqtt",
    "server": "mqtt://core-mosquitto",
    "user": "zigbee2mqtt",
    "password": "REDACTED"
  },
  "serial": {
    "port": "/dev/ttyACM0"
  },
  "advanced": {
    "pan_id": 6756,
    "channel": 11,
    "network_key": [
      5,
      5,
      5,
      4,
      9,
      11,
      13,
      15,
      0,
      1,
      1,
      6,
      8,
      10,
      12,
      13
    ],
    "availability_blacklist": [],
    "last_seen": "ISO_8601",
    "report": true
  },
  "ban": [],
  "whitelist": [],
  "queue": {},
  "zigbee_herdsman_debug": false
}
```

groups.yaml

```
'1':
    # Required: Name which will be used to control the group
    friendly_name: 'balcony_lights'
    # Required: Retain messages (true/false) (default: false)
    retain: false
    # Optional: Default transition to be used when e.g. changing brightness (in seconds) (default: 0)
    transition: 0
    # Optional: Change group state when one of the devices in it changes state, see 'State changes' below (default: true)
    optimistic: false
    # Optional: Devices of this group,
    # Note: this has to be the ieeeAddr of the device, not the friendly_name! (default: empty)
    devices:
      # Balcony Light 1
      - '0x14b457fffe2d8495'
      # Balcony Light 2
      - '0xccccccfffe3b5832' 

'2':
    # Required: Name which will be used to control the group
    friendly_name: 'hall_lights'
    # Required: Retain messages (true/false) (default: false)
    retain: false
    # Optional: Default transition to be used when e.g. changing brightness (in seconds) (default: 0)
    transition: 0
    # Optional: Change group state when one of the devices in it changes state, see 'State changes' below (default: true)
    optimistic: false
    # Optional: Devices of this group,
    # Note: this has to be the ieeeAddr of the device, not the friendly_name! (default: empty)
    devices:
      # friendly_name: Hall Light 1
      - '0x680ae2fffe16c465'
      # friendly_name: Hall Light 2
      - '0x14b457fffe2c4bb1'
      # friendly_name: Hall Light 3 
      - '0x14b457fffe2eb0c4'
      
'3':
    # Required: Name which will be used to control the group
    friendly_name: 'kitchen_lights'
    # Required: Retain messages (true/false) (default: false)
    retain: false
    # Optional: Default transition to be used when e.g. changing brightness (in seconds) (default: 0)
    transition: 0
    # Optional: Change group state when one of the devices in it changes state, see 'State changes' below (default: true)
    optimistic: false
    # Optional: Devices of this group,
    # Note: this has to be the ieeeAddr of the device, not the friendly_name! (default: empty)
    devices:
      # Kitchen Light 1
      - '0x14b457fffe760a5f'
      # Kitchen Light 2
      - '0x14b457fffe7f4975'
      # Kitchen Light 3
      - '0x14b457fffec56974'
      # Kitchen Light 4
      - '0x14b457fffed3c2a8'
      # Kitchen Light 5
      - '0x000d6ffffe609d2e'
      # Kitchen Light 6
      - '0x14b457fffe7f4c14'
      # Kitchen Extractor 1
      - '0x000d6ffffe597301'
      # Kitchen Extractor 2
      - '0x086bd7fffe21a538'
      # Kitchen Cabinet Lights 1
      - '0x680ae2fffe8ca1a2'
  
'4':
    # Required: Name which will be used to control the group
    friendly_name: 'lounge_lights'
    # Required: Retain messages (true/false) (default: false)
    retain: false
    # Optional: Default transition to be used when e.g. changing brightness (in seconds) (default: 0)
    transition: 1
    # Optional: Change group state when one of the devices in it changes state, see 'State changes' below (default: true)
    optimistic: true
    # Optional: Devices of this group,
    # Note: this has to be the ieeeAddr of the device, not the friendly_name! (default: empty)
    devices:
      # Lounge Light 1
      - '0x000d6ffffe5d683d'
      # Lounge Light 2
      - '0x14b457fffe3d85a8'
```

Thanks

Koenkk, 2020-06-06:
After removing all `transition` from your groups.yaml and restarting zigbee2mqtt, does this still occur?

marrold, 2020-06-06:
Ah, I had only tried setting it to 0 / 1 - I hadn't tried removing it.

I've now commented out all the transition lines, and it _looks_ like that's solved the issue. I'd have to wait a few more days to be sure.

Thanks for your help with this.

marrold, 2020-06-10:
Things have stable for a few days so I think it's safe to assume the `transistion` parameter is the problem. 

Thanks

Koenkk, 2020-06-10:
Good that this fixes it.

The reason why it happens:
Zigbee bulbs don't support custom time on/off transitions (but do for changing brightness). So as a workaround, instead of sending just off, we send a command to lower the brightness to 0%. When turning the bulb on again (from Zigbee2mqtt) we know that the bulb is still at 0% brightness, so we have restore it to the previous set brightness (before turning it off). The remote of course doesn't know this and just sends an on command, resulting in the bulbs turned on at the lowest brightness possible. Therefore this is something that cannot be fixed.

I assume this issue can be closed now.

closed 2020-06-10 by Koenkk

877dev, 2021-04-10:
> The reason why it happens:
> Zigbee bulbs don't support custom time on/off transitions (but do for changing brightness). So as a workaround, instead of sending just off, we send a command to lower the [...]

Glad I found this post, I just spent 1 hour checking all my node red flows for some error.

Deleting the transition for the group fixed for me, can this be added to docs in the groups section?