Koenkk/zigbee2mqtt #3335: Failed to call 'BridgeLegacy' after publish a touchlink/factory_reset

jcampoy, 2020-04-11:
# Bug Report
## What happened

I am trying a factory_reset of a Philips bulb via touch link, nothing happens and I am having errors in the log file: *zigbee2mqtt:error Failed to call 'BridgeLegacy'*

Log file: https://pastebin.com/APTQ1YuH

## What did you expect to happen

No error.

## How to reproduce it (minimal and precise)
Publish the zigbee2mqtt/bridge/config/touchlink/factory_reset with the versions below.

## Debug Info
**zigbee2mqtt version:** zigbee2mqtt-1.12.2
**CC253X firmware version:**
"coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190109}}

jcampoy, 2020-04-11:
BTW I got the same error in a zigbee2mqtt restart, so, it seems not to be only related with the touchlink factory reset:

https://pastebin.com/3W7CJfDN

Koenkk, 2020-04-12:
You are using an old firmware (`20190109`), first upgrade to the latest: https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing

IVI053, 2020-04-12:
Already posted in another issue and just found this one. So for reference: https://github.com/Koenkk/zigbee2mqtt/issues/2396#issuecomment-612676083
Got the same error with CC26X2R1 adapter with dev z-stack (https://github.com/Koenkk/Z-Stack-firmware/tree/develop/coordinator/Z-Stack_3.x.0/bin) and 1.12.2

closed 2020-06-20 by stale[bot]