Koenkk/zigbee2mqtt #3337: Xiaomi QBKG11LM not showing as Router?

derekoharrow, 2020-04-11:
I have this light switch which previously worked fine and was showing as a Routing node. Now, for some reason, this is not showing as a router - just an end node.

How can I get this recognised correctly as a router?

I've removed the device from my network and re-added it, and it is correctly identified as a QBKG11LM.

```
info  2020-04-11 10:58:43: Removing 'Cabin_Lights'
error 2020-04-11 10:58:53: Failed to remove Cabin_Lights (Error: AREQ - ZDO - mgmtLeaveRsp after 10000ms)
info  2020-04-11 10:58:53: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_removed_failed","message":"Cabin_Lights"}'
info  2020-04-11 10:59:37: MQTT publish: topic 'zigbee2mqtt/Cabin_Lights', payload '{"power":102.35,"consumption":18.61,"temperature":24,"linkquality":0,"state":"ON","last_seen":"2020-04-11T09:59:36.586Z","click":"single"}'
info  2020-04-11 11:00:18: Device 'Cabin_Lights' joined
info  2020-04-11 11:00:18: Starting interview of 'Cabin_Lights'
info  2020-04-11 11:00:18: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"Cabin_Lights"}}'
info  2020-04-11 11:00:18: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"Cabin_Lights"}}'
info  2020-04-11 11:00:28: Successfully interviewed 'Cabin_Lights', device has successfully been paired
info  2020-04-11 11:00:28: Device 'Cabin_Lights' is supported, identified as: Xiaomi Aqara single key wired wall switch (QBKG11LM)
info  2020-04-11 11:00:28: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"Cabin_Lights","model":"QBKG11LM","vendor":"Xiaomi","description":"Aqara single key wired wall switch","supported":true}}'
```

Zigbee2mqtt is detecting it as:
```
Zigbee Id is 0x00158d0002ec1d9e.
Address on Zigbee Network is 0xBFE9.
Link quality is unknown/255.
Model is lumi.ctrl_ln1.aq1 by LUMI.
Hardware version is unknown.
Firmware version is unknown.
This device type is EndDevice
```

My zigbee2mqtt setup is:
```
Version of Zigbee2MqttAssistant (this software): 0.3.147 (0.3.147+Branch.master.Sha.92d831d1df50a48c4a2fd38c201e9465dc7e88dd) Release
Version of Zigbee2Mqtt: 1.12.2-dev
Coordinator version: 20190608
```

sjorge, 2020-04-11:
```
info  2020-04-11 10:58:43: Removing 'Cabin_Lights'
error 2020-04-11 10:58:53: Failed to remove Cabin_Lights (Error: AREQ - ZDO - mgmtLeaveRsp after 10000ms)
info  2020-04-11 10:58:53: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_removed_failed","message":"Cabin_Lights"}'
info  2020-04-11 10:59:37: MQTT publish: topic 'zigbee2mqtt/Cabin_Lights', payload '{"power":102.35,"consumption":18.61,"temperature":24,"linkquality":0,"state":"ON","last_seen":"2020-04-11T09:59:36.586Z","click":"single"}'
info  2020-04-11 11:00:18: Device 'Cabin_Lights' joined
```

Seems to hit it was never fully removed. Maybe try again, if it errors out, reset the device physically? (I think holding the pair button for 5-10 seconds should do it) Then force remove it from zigbee2mqtt and repair after.

derekoharrow, 2020-04-11:
Took a bit of fiddling, but that seems to have fixed it! Thanks

closed 2020-04-11 by derekoharrow