Koenkk/zigbee2mqtt #3338: QBKG03LM single click latency

funcasta, 2020-04-11:
# Bug Report
## What happened
I installed 2 new QBKG03LM, but the single click takes quite a long time to trigger (maybe one second or more). The old QBKG03LM I have are working as expected.

## What did you expect to happen
Faster click with newer QBKG03LM

## How to reproduce it (minimal and precise)

## Debug Info
zigbee2mqtt version: 1.12.2 
CC253X firmware version: 20190608

I can see in the log file that the old ones are sending "genOnOff" for the single click, but the new ones are always sending "genMultistateInput" (maybe that's why?). Interestingly enough, the new ones are working fast if I use the xiaomi gateway.

With the new one:
`debug 2020-04-11 11:12:59: Received Zigbee message from '0x00158d00032115bc', type 'attributeReport', cluster 'genMultistateInput', data '{"presentValue":1}' from endpoint 1 with groupID 0
info  2020-04-11 11:12:59: MQTT publish: topic 'homeassistant/device_automation/0x00158d00032115bc/click_single/config', payload '{"automation_type":"trigger","type":"click","subtype":"single","payload":"single","topic":"zigbee2mqtt/0x00158d00032115bc/click","device":{"identifiers":["zigbee2mqtt_0x00158d00032115bc"],"name":"0x00158d00032115bc","sw_version":"Zigbee2mqtt 1.12.2","model":"Aqara single key wireless wall switch (WXKG03LM)","manufacturer":"Xiaomi"}}'
info  2020-04-11 11:12:59: MQTT publish: topic 'zigbee2mqtt/0x00158d00032115bc', payload '{"battery":100,"voltage":3025,"linkquality":49,"click":"single"}'
info  2020-04-11 11:12:59: MQTT publish: topic 'zigbee2mqtt/0x00158d00032115bc', payload '{"battery":100,"voltage":3025,"linkquality":49,"click":""}'
info  2020-04-11 11:12:59: MQTT publish: topic 'zigbee2mqtt/0x00158d00032115bc/click', payload 'single'`

With the old one:
`debug 2020-04-11 11:34:52: Received Zigbee message from 'salon_cocina_single', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
info  2020-04-11 11:34:52: MQTT publish: topic 'homeassistant/device_automation/0x00158d00015efd1b/click_single/config', payload '{"automation_type":"trigger","type":"click","subtype":"single","payload":"single","topic":"zigbee2mqtt/salon_cocina_single/click","device":{"identifiers":["zigbee2mqtt_0x00158d00015efd1b"],"name":"salon_cocina_single","sw_version":"Zigbee2mqtt 1.12.2","model":"Aqara single key wireless wall switch (WXKG03LM)","manufacturer":"Xiaomi"}}'
info  2020-04-11 11:34:52: MQTT publish: topic 'zigbee2mqtt/salon_cocina_single', payload '{"linkquality":76,"battery":97,"voltage":2995,"click":"single"}'
info  2020-04-11 11:34:52: MQTT publish: topic 'zigbee2mqtt/salon_cocina_single', payload '{"linkquality":76,"battery":97,"voltage":2995,"click":""}'
info  2020-04-11 11:34:52: MQTT publish: topic 'zigbee2mqtt/salon_cocina_single/click', payload 'single'`

Koenkk, 2020-04-12:
- Are you sure its not an interference issue? Is your CC2531 connected through a USB extension cable?
- Do you see `debug 2020-04-11 11:12:59: Received Zigbee message from '0x00158d00032115bc'` the the moment you click on the button or is there already a delay here.

funcasta, 2020-04-12:
1 - Yes, the coordinator is connected through an USB extension, and there's a router one meter away from one of the switches. All other switches (that are further away from the coordinator or the router) are working as expected.

2 - There's already a delay there

Koenkk, 2020-04-12:
@funcasta can you figure out (using sniffing: https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html) if there is also a delay in the device sending the message?

funcasta, 2020-04-12:
I don't have a spare CC2531 right now, but another one is on its way. I will test when I get the new one, although when I pair the WXKG03LM with the Xiaomi gateway, it works fine, so I guess the device is working fine :-\

closed 2020-06-20 by stale[bot]