Koenkk/zigbee2mqtt #3339: Aurora Motion Sensors and 4 Button Remote Support Request

mantis64, 2020-04-11:
Dear @Koenkk , thank you soo much for supporting the lights.  I was wondering if you could help add support for some motion sensors and a 4 button remote I have.  Below is the info I think you need.  

After this, I promise I'll leave you alone for a while :) 

_____________________________________

**Description**
Develco Motion Sensor

**Supported Features**
Occupancy, Temperature and Light

**Pairing**
Press and hold reset button till LED flashes red.  
Will flash every second to indicate it is in paring mode.  Will stop flashing to confirm paired

**Zigbee2Mqtt**
Device '0x0015bc001a0073d4' with Zigbee model 'MOSZB-130
' is NOT supported

**Database**
{"id":55,"type":"EndDevice","ieeeAddr":"0x0015bc001a0073d4","nwkAddr":24148,"manufId":4117,"manufName":"Develco Products A/S","powerSource":"Battery","modelId":"MOSZB-130","epList":[1,34,35,38,39,40,41],"endpoints":{"1":{"profId":49353,"epId":1,"devId":1,"inClusterList":[3,5,6],"outClusterList":[],"clusters":{},"binds":[]},"34":{"profId":260,"epId":34,"devId":263,"inClusterList":[0,3,1030],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"modelId":"MOSZB-130","manufacturerName":"Develco Products A/S","powerSource":3,"zclVersion":1}}},"binds":[]},"35":{"profId":260,"epId":35,"devId":1026,"inClusterList":[0,1,3,15,32,1280],"outClusterList":[10,25],"clusters":{"genBasic":{"attributes":{"modelId":"MOSZB-130","manufacturerName":"Develco Products A/S","powerSource":3,"zclVersion":1}}},"binds":[]},"38":{"profId":260,"epId":38,"devId":770,"inClusterList":[0,3,1026],"outClusterList":[3],"clusters":{"genBasic":{"attributes":{"modelId":"MOSZB-130","manufacturerName":"Develco Products A/S","powerSource":3,"zclVersion":1}}},"binds":[]},"39":{"profId":260,"epId":39,"devId":262,"inClusterList":[0,3,1024],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"modelId":"MOSZB-130","manufacturerName":"Develco Products A/S","powerSource":3,"zclVersion":1}}},"binds":[]},"40":{"profId":260,"epId":40,"devId":263,"inClusterList":[0,3,1030],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"modelId":"MOSZB-130","manufacturerName":"Develco Products A/S","powerSource":3,"zclVersion":1}}},"binds":[]},"41":{"profId":260,"epId":41,"devId":263,"inClusterList":[0,3,1030],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"modelId":"MOSZB-130","manufacturerName":"Develco Products A/S","powerSource":3,"zclVersion":1}}},"binds":[]}},"zclVersion":1,"interviewCompleted":false,"meta":{},"lastSeen":1586434467977}
**Weblink & Picture**
[https://fccid.io/2AHNM-MOSZB130/User-Manual/User-Manual-3694770]
(https://fccid.io/2AHNM-MOSZB130/User-Manual/User-Manual-3694770)

------------------------------------------------------------------

**Description**
Leedarson Motion Sensor 5AA-SS-ZA-H0

**Supported Features**
Occupancy and Light

**Pairing**
Press and hold reset button for 5 seconds

**Zigbee2Mqtt**
Device '0x00158d00017ac453' with Zigbee model 'ZHA-PIRSensor' is NOT supported

**Database**
{"id":58,"type":"EndDevice","ieeeAddr":"0x00158d00017ac453","nwkAddr":19242,"manufId":4456,"manufName":"Leedarson","powerSource":"Battery","modelId":"ZHA-PIRSensor","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":263,"inClusterList":[0,1030,3,1280],"outClusterList":[4,3,25],"clusters":{"genBasic":{"attributes":{"modelId":"ZHA-PIRSensor","manufacturerName":"Leedarson","powerSource":3,"zclVersion":1,"appVersion":1,"stackVersion":2,"hwVersion":1,"dateCode":"20170120-538","swBuildId":"2.0"}}},"binds":[]}},"appVersion":1,"stackVersion":2,"hwVersion":1,"dateCode":"20170120-538","swBuildId":"2.0","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1586537647593}
**Weblink & Picture**
[https://www.leedarson.com/iot.php/Product/detail/id/173#noscroll]
(https://www.leedarson.com/iot.php/Product/detail/id/173#noscroll)

------------------------------------------------------------------

**Description**
Aurora Motion Sensor AU-A1ZBPIRS

I cannot seem to get a zigbee model ID for this.  Will database entry be enough to support this device, do you need more info?

**Supported Features**
Occupancy and Light

**Pairing**
Press and hold reset button for 10 seconds until LED flashes red

**Zigbee2Mqtt**
Received message from unsupported device with Zigbee model 'undefined'

**Database**
{"id":56,"type":"EndDevice","ieeeAddr":"0x0015bc001a010388","nwkAddr":835,"manufId":4117,"epList":[1,34,35,39,40,41],"endpoints":{"1":{"profId":49353,"epId":1,"devId":1,"inClusterList":[3,5,6],"outClusterList":[],"clusters":{},"binds":[]},"34":{"profId":260,"epId":34,"devId":263,"inClusterList":[0,3,1030],"outClusterList":[],"clusters":{},"binds":[]},"35":{"profId":260,"epId":35,"devId":1026,"inClusterList":[0,1,3,15,32,1280],"outClusterList":[10,25],"clusters":{},"binds":[]},"39":{"profId":260,"epId":39,"devId":262,"inClusterList":[0,3,1024],"outClusterList":[],"clusters":{},"binds":[]},"40":{"profId":260,"epId":40,"devId":263,"inClusterList":[0,3,1030],"outClusterList":[],"clusters":{},"binds":[]},"41":{"profId":260,"epId":41,"devId":263,"inClusterList":[0,3,1030],"outClusterList":[],"clusters":{},"binds":[]}},"interviewCompleted":false,"meta":{},"lastSeen":1586480110887}
**Weblink & Picture**
[https://auroralighting.com/gb/ProductDetail/AU-A1ZBPIRS]
(https://auroralighting.com/gb/ProductDetail/AU-A1ZBPIRS)

-------------------------------------------------------------------

**Description**
Aurora 4 Button Aone Smart Remote AU-A1ZBRC

This seems to be a rebrand or an already supported remote from Leedarson.  
https://www.zigbee2mqtt.io/devices/6ARCZABZH.html

**Supported Features**
on/off, brightness up/down and click/hold/release, cct

**Pairing**
Press and hold reset button for 5 seconds

**Zigbee2Mqtt**
Device '0x00158d0002123314' with Zigbee model 'Remote50AU' is NOT supported

**Database**
{"id":59,"type":"EndDevice","ieeeAddr":"0x00158d0002123314","nwkAddr":28346,"manufId":4456,"manufName":"Aurora","powerSource":"Battery","modelId":"Remote50AU","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":2096,"inClusterList":[0,3,4096],"outClusterList":[4,3,6,8,5,4096,25],"clusters":{"genBasic":{"attributes":{"modelId":"Remote50AU","manufacturerName":"Aurora","powerSource":3,"zclVersion":2,"appVersion":1,"stackVersion":1,"hwVersion":1,"dateCode":"20181106","swBuildId":"2.4"}}},"binds":[]}},"appVersion":1,"stackVersion":1,"hwVersion":1,"dateCode":"20181106","swBuildId":"2.4","zclVersion":2,"interviewCompleted":true,"meta":{},"lastSeen":1586538381875}
**Weblink & Picture**
[https://auroralighting.com/gb/ProductDetail/AU-A1ZBRC]
(https://auroralighting.com/gb/ProductDetail/AU-A1ZBRC)

Koenkk, 2020-04-12:
For all devices, please provide the zigbee2mqtt debug logging when triggering actions on the device (occupancy, click, etc.)

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

mantis64, 2020-04-13:
Hi, I'm not sure which parts of the log you need. but have pasted the things I think you need.  If it is too much or too little, I apologise.  If you need me to do anything else, please let me know and I'll try my best.  Hope the info is helpful to enable you to support these devices. 

------------------------------------------------------------

Debug for Aurora 4 Button Aone Smart Remote AU-A1ZBRC

`warn  2020-04-13 10:58:05: Received message from unsupported device with Zigbee model 'Remote50AU'
warn  2020-04-13 10:58:05: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
debug 2020-04-13 10:58:24: Received Zigbee message from '0x00124b000be95914', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 8 with groupID 0
info  2020-04-13 10:58:24: MQTT publish: topic 'zigbee2mqtt/0x00124b000be95914', payload '{"state":true,"led_state":false,"linkquality":39}'
debug 2020-04-13 10:58:31: Saving state to file /share/zigbee2mqtt/state.json
debug 2020-04-13 10:58:32: Received Zigbee message from '0x00124b001b626e57', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
info  2020-04-13 10:58:32: MQTT publish: topic 'zigbee2mqtt/0x00124b001b626e57', payload '{"state":"ON","linkquality":57}'
debug 2020-04-13 10:58:32: Received Zigbee message from '0x00124b001b626e57', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
debug 2020-04-13 10:58:32: Received Zigbee message from '0x00124b001b626e57', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
debug 2020-04-13 10:58:32: Received Zigbee message from '0x00124b001b626e57', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
debug 2020-04-13 10:58:32: Received Zigbee message from '0x00124b001b626e57', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
debug 2020-04-13 10:58:32: Received Zigbee message from '0x00124b001b626e57', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
debug 2020-04-13 10:58:48: Received Zigbee message from '0x00158d0002123314', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"manufacturerCode":4456,"imageType":4609,"fileVersion":557062208}' from endpoint 1 with groupID 0
warn  2020-04-13 10:58:48: Received message from unsupported device with Zigbee model 'Remote50AU'
warn  2020-04-13 10:58:48: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.`

-------------------------------------------------------------------

Debug log for Leedarson Motion Sensor 5AA-SS-ZA-H0

`debug 2020-04-13 11:01:30: Device '0x00158d00017ac453' announced itself
info  2020-04-13 11:01:30: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"0x00158d00017ac453"}}'
debug 2020-04-13 11:01:40: Received Zigbee message from '0x00158d000451ad48', type 'attributeReport', cluster 'msIlluminanceMeasurement', data '{"measuredValue":37}' from endpoint 1 with groupID 0
info  2020-04-13 11:01:40: MQTT publish: topic 'zigbee2mqtt/0x00158d000451ad48', payload '{"battery":100,"voltage":3015,"illuminance":37,"linkquality":57,"occupancy":false,"illuminance_lux":37}'
debug 2020-04-13 11:01:40: Received Zigbee message from '0x00158d000451ad48', type 'attributeReport', cluster 'msOccupancySensing', data '{"occupancy":1}' from endpoint 1 with groupID 0
info  2020-04-13 11:01:40: MQTT publish: topic 'zigbee2mqtt/0x00158d000451ad48', payload '{"battery":100,"voltage":3015,"illuminance":37,"linkquality":60,"occupancy":true,"illuminance_lux":37}'
debug 2020-04-13 11:01:41: Received MQTT message on 'zigbee2mqtt/0x00158d0001801117/set' with data '{"state": "ON", "brightness": 255}'
debug 2020-04-13 11:01:41: Publishing 'set' 'brightness' to '0x00158d0001801117'
info  2020-04-13 11:01:41: MQTT publish: topic 'zigbee2mqtt/0x00158d0001801117', payload '{"state":"ON","color_temp":330,"color":{"x":0.434810358986718,"y":0.403334643366923},"brightness":255}'
debug 2020-04-13 11:01:51: Device '0x00158d00017ac453' announced itself
info  2020-04-13 11:01:51: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"0x00158d00017ac453"}}'
debug 2020-04-13 11:01:56: Received Zigbee message from '0x00158d00017ac453', type 'attributeReport', cluster 'msOccupancySensing', data '{"occupancy":1}' from endpoint 1 with groupID 0
warn  2020-04-13 11:01:56: Received message from unsupported device with Zigbee model 'ZHA-PIRSensor'
warn  2020-04-13 11:01:56: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
debug 2020-04-13 11:01:56: Received Zigbee message from '0x00158d00017ac453', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"zonestatus":17,"extendedstatus":0}' from endpoint 1 with groupID 0
warn  2020-04-13 11:01:56: Received message from unsupported device with Zigbee model 'ZHA-PIRSensor'
warn  2020-04-13 11:01:56: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
debug 2020-04-13 11:01:56: Received Zigbee message from '0x00158d00017ac453', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"zonestatus":17,"extendedstatus":0}' from endpoint 1 with groupID 0
[... cut ...]
info  2020-04-13 11:18:45: MQTT publish: topic 'zigbee2mqtt/0x00158d00042152f8', payload '{"battery":100,"voltage":3025,"contact":true,"linkquality":60}'
debug 2020-04-13 11:19:03: Received Zigbee message from '0x0015bc001a010388', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"zonestatus":49,"extendedstatus":0}' from endpoint 35 with groupID 0
warn  2020-04-13 11:19:03: Received message from unsupported device with Zigbee model 'MotionSensor51AU'
warn  2020-04-13 11:19:03: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
`

Koenkk, 2020-04-13:
Please try with the changes from: https://github.com/Koenkk/zigbee-herdsman-converters/blob/3339/devices.js (how to get to `devices.js`: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html)

Next time when providing logs, please post them on pastebin.com and link them here, otherwise this issue will get very long.

- AU-A1ZBRC: as this has 4 buttons, please provide the log when pressing each buttons
- 5AA-SS-ZA-H0: should work now
- MOSZB-130: should work now
- AU-A1ZBPIRS: should work now

mantis64, 2020-04-13:
Dear @Koenkk, thanks for this.  

I thought I had pressed all the buttons, appologies if I had missed a part of the log out.

Also I thought you may be able to use the configuration for an already supported 4 button remote. which I think is exactly the same but with a different model number:

https://www.zigbee2mqtt.io/devices/6ARCZABZH.html

Are you not able to use the same config from this?

I will re-install edge version and try out the motion sensors, so I just need the have the latest devices.js and the motion sensors will work, will they be auto discovered with the latest edge version?

mantis64, 2020-04-13:
I think I did this correctly this time. I pressed all the buttons on the remote a few times:

https://pastebin.com/pwZNpYbX

Not sure if it worked, but please let me know if not done properly

Koenkk, 2020-04-14:
@mantis64 

- What is the ieeeAddr of the remote?
- Just replace the devices.js file, I won't enable the auto discovery but you should at least see the correct messages published to MQTT.

mantis64, 2020-04-16:
@Koenkk 

How do I find the ieeeAddr, I think the messages are very delayed, I get strange messages relating to the remote later on in the log, however I have already taken the batteries out.  

Also none of the motion sensors are working.  This is the error for the AU-A1ZBPIRS

---------------------------------------------------

zigbee2mqtt:info  2020-04-16 14:46:47: MQTT publish: topic 'zigbee2mqtt/0x0015bc001a010388', payload '{"occupancy":true,"tamper":false,"battery_low":false,"linkquality":63}'
(node:236) UnhandledPromiseRejectionWarning: TypeError: Cannot read property 'includes' of undefined
    at HomeAssistant.onPublishEntityState (/app/lib/extension/homeassistant.js:1663:59)
    at EventBus.<anonymous> (/app/lib/extension/homeassistant.js:1570:63)
    at EventBus.emit (events.js:223:5)
    at EventBus.emit (/app/lib/eventBus.js:15:15)
    at Controller.publishEntityState (/app/lib/controller.js:292:23)
    at publish (/app/lib/extension/deviceReceive.js:146:22)
    at DeviceReceive.onZigbeeEvent (/app/lib/extension/deviceReceive.js:176:13)
    at Controller.callExtensionMethod (/app/lib/controller.js:336:44)
    at Controller.onZigbeeEvent (/app/lib/controller.js:222:14)
    at Zigbee.emit (events.js:223:5)
(node:236) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 5)

mantis64, 2020-04-16:
Do you mean this:

info  2020-04-16 14:44:36: 0x00158d0002123314 (0x00158d0002123314): AU-A1ZBRC - Aurora Lighting AOne smart remote (EndDevice)

If so I have pressed the buttons many times but the log does not show anything for this device apart from when it anounces itself

mantis64, 2020-04-16:
Here is the log for the MOSZB-130 

zigbee2mqtt:info  2020-04-16 17:53:51: MQTT publish: topic 'zigbee2mqtt/0x0015bc001a0073d4', payload '{"occupancy":true,"tamper":false,"battery_low":false,"linkquality":36}'
(node:236) UnhandledPromiseRejectionWarning: TypeError: Cannot read property 'includes' of undefined
    at HomeAssistant.onPublishEntityState (/app/lib/extension/homeassistant.js:1663:59)
    at EventBus.<anonymous> (/app/lib/extension/homeassistant.js:1570:63)
    at EventBus.emit (events.js:223:5)
    at EventBus.emit (/app/lib/eventBus.js:15:15)
    at Controller.publishEntityState (/app/lib/controller.js:292:23)
    at publish (/app/lib/extension/deviceReceive.js:146:22)
    at DeviceReceive.onZigbeeEvent (/app/lib/extension/deviceReceive.js:176:13)
    at Controller.callExtensionMethod (/app/lib/controller.js:336:44)
    at Controller.onZigbeeEvent (/app/lib/controller.js:222:14)
    at Zigbee.emit (events.js:223:5)
(node:236) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 10)

Koenkk, 2020-04-17:
that includes exception should be fixed in the latest hassio edge.

mantis64, 2020-04-17:
@Koenkk The leedarson motion sensor does not show this error and it seems to work as it should.  However I cannot make use of it as it is not auto discovered.  

Will you turn on auto discovered in the latest edge for the 3 different motion sensors?

No luck getting any MQTT messages from the 4 button remote so far.

mantis64, 2020-04-19:
Just some updates on the motion sensors.  

I have been testing with some automations to see if the motion sensor is working as it should.  If I do an MQTT trigger with the motion sensors I can see if it works.  

With the 'MOSZB-130 - Develco Motion sensor' and 'AU-A1ZBPIRS - Aurora Lighting AOne PIR sensor' 

If I do a condition template of '{{ trigger.payload_json.occupancy == true }}' when the motion sensor is triggered the automation happens as it should.  

However with the '5AA-SS-ZA-H0 - Leedarson Motion sensor' it does not work.  

Looking at the z2mqtt log I can see that it has a different message for occupancy:

`zigbee2mqtt:debug 2020-04-19 20:37:43: Received Zigbee message from '0x00158d00017ac453', type 'attributeReport', cluster 'msOccupancySensing', data '{"occupancy":1}' from endpoint 1 with groupID 0`

So I tried:  '{{ trigger.payload_json.occupancy ==  1}}', however this does not work and does not trigger an automation.  

Perhaps the above will help in some way as the leedarson motion sensor seems to work in a different way to the other 2 motion sensors.  

Can you please enable auto discovery so that I can utilise these motion sensors properly. 

Many thanks again.

Koenkk, 2020-04-20:
Please try if the 5AA-SS-ZA-H0 works with: https://github.com/Koenkk/zigbee-herdsman-converters/blob/3339/devices.js

If it does I will merge it and add discovery support.

mantis64, 2020-04-20:
@Koenkk thats great thank you.  Its now working as it should!  

I still can't get the 4 button remote to work.  I have 4 of them and managed to pair 2 of them previously.  However the remaining 2 I cannot seem to pair anymore, since I wanted to see if the other 2 I paired were faulty.  

However none of them seem to pair anymore, even the original 2 that I paired before.  Surely they can't all be faulty?

mantis64, 2020-04-20:
Ok, I have looked in the log and there does seem to be some messages from the remote:

Here is the pastbin link to my log:

https://pastebin.com/QSHLUSyT

The ieeeAddr for this remote is:

0x00158d00021a5040

Hope this can help

Koenkk, 2020-04-21:
Those messages are not from clicking buttons, please try with: https://github.com/Koenkk/zigbee-herdsman-converters/blob/3339/devices.js , if it doesn't work also try repairing the device.

mantis64, 2020-04-22:
Hi @Koenkk 

There seems to be some messages now :) here is the log:

https://pastebin.com/uxa75HKU

The ieeeAddr is 0x00158d00021a5040

Hope you can use to integrate it.

Many thanks for all your help and work.

Koenkk, 2020-04-22:
Added all devices + discovery support, please check the latest edge addon in 5 hours from now.

mantis64, 2020-04-27:
Dear @Koenkk 

Thank  you for enabling autodiscovery.  I can now utilise these however the Aurora 'AU-A1ZBPIRS' and Develco 'MOSZB-130' motion sensors are not showing the light/lux levels.  The leedarson does show the light but these 2 are not showing this in lovelace.  Additionally the MOSZB-130 has a temperature sensure.  

Do you need anything else in order to support the light and temperature sensors. 

Also the 4 button remote, the bottom button does not work, but does seem to send a message to z2m.  

zigbee2mqtt:info  2020-04-27 17:16:46: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"0x00158d00021a5040"}}'
zigbee2mqtt:debug 2020-04-27 17:16:47: Received Zigbee message from '0x00158d00021a5040', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-04-27 17:16:47: MQTT publish: topic 'zigbee2mqtt/0x00158d00021a5040', payload '{"linkquality":39,"action":"on"}'
zigbee2mqtt:info  2020-04-27 17:16:47: MQTT publish: topic 'zigbee2mqtt/0x00158d00021a5040', payload '{"linkquality":39,"action":""}'
zigbee2mqtt:info  2020-04-27 17:16:47: MQTT publish: topic 'zigbee2mqtt/0x00158d00021a5040/action', payload 'on'
zigbee2mqtt:debug 2020-04-27 17:16:47: Received Zigbee message from '0x90fd9ffffe541936', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"manufacturerCode":4456,"imageType":4385,"fileVersion":286417456}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-04-27 17:16:47: Received Zigbee message from '0x00158d00021a5040', type 'commandOff', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-04-27 17:16:47: MQTT publish: topic 'zigbee2mqtt/0x00158d00021a5040', payload '{"linkquality":39,"action":"off"}'
zigbee2mqtt:info  2020-04-27 17:16:47: MQTT publish: topic 'zigbee2mqtt/0x00158d00021a5040', payload '{"linkquality":39,"action":""}'
zigbee2mqtt:info  2020-04-27 17:16:47: MQTT publish: topic 'zigbee2mqtt/0x00158d00021a5040/action', payload 'off'
zigbee2mqtt:debug 2020-04-27 17:16:48: Received Zigbee message from '0x00158d00021a5040', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-04-27 17:16:48: MQTT publish: topic 'zigbee2mqtt/0x00158d00021a5040', payload '{"linkquality":39,"action":"on"}'
zigbee2mqtt:info  2020-04-27 17:16:48: MQTT publish: topic 'zigbee2mqtt/0x00158d00021a5040', payload '{"linkquality":39,"action":""}'
zigbee2mqtt:info  2020-04-27 17:16:48: MQTT publish: topic 'zigbee2mqtt/0x00158d00021a5040/action', payload 'on'
zigbee2mqtt:debug 2020-04-27 17:16:48: Received Zigbee message from '0x00158d00021a5040', type 'commandStep', cluster 'genLevelCtrl', data '{"stepmode":0,"stepsize":26,"transtime":3}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-04-27 17:16:48: MQTT publish: topic 'zigbee2mqtt/0x00158d00021a5040', payload '{"linkquality":39,"action":"brightness_step_up","action_step_size":26}'
zigbee2mqtt:info  2020-04-27 17:16:48: MQTT publish: topic 'zigbee2mqtt/0x00158d00021a5040', payload '{"linkquality":39,"action":""}'
zigbee2mqtt:info  2020-04-27 17:16:48: MQTT publish: topic 'zigbee2mqtt/0x00158d00021a5040/action', payload 'brightness_step_up'
zigbee2mqtt:debug 2020-04-27 17:16:48: Received Zigbee message from '0x00158d00021a5040', type 'commandStep', cluster 'genLevelCtrl', data '{"stepmode":1,"stepsize":26,"transtime":3}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-04-27 17:16:48: MQTT publish: topic 'zigbee2mqtt/0x00158d00021a5040', payload '{"linkquality":39,"action":"brightness_step_down","action_step_size":26}'
zigbee2mqtt:info  2020-04-27 17:16:48: MQTT publish: topic 'zigbee2mqtt/0x00158d00021a5040', payload '{"linkquality":39,"action":""}'
zigbee2mqtt:info  2020-04-27 17:16:48: MQTT publish: topic 'zigbee2mqtt/0x00158d00021a5040/action', payload 'brightness_step_down'
zigbee2mqtt:debug 2020-04-27 17:16:49: Received Zigbee message from '0x00158d00021a5040', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"manufacturerCode":4456,"imageType":4609,"fileVersion":285278246}' from endpoint 1 with groupID 0
zigbee2mqtt:error 2020-04-27 17:16:49: Failed to call 'OTAUpdate' 'onZigbeeEvent' (TypeError: Cannot read property 'commandResponse' of undefined
    at OTAUpdate.onZigbeeEvent (/app/lib/extension/otaUpdate.js:59:24)
    at Controller.callExtensionMethod (/app/lib/controller.js:348:44))

Koenkk, 2020-04-28:
- Please use devices.js from: https://github.com/Koenkk/zigbee-herdsman-converters/blob/3339/devices.js, with this AU-A1ZBPIRS and MOSZB-130 should start reporting illuminance (no auto discovery yet but check if it's in the published MQTT payload). If it doesn't work share your `database.db` entries of these devices. Same goes for MOSZB-130 temperature.

> The leedarson does show the light but these 2 are not showing this in lovelace
- If the sensor in HA is showing values, this is not a z2m issue, probably you need to configure HA correctly

> Also the 4 button remote, the bottom button does not work, but does seem to send a message to z2m.

Looks like it's working as it's sending messages with `action` in it. See https://www.zigbee2mqtt.io/integration/home_assistant.html#responding-to-button-clicks how to use it. The exception you got should be fixed in latest dev branch.

mantis64, 2020-04-29:
With  https://github.com/Koenkk/zigbee-herdsman-converters/blob/3339/devices.js

Z2M does not start and I get the error below.  If I use an older devices.js it starts as it should.

The error is below:

----------------------------

npm ERR! errno 1
npm ERR! zigbee2mqtt@1.12.2 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.12.2 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2020-04-29T09_43_34_623Z-debug.log
2020-04-29T10:43:36: PM2 log: App [npm:0] exited with code [1] via signal [SIGINT]
2020-04-29T10:43:36: PM2 log: App [npm:0] starting in -fork mode-
2020-04-29T10:43:36: PM2 log: App [npm:0] online
> zigbee2mqtt@1.12.2 start /zigbee2mqtt-1.12.2
> node index.js
/zigbee2mqtt-1.12.2/node_modules/zigbee-herdsman-converters/index.js:9
    for (const zigbeeModel of device.zigbeeModel) {
                                     ^
TypeError: device.zigbeeModel is not iterable
    at Object.<anonymous> (/zigbee2mqtt-1.12.2/node_modules/zigbee-herdsman-converters/index.js:9:38)
    at Module._compile (internal/modules/cjs/loader.js:955:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:991:10)
    at Module.load (internal/modules/cjs/loader.js:811:32)
    at Function.Module._load (internal/modules/cjs/loader.js:723:14)
    at Module.require (internal/modules/cjs/loader.js:848:19)
    at require (internal/modules/cjs/helpers.js:74:18)
    at Object.<anonymous> (/zigbee2mqtt-1.12.2/lib/util/utils.js:1:34)
    at Module._compile (internal/modules/cjs/loader.js:955:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:991:10)
npm
 ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.12.2 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.12.2 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2020-04-29T09_43_45_333Z-debug.log
2020-04-29T10:43:47: PM2 log: App [npm:0] exited with code [1] via signal [SIGINT]
2020-04-29T10:43:47: PM2 log: App [npm:0] starting in -fork mode-
2020-04-29T10:43:47: PM2 log: App [npm:0] online
> zigbee2mqtt@1.12.2 start /zigbee2mqtt-1.12.2
> node index.js
/zigbee2mqtt-1.12.2/node_modules/zigbee-herdsman-converters/index.js:9
    for (const zigbeeModel of device.zigbeeModel) {
                                     ^
TypeError: device.zigbeeModel is not iterable
    at Object.<anonymous> (/zigbee2mqtt-1.12.2/node_modules/zigbee-herdsman-converters/index.js:9:38)
    at Module._compile (internal/modules/cjs/loader.js:955:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:991:10)
    at Module.load (internal/modules/cjs/loader.js:811:32)
    at Function.Module._load (internal/modules/cjs/loader.js:723:14)
    at Module.require (internal/modules/cjs/loader.js:848:19)
    at require (internal/modules/cjs/helpers.js:74:18)
    at Object.<anonymous> (/zigbee2mqtt-1.12.2/lib/util/utils.js:1:34)
    at Module._compile (internal/modules/cjs/loader.js:955:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:991:10)
npm
 ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.12.2 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.12.2 start script.
npm ERR!
 This is probably not a problem with npm. There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2020-04-29T09_43_56_464Z-debug.log
2020-04-29T10:43:57: PM2 log: App [npm:0] exited with code [1] via signal [SIGINT]
2020-04-29T10:43:57: PM2 log: App [npm:0] starting in -fork mode-
2020-04-29T10:43:58: PM2 log: App [npm:0] online
> zigbee2mqtt@1.12.2 start /zigbee2mqtt-1.12.2
> node index.js

mantis64, 2020-05-13:
Hi @Koenkk 

The MOSZB-130 doesn't appear to be reporting back illuminance.  Here is the log:

```
debug 2020-05-13 15:22:40: Received Zigbee message from '0x0015bc001a0073d4', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"zonestatus":52,"extendedstatus":0}' from endpoint 35 with groupID 0
info  2020-05-13 15:22:40: MQTT publish: topic 'zigbee2mqtt/0x0015bc001a0073d4', payload '{"occupancy":false,"tamper":true,"battery_low":false,"linkquality":34}' 
```
Also here is the database:

```
0x0015bc001a0073d4","nwkAddr":24148,"manufId":4117,"manufName":"Develco Products A/S","powerSource":"Battery","modelId":"MOSZB-130","epList":[1,34,35,38,39,40,41],"endpoints":{"1":{"profId":49353,"epId":1,"devId":1,"inClusterList":[3,5,6],"outClusterList":[],"clusters":{},"binds":[]},"34":{"profId":260,"epId":34,"devId":263,"inClusterList":[0,3,1030],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"dateCode":"2018-04-16 08:00"}}},"binds":[]},"35":{"profId":260,"epId":35,"devId":1026,"inClusterList":[0,1,3,15,32,1280],"outClusterList":[10,25],"clusters":{"genBasic":{"attributes":{}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00124b000be86d6c","zoneState":1}}},"binds":[]},"38":{"profId":260,"epId":38,"devId":770,"inClusterList":[0,3,1026],"outClusterList":[3],"clusters":{"genBasic":{"attributes":{}},"msTemperatureMeasurement":{"attributes":{"measuredValue":1875}}},"binds":[{"cluster":1026,"type":"endpoint","deviceIeeeAddress":"0x00124b000be86d6c","endpointID":1}]},"39":{"profId":260,"epId":39,"devId":262,"inClusterList":[0,3,1024],"outClusterList":[],"clusters":{"genBasic":{"attributes":{}},"msIlluminanceMeasurement":{"attributes":{"measuredValue":18473}}},"binds":[{"cluster":1024,"type":"endpoint","deviceIeeeAddress":"0x00124b000be86d6c","endpointID":1}]},"40":{"profId":260,"epId":40,"devId":263,"inClusterList":[0,3,1030],"outClusterList":[],"clusters":{"genBasic":{"attributes":{}}},"binds":[]},"41":{"profId":260,"epId":41,"devId":263,"inClusterList":[0,3,1030],"outClusterList":[],"clusters":{"genBasic":{"attributes":{}}},"binds":[]}},"dateCode":"2018-04-16 08:00","zclVersion":1,"interviewCompleted":true,"meta":{"configured":1},"lastSeen":1589379764803}
{"id":56,"type":"EndDevice","ieeeAddr":"

The AU-A1ZBPIRS is reporting illuminance, so I just need to wait for auto-discovery on this one.  

For the 4 button remote, I am still not getting anything for the last bottom button.  In Z2M log it does not register anything for the last button but I do get this error in the log:

```
zigbee2mqtt:error 2020-05-13 15:38:03: Failed to call 'Receive' 'onZigbeeEvent' (TypeError: Cannot read property 'type' of undefined
    at /app/lib/extension/receive.js:105:42
    at Array.filter (<anonymous>)
    at Receive.onZigbeeEvent (/app/lib/extension/receive.js:104:65)
    at Controller.callExtensionMethod (/app/lib/controller.js:350:44)
    at Controller.onZigbeeEvent (/app/lib/controller.js:236:14)
    at Zigbee.emit (events.js:223:5)
    at Controller.<anonymous> (/app/lib/zigbee.js:67:52)
    at Controller.emit (events.js:223:5)
    at Controller.<anonymous> (/app/node_modules/zigbee-herdsman/dist/controller/controller.js:496:22)
    at Generator.next (<anonymous>))
```
And I am using the devices.js from 3339

Hope you can help.

Koenkk, 2020-05-14:
For the `MOSZB-130` I see an illuminance level in the database. `"msIlluminanceMeasurement":{"attributes":{"measuredValue":18473}}},"binds":`, so I would also expect it to send that to MQTT, maybe wait a bit longer?

Regarding the button, this error occurs when zigbee2mqtt is outdated. Please update zigbee2mqtt to 1.13.0 and reapply the `devices.js` file.

closed 2020-07-20 by stale[bot]

oronejustin-cloud, 2026-08-30:
Its a good camera

oronejustin-cloud, 2026-08-30:
Notaccepting anymore