Koenkk/zigbee2mqtt #3340: Can't pair Xiaomi A6121 TUYA Version

dersparkiie, 2020-04-11:
I have problems pairing a Xiaomi A6121 Smart Lock. After pairing I get a message that the device is not support!

The Box is marked with TUYA Version. Does this  make any difference?

Thanks for your support!

closed 2020-04-13 by dersparkiie

angeck, 2020-05-08:
Hello,

J'ai le même problème.

APrès plusieurs recherches je me rends compte qu'il y a 2 modèles.
Vima fabrique pour Xiaomi le Vima Smart Lock A6121
Vima fabrique pour lui même le Vima Lock A6121

Le premier marche avec Mi Home et le second avec Tuya Smart. Il s'agit exactement des mêmes appareils mais ils n'ont pas le même nom enregistré pour Zigbee donc le Vima Lock n'est pas pris en charge dans zigbee2mqtt.

Je suis entrain d'essayer de le rajouter mais les lignes de commandes ce n'est pas mon truc donc je galère.

J'ai bien réussi à le lier, ajouter dans device.js
{
    zigbeeModel: ['TY0A01'],
    model: 'A6121', // 
    vendor: 'VIMA', // 
    description: 'VIMA LOCK ',
    supports: 'inserted, forgotten and key error',
    fromZigbee: [fz.xiaomi_lock_report],
    toZigbee: [],
},

En prenant le même contener que le Xiaomi mais après rien de plus... pour le moment.

angeck, 2020-05-09:
zigbee2mqtt:info  2020-05-09 08:44:46: Successfully interviewed '0xec1bbdfffe781b28', device has successfully been paired
zigbee2mqtt:info  2020-05-09 08:44:46: Device '0xec1bbdfffe781b28' is supported, identified as: VIMA VIMA LOCK  (A6121)
zigbee2mqtt:info  2020-05-09 08:44:46: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"0xec1bbdfffe781b28","model":"A6121","vendor":"VIMA","description":"VIMA LOCK ","supported":true}}'

Pour le moment je bloque là. Avec le conteneur de Xiaomi ou rien, je n'ai pas de retour d'info. Uniquement 

zigbee2mqtt:debug 2020-05-09 08:43:50: Received Zigbee message from '0xec1bbdfffe781b28', type 'attributeReport', cluster 'genBasic', data '{"65506":22,"appVersion":85}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-05-09 08:43:55: Received Zigbee message from '0xec1bbdfffe781b28', type 'attributeReport', cluster 'genBasic', data '{"65503":"�)I&\u0013�)I&\u0012"}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-05-09 08:44:26: Received Zigbee message from '0xec1bbdfffe781b28', type 'readResponse', cluster 'genBasic', data '{"modelId":"TY0A01","manufacturerName":"_TYST12_C5IaBuQl"}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-05-09 08:44:31: Received Zigbee message from '0xec1bbdfffe781b28', type 'readResponse', cluster 'genBasic', data '{"powerSource":3,"zclVersion":3}' from endpoint 1 with groupID 0